technyon/nuki_hub #272: "Register as app" functionality does not work

micronax, 2024-01-17:
Hi again 🙂 

I tried using the `NUKI Bridge is running alongside NUKI Hub (needs re-pairing if changed)` functionality as it is easier to create eg. new keypad passcodes remotely with the Nuki app instead of doing this via MQTT.

I did re-pair the esp32 nukihub and afterwards it shows:

NUKI Lock paired | Yes (BLE Address <redacted>)
-- | --
NUKI Lock state | undefined

Lock operations are not possible and fail.

All other information published in MQTT are also missing:

[image]

Here comes my system information:

```
NUKI Hub version: 8.29
run: true
deviceId:  <redacted>
deviceIdOp:  <redacted>
mqttbroker: <redacted>
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: nukiopener
maxkpad: 1
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
rssipb: 60
hostname: esp32-nukihub
nettmout: -1
restdisc: false
rstbcn: -1
lockStInterval: 1800
configInterval: 600
batInterval: 1800
kpInterval: 600
kpEnabled: true
accLvl: 0
regAsApp: true
nrRetry: 3
rtryDelay: 100
crdusr: ***
crdpass: ***
pubauth: true
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 
Lock hardware version: 
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: No
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 1 minutes
Heap: 83940
Stack watermarks: nw: 5976, nuki: 680, pd: 264
Restart reason FW: ConfigurationUpdated
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

If I disable the "register as app" function and re-pair as bridge, status as well as lock operations work again.

technyon, 2024-01-18:
I'm not sure why this would happen, but I had some trouble when re-pairing too. For me it helped to factory reset the ESP before flashing, so that all settings are deleted.

micronax, 2024-01-18:
OK thank you, I will also try that. Maybe it also helps with my other problem in https://github.com/technyon/nuki_hub/issues/270. I will let you know later today.

micronax, 2024-01-22:
After re-installing, this worked. Thank you1

closed 2024-01-22 by micronax