technyon/nuki_hub #273: With a deauth-command, the WIFI access point opens!

StopAndG0, 2024-01-22:
Hallo,

if I disable the Nuki hub with a deauth-command, the access point opens and I can connect the hub to a new WLAN!
I think this is a major security flaw. Can i deactivate the access point?

Sorry for my bad English, it's a google-translate.

technyon, 2024-01-23:
This is the default behavior of the Wifimanager library, so you can reconfigure in case no connection is necessary. I see your point though, this could be used for an attack. At this time you can't disable the access point, I'll look into adding it as an option.

For now to mitigate it, you should add a username and password so that at least the broker can't be reconfigured.

StopAndG0, 2024-01-24:
Hello,

thanks for the quick feedback.

OK, so far it works. But I can't save MQTT or the Network- configuration with this Firmware. If I do that, I get kicked off the Network and it opens the access point to reconnect.

Thank you for your great work!

technyon, 2024-01-24:
Hi. The code for saving configuration hasn't been touched at all compared to the released 8.29 binary, and I have no problem at all saving it. Can you get serial logs to see what's going on? Speed is 115200 baud.

StopAndG0, 2024-01-24:
ok, i haven't done that yet, it's a new hobby for me. I'll read up on it and give it a try.

technyon, 2024-01-24:
Connect your ESP to your PC, and then use for example HTERM to read serial output of the ESP

StopAndG0, 2024-01-24:
Hi, I did it with Arduino IDE. I reflashed the ESP (M5StackATOM LITE) and connected it to my WLAN. Here is the log when configuring the MQTT broker:

`MQTT Broker not configured, aborting connection attempt.
Restarting
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
E (552) esp_core_dump_flash: �� core dump partition found!
E (552) esp_core_dump_flash: No core dump partition found!
NUKI Hub version 8.29-pre-2
IP address empty, falling back to DHCP.
IP configuration: IP address: 0.0.0.0, Subnet: 0.0.0.0, Gateway: 0.0.0.0, DNS: 0.0.0.0
Hardware detect     : 0
Network device: Wifi only
MQTT without TLS.
*wm:AutoConnect 
*wm:No wifi saved, skipping 
*wm:AutoConnect: FAILED 
*wm:StartAP with SSID:  ESP32_9FDAD4D4
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal `

StopAndG0, 2024-01-25:
Hi. Sorry for my mistake. I did not flash the new firmware via OTA, but directly. I would not have thought that OTA is possible, but that's how it works.

Thank you very much for your support!

technyon, 2024-01-25:
OTA should be possible actually. It works now?

mundschenk-at, 2024-01-27:
I agree, otherwise the ESP doesn't come back online after a WiFi outage.

technyon, 2024-01-27:
ok, I'll look into it. At least for now reconfiguring Wifi is prevented.

technyon, 2024-01-30:
Here's a binary to test. You'll find the option under the selection for network hardware. The behavior is slighly different depending on which hardware is configured:

- If "Wifi only" is selected, the Wifi configuration portal is disabled, and the ESP will just reboot until it can eventually connect to Wifi again
- If some LAN hardware is elected, fallback to the Wifi device is disabled, and thus the Wifi config portal will not open. The ESP will reboot until it can connect to LAN again.

To the user, both behaviors should look very similar. Please test, if it works as intended, I'll release a new version. 

[nuki_hub-8.31-pre-1.zip](https://github.com/technyon/nuki_hub/files/14098448/nuki_hub-8.31-pre-1.zip)

StopAndG0, 2024-01-30:
Hi, thanks for the quick response! I'll get back to you when I've tested it....

alexdelprete, 2024-01-30:
> if I disable the Nuki hub with a deauth-command, the access point opens and I can connect the hub to a new WLAN!
> I think this is a major security flaw

Can you explain why this is a security flaw? Firt of all, the attacker would have access to the LAN, then, it would have to have the admin rights to issue a deauth command, then it would have to reconfigure Nuki Hub to another local wifi, but you would notice it because Nuki Hub would not be reachable anymore.

When I install Tasmota or other fw, it goes in AP mode so you can connect to it and configure the wifi. I don't see that as a "major security flaw" at all. But probably I'm missing something, that's why I ask...:)

StopAndG0, 2024-01-30:
I just had time and tried it out.

Only tested with "Wifi only", I don't have an ESP32 with LAN connection. 

The Wifi configuration portal is deactivated, which is OK. But the ESP no longer logs into the WLAN after the attack, even after 5 minutes.

technyon, 2024-01-30:
With reboot you mean you have to power-cycle it.

StopAndG0, 2024-01-30:
yes, or with the reset button, it works with both

technyon, 2024-01-30:
What software do you use to send the deauth packets?

technyon, 2024-01-31:
I've added some code to deinitialize the wifi driver before rebooting, please give it a try.

Also, upon sending the deauth give it some time to reconnect. I didn't try with deauth, but just renamed by AP SSID. After renaming it back to the correct name, it took about 30 seconds to reconnect.

[nuki_hub-8.31-pre-2.zip](https://github.com/technyon/nuki_hub/files/14112080/nuki_hub-8.31-pre-2.zip)

StopAndG0, 2024-01-31:
Great, now it works! 

I have now tested it several times. No AP is displayed during the attack and after the attack the ESP is back in the WLAN after 20 seconds at the latest.

Many Thanks

technyon, 2024-02-01:
Included in release 8.31, issue resolved :)

closed 2024-02-01 by technyon