technyon/nuki_hub #274: Are you using this with a full local installation?

cagnulein, 2024-01-23:
Hi, thanks for this great application!
I just installed and it works flawlessly!

Just a thought: i'm not a fan of closed ecosystem so i would like to DON'T use the nuki app at all for my phone and my wife's phone and use this and the hass integration to do everything.

Does anyone use this in a similar scenario? My idea is that i don't want any IoT device dependent from the cloud or from any 3rd party cloud service 

I'm planning to donate to this project in the next weeks if i will keep the nuki device for my house

Thanks!

bartsnijder, 2024-01-23:
The Nuki app is standby for any possible firmware updates, but other than that the NUKI is indeed used through nuki_hub only (through mqtt/ domotica software). Works flawlessly.

cagnulein, 2024-01-23:
ok so how could we handle the bluetooth authtentication in this scenario without the Nuki app? using the HASS state for each person?

bartsnijder, 2024-01-23:
I do not follow what you mean. The bridge is connected/authenticated to the lock through bluetooth (as only device). The bridge has now become your primary interface (through MQTT). How you deal with e.g. multi-users, the ability to see the locked status and/or (un)lock at different times has now shifted to your domotica software.  I do not us HASS (but Openhab), but elaborating on this might be better done on a HASS forum I guess.

cagnulein, 2024-01-23:
Yes indeed "How you deal with e.g. multi-users, the ability to see the locked status and/or (un)lock at different times has now shifted to your domotica software."

And so you asked to my question, you're using the ability of the domotica software to unlock or lock the nuki based on certain event. Got it.

so @bartsnijder are you using the nuki without the nuki app at all on your setup?

bartsnijder, 2024-01-23:
In the daily setup: no I do not use the NUKI app.
But I do have the app standby for possible firmware updates.

cagnulein, 2024-01-23:
ok thanks for the clarification!

Roberto Viola
Software engineer and open source enthusiast
http://robertoviola.cloud

bartsnijder, 2024-01-23:
I use a few means to automatically open the lock
* BLE presence (of NUKI hub): I do limit based on BLE signal strength
* Mobile phone and/or car connected to my LAN on a specific AP (+signal strength) (this does have some extra delay)

cagnulein, 2024-01-23:
ok so you have also a nuki hub...

Roberto Viola
Software engineer and open source enthusiast
http://robertoviola.cloud

bartsnijder, 2024-01-23:
No, I edited the command. By accident I named it a HUB. I only have the
bridge.

technyon, 2024-01-23:
The app on the smartphone is not necessarily cloud connected. If you use it without the NUKI Bridge, it directly connects to the lock via bluetooth without communicating with the cloud.

cagnulein, 2024-01-23:
> No, I edited the reply. By accident I named it a HUB. I only have the bridge.

sorry, just to understand correctly, you mean you only have the ESP32 with the nuki hub? if so, does it expose any data from the bluetooth devices around it?

> The app on the smartphone is not necessarily cloud connected. If you use it without the NUKI Bridge, it directly connects to the lock via bluetooth without communicating with the cloud.

hah i didn't know this! thanks

bartsnijder, 2024-01-23:
> > No, I edited the reply. By accident I named it a HUB. I only have the bridge.
> 
> sorry, just to understand correctly, you mean you only have the ESP32 with the nuki hub? if so, does it expose any data from the bluetooth devices around it?

Yes it does, do you have an MQTT browser/viewer (like mqtt explorer). It states all the BLE device found (including signal strength). I think your phone BLE will not be visible by default though; might need an app for that. I use different BLE devices as keys.

cagnulein, 2024-01-23:
@bartsnijder that's great! i didn't know this! I'm installing https://community.home-assistant.io/t/addon-mqtt-explorer-new-version/603739 (for future reference) to check it and I will report it back. If it's so I guess I'm ready to give it a go! Thanks!

technyon, 2024-01-23:
> Yes it does, do you have an MQTT browser/viewer (like mqtt explorer). It states all the BLE device found (including signal strength). I think your phone BLE will not be visible by default though; might need an app for that. I use different BLE devices as keys.

Correct, you have to install an app that broadcasts ble beacons for your phone to become visible for NUKI Hub. Also, it has to broadcast a name, otherwise the beacon will be ignored.

cagnulein, 2024-01-23:
> Correct, you have to install an app that broadcasts ble beacons for your phone to become visible for NUKI Hub. Also, it has to broadcast a name, otherwise the beacon will be ignored.

that's awesome @technyon thanks! do you know if it also possible on ios? which app do you suggest?

technyon, 2024-01-23:
I'm not using an app myself, but a Siemens G-Tag. I think many people use the Home Assistant companion app, but that only makes sense if you're using home assistant. Just search on the play store ("beacon simulator"), there are a few apps that can do it.

I'm not sure about IOS, I'm strictly an android user :).

iranl, 2024-03-08:
In accordance with https://docs.github.com/en/get-started/using-github/communicating-on-github#scenarios-for-github-discussions this open-ended conversation is more suitable for the discussions section imho.

@technyon: Now that GitHub discussions is enabled on the repository I propose we convert this issue to a discussion.