technyon/nuki_hub #275: Trouble with 2nd NukiHub

mkomarac, 2024-01-23:
Hello.
First of all, I woud like to say that this is a great project, and a lot of effort put into it.
Couple a weeks ago, I have had one esp32 laying around, and implemented into HASS seamlessly.
Four days ago finally arrived 2nd esp32, and I wanted to add it to my 2nd Nuki lock.
renamed as noted, and all of data coming into MQTT broker.
but device never showed up in mqtt integration in home assistant as device. Restarted, rebooted HASS, but no luck
I have tried with multiple ESP's but 2nd is never discovered by mqtt integration.
Have I missed something?

If I put nuki name as default, the device showed up, but controlling 2 locks as one, what's not an option.

thanks in advance

alexdelprete, 2024-01-24:
If the correct entities are created in MQTT, the device has to show up in HA. So please check the MQTT broker, and post a screenshot of the topics of the second lock.

technyon, 2024-01-24:
Also post the contents of your the system information pages of your ESPs (click on the firmware version)

mkomarac, 2024-01-25:
Here are more details:
Both of hubs are connecting to each lock, and read states
NukiHub that is discoverable by HomeAssistant is FW v.8.28, and other one 8.29
[image]
2nd Hub path is changed as noted
[image]
MQTT Topic
[image]

MQTT Explorer
[image]
[image]

Under homeassistant path
[image]

and in Home Assistant there is only one nuki
[image]

[image]
[image]

if any more information is needed, just ask

thanks

mkomarac, 2024-01-25:
System Information
NUKI Hub version: 8.28
run: true
deviceId: 3163309255
deviceIdOp: 3163309255
mqttbroker: 10.10.10.10
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant/KitchenDoor
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
rssipb: 60
hostname: nukihub1
nettmout: -1
restdisc: false
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 0
regAsApp: false
nrRetry: 0
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 4.0.36
Lock hardware version: 8.3
Lock paired: Yes
Lock PIN set: No
Lock has door sensor: No
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 105 minutes
Heap: 90432
Stack watermarks: nw: 6104, nuki: 664, pd: 236
Restart reason FW: RequestedViaMqtt
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.

System Information
NUKI Hub version: 8.29
run: true
deviceId: 223456177
deviceIdOp: 223456177
mqttbroker: 10.10.10.10
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: main
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant/FrontDoor
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
rssipb: 60
hostname: nukihub11
nettmout: -1
restdisc: false
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 0
regAsApp: false
nrRetry: 0
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 4.0.36
Lock hardware version: 8.3
Lock paired: Yes
Lock PIN set: No
Lock has door sensor: No
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 146 minutes
Heap: 91060
Stack watermarks: nw: 6104, nuki: 664, pd: 232
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_INT_WDT: Reset (software or hardware) due to interrupt watchdog

alexdelprete, 2024-01-25:
> Here are more details:
> Both of hubs are connecting to each lock, and read states
> NukiHub that is discoverable by HomeAssistant is FW v.8.28, and other one 8.29

[image]

This is wrong configuration: you need to specify the main HA topic, not the subtopic. None of the hubs should be discoverable by HA with this configuration.

HA main topic has the entity types under it...

[image]

And under the lock entity subtopic you have the Nuki Hubs:

[image]

That is your problem, you can easily fix it.

@technyon doc and description on the homeassistant config field should be more clear about this. I would call that "Home Assistant root audotiscovery topic". And in the docs I would specify it's the root, and 99.99999% it's not to be changed.

technyon, 2024-01-25:
OK, let's update the documentation

alexdelprete, 2024-01-25:
> OK, let's update the documentation

If you can give me access just to the documentation section, I can update it.

Or, I was thinking: can you enable the wiki? We could create a sort of FAQ.

mkomarac, 2024-01-26:
> ![image](https://private-user-images.githubusercontent.com/7027842/299656588-bf4afdde-4fff-4cb4-be20-ee56a3226789.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

Yes Sir.
That's the one!
Two NukiHubs in my Home Assistant
Thank You

alexdelprete, 2024-01-26:
> Yes Sir.
> That's the one!
> Two NukiHubs in my Home Assistant
> Thank You

Glad it's working now. :)

Can you please close the issue? Thanks.

closed 2024-01-26 by mkomarac

technyon, 2024-01-26:
@alexdelprete I would give you access no problem, but that's unfortunately not possible with a free github account. You'll have to create a PR unfortunately.

alexdelprete, 2024-01-26:
> @alexdelprete I would give you access no problem, but that's unfortunately not possible with a free github account. You'll have to create a PR unfortunately.

I have repos with other contributors that have free accounts. Are you sure about that?

mundschenk-at, 2024-01-26:
Yeah, it should be possible to add collaborators in the repository settings: https://docs.github.com/en/account-and-profile/setting-up-and-managing-your-personal-account-on-github/managing-access-to-your-personal-repositories/inviting-collaborators-to-a-personal-repository

technyon, 2024-01-27:
ok then I just couldn't figure it out last time :D.

I've invited the two of you.

alexdelprete, 2024-01-27:
> I've invited the two of you.

Make sure we don't break things accidentally!! I'll limit myself to edit docs/wiki. :D

technyon, 2024-01-27:
With great power comes great responsibility :D.

I trust you not to break things.

alexdelprete, 2024-01-27:
> I trust you not to break things.

And I trust you have local backups. :)

technyon, 2024-01-27:
Git does have history.