technyon/nuki_hub #280: [Feature request] Nuki Keypad Code Limit Access Time

sjors-lemniscap, 2024-01-30:
Unfortunately due to the instability of the Nuki Cloud API I'm forced to look at alternatives and stumbled upon this open Nuki Hub implementation.

Most of the functionalities are there and super happy to see that the Nuki Keypad is supported as well. However, when using the official Nuki API there's a possibility to program a entry code with limited access time. This is being done through the following API endpoint: https://api.nuki.io/#/SmartlockAuth/SmartlockAuthsResource_put_put by setting the:
```
  "allowedFromDate": "2024-01-30T13:23:01.677Z",
  "allowedUntilDate": "2024-01-30T14:23:01.677Z",
```
Are there any plans to extend the Keypad control code creation with those 2 properties since the Nuki Keypad itself supports this functionality.

By the way, just bought you some coffees as well, love this project!

iranl, 2024-03-07:
It's on my longlist of features to develop, hopefully somewhere in the coming month.

Probably as a part of adding the option to control the keypad using JSON as an alternative to the current implementation.

Since the nuki_ble library that Nuki Hub uses already supports the functionality you're asking for it should not be that hard to include in Nuki Hub.

sjors-lemniscap, 2024-04-03:
> It's on my longlist of features to develop, hopefully somewhere in the coming month.
>
> Probably as a part of adding the option to control the keypad using JSON as an alternative to the current [...]

Thanks a lot! Any updates? Been following the progress and this would be a very welcomed addition to Nuki Hub. If there's anything I can do, please let me know. Happy to test out an early release.

iranl, 2024-04-04:
I'm currently working on finishing PR #332 which has a lot of the base code which i'll reuse for  implementing the complete api for adding/modifying/removing keypad codes.

When I finish #332 I'll start work on this request and link this issue in the PR. You should be able to beta test using build artifacts from that PR. I'm targeting this month.

sjors-lemniscap, 2024-04-04:
> I'm currently working on finishing PR #332 which has a lot of the base code which i'll reuse for implementing the complete api for adding/modifying/removing keypad codes.
>
> When I finish #332 I'll [...]

Looking forward to it, will keep an eye out!

iranl, 2024-04-12:
See #336 for the binary for testing and examples on how to format the JSON.

Still need to update the README and check with @technyon if we can target this for 8.34 or 8.35.

PR #336 (2024-04-14): Extend keypad
Todo
- [x] Publish all keypad code info to `keypad/json`
- [x] Allow changing all keypad code info through `keypad/actionJson`
- [x] Update README

Closes #280 

Adds all available keypad functions from the [API](https://developer.nuki.io/page/nuki-smart-lock-api-2/2/#heading--add-keypad-code) to Nuki Hub.

All available information from keypad codes (except codes themselves) is published to `keypad/json`

Example:

```
[
  {
    "codeId": 1234,
    "enabled": 1,
    "name": "Test",
    "dateCreated": "2024-01-01 10:00:00",
    "lockCount": 100,
    "dateLastActive": "2024-04-12  [... cut]
closed 2024-04-14 by technyon