technyon/nuki_hub #283: Upgraded to 8.30, lost BT and now can't lock/unlock

binderth, 2024-01-31:
So I upgraded to 8.30 from 8.27 and lost the Bluetooth pairing while upgrading. 

So I 

- unpaired the lock
- paired it again

Now I always get the following on the starting page:
[image]

Now everytime I enter the PIN, the nukihub restarts and the same message keeps popping up - and still no lock/unlock command goes through.

What can I do?

The configuration is like this:
```
NUKI Hub version: 8.30
run: true
deviceId: 39xxx
deviceIdOp: 38yyy
mqttbroker: 192.168.78.20
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: true
lockena: true
mqttpath: devices/NukiHub
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: 
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 2
rssipb: -1
hostname: nukihub
nettmout: -1
restdisc: true
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 
regAsApp: false
nrRetry: 5
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: true
macb0: -16
macb1: 97
macb2: -35
MQTT connected: Yes
Lock firmware version: 2.15.3
Lock hardware version: 11.1
Lock paired: Yes
Lock PIN set: No
Lock has door sensor: Yes
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 1576 minutes
Heap: 90760
Stack watermarks: nw: 6104, nuki: 488, pd: 232
Restart reason FW: NetworkDeviceCriticalFailure
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

mundschenk-at, 2024-02-01:
I would try pairing again, something likely went wrong there.

technyon, 2024-02-04:
@binderth Were you able to fix your issues?

binderth, 2024-02-04:
Oh, yes. I did a reset of my ESP and now it works as intended. Something was off I guess.

binderth, 2024-02-04:
I'll close the issue

technyon, 2024-02-04:
Perfect

iranl, 2024-02-25:
@binderth @technyon: Close?

binderth, 2024-02-25:
Yes. Wirry

closed 2024-02-25 by binderth