technyon/nuki_hub #287: Feature Request: Support for Controlling Multiple Nuki Locks with a Single ESP32

stoykow, 2024-02-04:
Hello,

I've been working with your nuki_hub project and it's been great for single-lock control. However, I have a scenario where I'd like to manage multiple Nuki locks using a single ESP32 to simplify my setup and reduce hardware redundancy.

My idea involves modifying the codebase to create and manage multiple instances of NukiLock, each with its own unique identifier, enabling the ESP32 to send commands to and receive statuses from multiple locks independently. This could potentially be achieved by extending the current implementation to handle an array of locks, each with distinct configuration parameters.

Would this be feasible with the current architecture of nuki_hub, or are there known limitations that would prevent this from being implemented? I’m willing to contribute to the development and testing of this feature if you think it's a viable addition to the project.

Looking forward to your thoughts and any guidance you can provide.

Best regards,
Niko (stoykow)

mundschenk-at, 2024-02-04:
@technyon can probably answer this in more detail, but you are going to be limited by the lack of memory, the low CPU power, and the single radio.

technyon, 2024-02-05:
This was discussed before, unfortunately I can't the issue. The point is that supporting multiple locks would add a lot of complexity to the code. Not only you need to keep track of multiple locks, the UI would need to be changed to support an unknown number of locks. Adding complexity always has a good chance to add new bugs, and considering how cheap ESPs are, it's easier if you just get a second (or third) ESP, which has the advantage that you can move them closer to each lock to improve bluetooth reception.

closed 2024-03-02 by technyon