technyon/nuki_hub #294: Lock Status and authorizations are not properly updated

miccico, 2024-02-08:
Hi! 

I got my Nuki 4 (non Pro) today and set it up without a Nuki Bridge and paired to an ESP32 flashed with NukiHub 8.31
The initial setup worked flawlessly and also homeassistant picked up the device without any problems.

When reworking my automations from my previous Homematic Keymatic opener i tried to detect manual activations (as those should never happen in our household).

When playing with the lock (unlocking by turning the knob or unlocking via key) i can see that the lock state updates to unlocked after several seconds but the authorization id remains unchanged. Also there is no indication at all that the lock was manually operated. When then querying for lockstate the authorization is updated but to a wrong value (seems to be the first from log). Also the log in the app features 30+ action where in mqtt i only see 6 that seem to be hours old. 

This screenshot is from after the system picked up the manual unlocking. Even several minutes after it remained unchanged

[image]

I have two phones authorized and the bridge - no other devices connected to the nuki.

Is there anything i need to do to enable to capture these properly? From the older issues i saw that some people reported similar problems but i did not find any solutions there.

Thanks for the great work + best regards

Michael

technyon, 2024-02-10:
There's a related issue #270. It seems there's some kind of fluke with the 4.0 locks when it comes to transmitting the log.

miccico, 2024-02-10:
That's probably true. What however bothers me a bit is that even when i disable the "publish auth data" the problem still appears.

The trigger for instance has been set to "system" once and never changed since. From my code browsing that should be part of KeyTurnerState and come independent from the log, shouldn't it? Also state has only ever been locked and unlocked - nothing inbetween ever captured. 

If i look on the log on the nuki ios app i see all actions properly. First load of the log loads 15 actions and they show everything (Manual, Manual (Button), Auto Unlock, ....). Is there a way to increase the number of log items loaded from the default 5 towards the 15 the app seemingly uses?

technyon, 2024-03-02:
@miccico Sorry for the late answer. Unfortunately 5 is the limit at this time due to using json for the log entries, which uses a lot of memory when encoding.

iranl, 2024-05-19:
There was a bug in the Nuki (4.0) firmware that prevented the correct trigger being sent, this has recently been fixed on the Nuki side in my own testing.

The authorization log is fixed in #368

closed 2024-05-21 by technyon