technyon/nuki_hub #307: [Question] Is it safe to update the firmware on our NUKI? (and will it stay safe?)

Rubeast, 2024-02-26:
Hi! 

Not really an issue, but more a question.
I saw that my Nuki had a firmware update available. 
While it might be safe now to update, do you think it will stay safe?
I was wondering if they might able to block access to this hub in the future.

BTW: Love the NUKI-Hub. I even created a custom case for it.
[image]
[image]

technyon, 2024-02-26:
Nuki is actually supporting projects like this. The API for the bluetooth communication is available directly from Nuki, and there's a developer forum to ask questions.

Nuki Hub uses the same protocol that the smartphone app and the Nuki Bridge use. If they want to lock out third party implementations, they'd have to change the protocol and force users to update all devices connected to the lock ... a difficult thing to do.

I'd say don't worry too much, as this is endorsed by Nuki.

Rubeast, 2024-02-26:
Ok. Great to hear!

closed 2024-02-26 by Rubeast