technyon/nuki_hub #310: Issue with Opener: Only Ring to Open is working not Open directly

riyas1980, 2024-02-28:
Hello,

First of all, thanks for your great work. I am trying nuki hub with my opener. I am using version 8.32

When added, my MQTT shows 2 entities (1 is enabled and 1 is disabled)

Intercom is enabled with Unlock as option
Unlatch is disabled

When I click on Unlock, it doesn't open my entrance door. The state change to Ring to Open. So If I ring my intercom, it's opening. 

I want to directly open the door without Ring to open. Is that possible ?

Thanks for your help. 

Kind Regards,

riyas1980, 2024-02-28:
And I've enabled the unlatch button in Home Assistant. But when I press, it doesn't do anything

iranl, 2024-02-28:
Unlock enables Ring to Open and lock disables Ring to Open..
Unlatch (equals Electronic Strike Actuation for the opener) opens a door.

The Unlatch button in Home Assistant only works when unlatch is enabled in the Nuki Hub configuration (Nuki Configuration -> Advanced Nuki Configuration). Set access level to "Full"

[image]

riyas1980, 2024-02-29:
> Unlock enables Ring to Open and lock disables Ring to Open. Unlatch (equals Electronic Strike Actuation for the opener) opens a door.
>
> The Unlatch button in Home Assistant only works when unlatch is [...]
Hi,

Thanks for your reply. Actually it is already in Full Access level. But When I press the Unlatch button, Opener doesn't do anything. But in logbook, we can see that : "Unlatch changed to February 28, 2024 at 12:28 PM triggered by service Button: Press 12:28:42 PM - 21 hours ago".

For your information, when MQTT discovered Nuki Hub, only "Unlock" button was visible. Unlatch was disabled by default. I've enabled it but unfortunately it's not working. 

[image]

iranl, 2024-02-29:
Unlatch button is disabled in HA by default. This is by design as a security measure.

The Home Assistant log will probably not be very helpfull here as it is only able to tell if it published something to the MQTT topic, what Nuki Hub does or doesn't do based on this can not really be discerned from within Home Assistant.

Just to make sure: You are able to unlatch/open your door without ringing using the Nuki smartphone app?

riyas1980, 2024-02-29:
> Unlatch button is disabled in HA by default. This is by design as a security measure.
>
> The Home Assistant log will probably not be very helpfull here as it is only able to tell if it published [...]

Yes I am able to unlatch/open my door without ringing from my Nuki app. I really don't know what's wrong with my HA and Nuki Hub

iranl, 2024-03-02:
Have you tried setting `nukiopener/lock/action` to `electricStrikeActuation` manually (using an app like MQTT Explorer)? 
To rule out an issue with Home Assistant. 
Please also check `nukiopener/lock/action` changes to `electricStrikeActuation` when pushing the Home Assistant "Unlatch" button.

What does `nukiopener/lock/completionStatus` and `nukiopener/lock/commandResult` show after setting action to `electricStrikeActuation`?
And does the door open when setting action to `electricStrikeActuation` manually?

riyas1980, 2024-03-04:
> Have you tried setting `nukiopener/lock/action` to `electricStrikeActuation` manually (using an app like MQTT Explorer)? To rule out an issue with Home Assistant. Please also check [...]

Thanks for your help. I am out of town. Tomorrow I'll try and inform you. Thanks again

riyas1980, 2024-03-04:
I am trying to change the setting to electricStrikeActuation with MQTT Explorer but I don't know where exactly I need to change. I am sorry. I am new to MQTT.

[image]

I've even updated the Nuki Hub to 8.33 and I can see the continuous mode also. 

With MQTT explorer, this is what I get, when I click on unlock (action : ack)

[image]

But when I click on Unlatch, I have this message (action = Unknown_action): 

[image]

In the history, I have 2 lines : First one is Unlatch and second one is Unknown_action

[image]

I don't know if my message is clear to you. I don't know how to edit the setting to "electricStrikeActuation"

Thanks again for your help

iranl, 2024-03-04:
There is a bug in the code that exposes the "Unlatch" button in Home Assistant for the opener.

When using a Opener Home Assistant should not sent a "unlatch" command but the "electricStrikeActuation" command.

PR #316 should fix this bug.

You can download the binary that is created by GitHub with these changes from https://github.com/technyon/nuki_hub/actions/runs/8147478237/artifacts/1296509546

Please try the binary and report back if your issue is solved. You will know Home Assistant has picked up the changes if the name of the "Unlatch" button has changed to "Open".

technyon, 2024-03-05:
In MQTT explorer, you should set lock/action to unlatch. When correct, Nuki Hub will respond with "ack" to the same topic.

iranl, 2024-03-05:
Good to hear it works as expected now.

Nuki Hub is known to have varying delay of a couple of seconds between issueing a command and it's execution.
This is due to the nature of the bluetooth communication between the ESP and the lock/opener.

PR #316 (2024-03-05): Fix Home Assistant Unlatch button for opener
The Unlatch button added in #284 only works for the Lock and not for the Opener because the command that is sent to nuki(opener)/lock/action is harcoded to unlatch instead of unlatch for the lock and electronicStrikeActuation for the opener.

Thix PR fixes this and changes the name of the button to "Open" instead of "Unlatch" as previously discussed.

Closes #310