technyon/nuki_hub #317: [Feature request] Lock Online/Offline Status

OzZyHH, 2024-03-05:
Hi,
I am currently configuring Nuki Hub and my Smart Lock 3.0 (non Pro) in OpenHAB. One optional MQTT field is for so called "Thing" (Device) availability status and according to readme I couldn't find a mqtt topic for this. Is this not implemented yet or do I overlook the possibility?

For example Tasmota has implemented this with "[Last Will Testament (LWT)](https://tasmota.github.io/docs/MQTT/#lwt-topic-last-will-and-testament)"

More details here: https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/

Would this be possible to add in a future release?

Keep up the great work!

Ozzy

iranl, 2024-03-05:
The LWT MQTT topic is `nuki/maintenance/mqttConnectionState`

OzZyHH, 2024-03-05:
Is this topic for the bridge or the smart lock device?

iranl, 2024-03-05:
LWT is a part of the specification for the connection between a MQTT client (Nuki Hub in this case) and a MQTT broker.

So the answer to your question is Nuki Hub / ESP / Bridge.

OzZyHH, 2024-03-05:
Ok thanks for clarification. Without using LWT would it be possible to publish the device state? For example in zigbee2mqtt every device has an availability topic with "online" or "offline" payload. "Offline" might be when there is no more communication for a given time between smart lock and esp/bridge/hub

technyon, 2024-03-06:
This is exactly what last will does. The broker will change the topic to offline, because obviously the device is offline and can't set it anymore. This is based on a timeout, so it might take a bit until it changes.

iranl, 2024-03-06:
In regard to the Nuki lock itself being "offline" (as in not connected to Nuki Hub): This does not really happen when "Restart if bluetooth beacons not received" in "Advanced Nuki Configuration" is set to a positive integer (default 60 seconds). This setting is primarly implemented because the bluetooth stack of ESP32 sometimes silently crashes but should also pick up the Nuki Lock/Opener being offline (only probable reason I can think of is dead battery or maybe having the ESP right on the edge of the bluetooth range and (unlikely) a firmware crash of the lock)

In this case the ESP will just restart whenever the set time expires.

If  you  really insist in wanting to get an offline/online state of the Lock (noone around here has ever seen the need and the OpenHab  availability status is meant for the LWT topic and not for what you are asking):

If you open http://NUKIHUBIP/debugon in your browser once you can enable publishing some additional information to the MQTT topic `maintenance/restartReasonNukiHub`. If the Nuki Hub restarted because of not receiving bluetooths beacons from the lock/opener this topic will be set to `BLEBeaconWatchdog`. This state in combination with the Nuki Hub endlessly restarting and the lock state changing to `undefined` should be a pretty good indication something is wrong with the lock/opener.

OzZyHH, 2024-03-06:
> This is exactly what last will does. The broker will change the topic to offline, because obviously the device is offline and can't set it anymore. This is based on a timeout, so it might take a bit until it changes.

Great, than I will pick up this mqtt topic for OpenHAB "Thing" status. 

@iranl you provide some good details here, good to know this is possible.

Solution is provided so issue can be closed. 
Thanks for your support!

Ozzy

closed 2024-03-06 by OzZyHH