technyon/nuki_hub #318: Controlling the Nuki in Home Assistant

chrizstone, 2024-03-05:
I successfully configured an ESP32 with Nukihub and MQTT is connected and the lock is paired. How do I control it now in HA? Ist there an Add-on? How does it work? Sorry if this is a very nooby question...

iranl, 2024-03-05:
RT(F)M: See the [README section about Home Assistant Discovery](https://github.com/technyon/nuki_hub?tab=readme-ov-file#home-assistant-discovery-optional)

TLDR: Assuming you have already setup the general MQTT integration in Home Assistant: Basically set "Home Assistant discovery topic" on the Nuki Hub "MQTT Configuration" page to "homeassistant".

chrizstone, 2024-03-06:
Ok, I did that but cannot see any Lock devices in HA. Where can I see my default discovery topic in HA?

closed 2024-03-06 by technyon