technyon/nuki_hub #320: Cant pair my Nuki Lock 2.0

chrizstone, 2024-03-06:
I am not able to pair my Nuki. Trie dseveral ESPs and Reflashed everything again and again. But I am not able to get the Nuki Hub paired with my log. Any help?

System Information

Nuki Hub version: 8.33
run: true
confVersion: 833
deviceId: 677313223
deviceIdOp: 677313223
mqttbroker: 192.168.178.72
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: false
lockena: true
mqttpath: nuki
openerena: false
openercont: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
hassConfigUrl: 
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
nwwififb: false
rssipb: 60
hostname: nukihub
nettmout: -1
restdisc: false
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 
regAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 
Lock hardware version: 
Lock paired: No
Lock PIN set: -
Lock has door sensor: No
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 1 minutes
Heap: 88852
Stack watermarks: nw: 6200, nuki: 2296, pd: 264
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.

technyon, 2024-03-07:
Hi,

Did you already factory reset the ESP before flashing? That often helps especially if the ESP was paired before.

chrizstone, 2024-03-07:
Yes, and I already tried out a brand new ESP. Same result...

technyon, 2024-03-07:
It's hard to say, but the 2.0 lock is supported and users have confirmed it works.

You can try:
- make sure your lock and ESP are in close proximity
- factory reset your lock. Not convenient, but it does help from time to time
- observe serial logs, for example with hterm. Baud rate is 115200

technyon, 2024-03-07:
2 meters should be fine.

chrizstone, 2024-03-07:
After reset of the lock. Do I have to pair with the Smartphone App first?

technyon, 2024-03-08:
No not necessarily, but it maybe makes things a bit easier, you can for example check paired devices

technyon, 2024-03-08:
Very strange. Could you check with an older version of NUKI Hub? There were quite a few changes recently, though I don't believe any related to pairing. 8.30 for example

JanBe, 2024-03-08:
I am also having issues pairing my Nuki Lock 3.0 with Nuki Hub version 8.33 on a brand new ESP.
I have a different ESP (that I want to replace) running 8.32 that had no issues pairing with the lock. Maybe this has to do with version 8.33?

JanBe, 2024-03-08:
Read your comment right after posting mine @technyon and just installed 8.32 on the new ESP. Was able to pair my Nuki on the first try. So at least for me it seems to be connected with 8.33

iranl, 2024-03-08:
Ran through the diff between 8.33 and 8.32 (https://github.com/technyon/nuki_hub/compare/8.32...8.33) but can't find any obvious changes that explains why 8.33 would have trouble pairing.

technyon, 2024-03-09:
I agree there's nothing obvious in the diff. We changed from sdk 2.0.9 to 2.0.14 though, that would update the bluetooth stack as well. Could you retry with this build? It's 8.33 rebuild with sdk 2.0.9.

[nuki_hub-8.33-sdk-2.0.9.zip](https://github.com/technyon/nuki_hub/files/14544780/nuki_hub-8.33-sdk-2.0.9.zip)

JanBe, 2024-03-09:
Thanks! I am currently on a weekend trip, will test it on Monday

chrizstone, 2024-03-09:
@technyon I tried that build. But sadly no pairing possible :(

chrizstone, 2024-03-09:
Also tried 8.30. Also no pairing possible. I also resetted the lock again and didnt paired with the app first. Also no success with Nukihub. Anything else what I can try?

chrizstone, 2024-03-09:
Ok, weird thing. Not I got it paired. I set a username and password before pairing. No, it paired without a problem. Could that be a reason with username and password didnt set?

technyon, 2024-03-09:
Do you mean credentials to log into nuki hub?

chrizstone, 2024-03-09:
Yes, exactly

JanBe, 2024-03-11:
> I agree there's nothing obvious in the diff. We changed from sdk 2.0.9 to 2.0.14 though, that would update the bluetooth stack as well. Could you retry with this build? It's 8.33 rebuild with sdk 2.0.9.

Thanks for supplying the 8.33 build with the old SDK @technyon 🙏 
I was able to test it out now. However, I still have the same error – pairing is not happening.
In the log there is just a continuous stream of
```
Nuki lock start pairing
```
messages. However, after setting the Lock PIN in the settings, there is also a continuous and very fast spam of this error message:
```
scan_evt timeout
```
interrupted by the occasional `Nuki lock start pairing`. I hope that helps!

technyon, 2024-03-11:
Operating the lock via bluetooth can be slow, it's more or less expected. It should be reliable though, you can increase the number of retries if necessary. All of this points to reception problems, though 1m is more than close enough. Does control with the smartphone app work without problems?

JanBe, 2024-03-11:
Controlling it with the smartphone app works without problems, yes. I also use Auto Unlock, which works from a considerable distance. Honestly, I am at the point where I just ordered the official Nuki bridge and opener 😅. Even when the connection between ESP and lock was working reliably < 8.33, the hub went inactive in Home Assistant after a few hours, sometimes a day and needed to be restarted.
I guess the official bridge will make it transparent if the reception problems came from the lock or the ESP.

technyon, 2024-03-11:
You can enable restart if no  beacons received, at least it should restart on it's own.

okastl, 2024-05-08:
I can confirm the pairing problem with Nuki Hub 8.33 and a Nuki Lock 4.
It is a fresh ESP32 with a fresh Nuki Lock.
After downgrading to Nuki Hub 8.32 via OTA, the lock immediately paired.

okastl, 2024-05-12:
> I agree there's nothing obvious in the diff. We changed from sdk 2.0.9 to 2.0.14 though, that would update the bluetooth stack as well. Could you retry with this build? It's 8.33 rebuild with sdk [...]

I can confirm, that with this version pairing with my Nuki Lock 4.0 worked right away. Thanks!

iranl, 2024-05-12:
I've paired ESP32, ESP32-S3 and ESP32-C3 devices without issue to a Nuki Lock 4.0 Pro using Nuki Hub 8.33 and 8.34.
As Nuki Hub has only been build for the S3 and C3 with 2.0.15 it is unlikely that there is a structural problem with the newer Arduino Core version.

iranl, 2024-05-26:
Pairing can sometimes be finicky but there is no structural problem with Nuki Hub 8.33 and up.
Pairing is confirmed working in both app and bridge mode on multiple different ESP's (all models) with Nuki 4.0 and Nuki 4.0 Pro on 8.34.

When pairing is difficult it can help to restart the ESP a couple of times (when pairing fails within 60 seconds) and restarting the pairing process on the Nuki Lock/Opener a couple of times by pressing the button. Eventually it should pair.

closed 2024-05-26 by iranl

mariomare22, 2024-09-22:
Hello!
on nuki lock 4.0 not Pro.
I am not able to pair it to esp32-s3 geek from waveshare tried firmware 8.35 and 9.00. the ESP is connected to MQTT and i can see from mqtt explorer but I the pairing of the nuki shows NO.

what else can I try?

technyon, 2024-09-22:
Move your as ESP as close to the lock as possible and keep trying.

iranl, 2024-09-22:
And make sure to put your Nuki in pairing mode using the physical button (and having pairing enabled in your Nuki device settings before pressing the button)

mariomare22, 2024-09-22:
So i need to add the device to the nuki app on my phone?
They are next to each other esp and lock

technyon, 2024-09-22:
What do you mean by "the device"? The smart lock yes you add it to the app on your phone, the ESP no. Enable pairing mode on your lock by long-pressing the button on your lock, then power up the ESP, pairing should be automatic.

technyon, 2024-09-23:
@mariomare22  So did it finally work?

mariomare22, 2024-09-23:
Yes i resetted the lock to factory without adding to the app