technyon/nuki_hub #327: Ring suppression

fir3drag0n, 2024-03-15:
Is there an option to enable the Ring suppression as it is described here for nuki ng (https://github.com/kvj/hass_nuki_ng/issues/76)?

Best wishes

chrisgrou, 2024-03-17:
I think this is the only important feature missing, otherwise it's perfect!

iranl, 2024-03-17:
It is not part of Nuki Hub at this time.

But since the nuki_ble library that Nuki Hub uses already supports the functionality you're asking for it should not be that hard to include in Nuki Hub.

I'll add it to the list of features I'm working on.

iranl, 2024-04-06:
Getting Ring suppression state and changing is part of #332. An update to the README and Home Assistant (select) switch still needs to be made but you can already beta test the current version using the binary from https://github.com/technyon/nuki_hub/actions/runs/8579693856/artifacts/1390528720.

You should see 4 new MQTT topics after installing the binary. Current state of `doorbellSuppression` will be under `configuration/advancedJson`. Current state should be a number between 0 and 7 based on having ring, rto and/or cm suppression enabled.

You can change the setting by sending a JSON value of `{ "doorbellSuppression": "[0-7]" }` e.g. `{ "doorbellSuppression": "5" }`   to `configuration/action`.

Result of the action will be in `configuration/commandResult`

chrisgrou, 2024-04-14:
Hello and thank you for your time.
First of all, here are the states:
0: All off
1: CM
2: RTO
3: RTO + CM
4: Ring
5: Ring + CM
6: Ring + RTO
7: All on

Assuming I'm doing this correctly (via MQTT Explorer)...
[image]

How long should it take for the configuration to update after sending the JSON data? Does it have to do with the query interval? At first I had to restart the hub in order to see the new value, now it won't update at all. It also won't reflect the app settings. I suspect I'm doing something wrong, so I'll wait for a stable release. I'd be happy to help though if you need anything

iranl, 2024-04-14:
The binary posted here was a very early one and development since has been significant. We are currently testing a pre-release of 8.34 using the binary [nuki_hub-8.34-pre-2.zip](https://github.com/technyon/nuki_hub/files/14958890/nuki_hub-8.34-pre-2.zip)

As I don't own an Opener testing of features only available in the Opener is greatly appreciated.

Steps:
-Install the above binary
-Check "Access Level Configuration" in the Nuki Hub Web configuration and make sure any setting you want to be able to change is enabled and save the config. Enable at least doorbellSuppression.
-If you are using Home Assistant within a minute or so there should be a new disabled/hidden select switch for doorbellSuppression. Enable the entity. Wait a minute.
-Using MQTT Explorer check the new values for doorbellSuppression. This has changed from numbers to text representing the state of doorbellSuppression.
-Send a new value using MQTT explorer to `nukiopener/configuration/action` e.g. `{ "doorbellSuppression": "CM & Ring" }`
-Or use the Home Assistant select entity to change the state of the doorbell
-Check `nukiopener/configuration/commandResult` in MQTT explorer for the result of the command. You should expect a result in about 5 seconds.

chrisgrou, 2024-04-14:
I installed it and enabled everything under access level.

Web UI:
-The hub is now very very unstable, it crashes very frequently, sometimes right after a reboot. It actually won't reboot by itself, I have to power cycle it manually.
-Restart on disconnect doesn't seem to make a difference.
-The Opener appears as paired but with undefined status. I did remove and re-pair it, no difference.

Home Assistant/MQTT:
-No new entities for the Opener in HA. For the Lock I did get everything. I enabled under access level
-Changing a setting in the app is reflected almost instantly in HA and MQTT. The opposite causes a crash.
-I tried the MQTT command once and I got a { "general": "invalidconfig" } response.

I don't know if my actions are triggering the reboots or it's something in the background.
Reboot reason is usually: ESP: ESP_RST_PANIC: Software reset due to exception/panic.
Bear in mind that I eventually installed a shelly plug so I could reboot it from afar, I don't know if that would mess with the logs or anything.
Do you need me to provide any logs?

iranl, 2024-04-14:
Can you post the output of your /info page?
Especially interested in what the watermarks and heap numbers are.

And is the nukiopener/configuration/basicJson and nukiopener/configuration/advancedJson topic being updated at all after a reboot of the ESP?

Seems something is going wrong with getting the config for the opener. Could be a bug or resource issue (combining lock and opener might overwhelm the ESP). This likely also triggers the frequent reboots/instability.

Not having a valid config will also block changing the config through MQTT and lead to the 'invalidconfig' state.

chrisgrou, 2024-04-14:
I'm on my phone currently but yes, the topics seem to get updated after a reboot (they're flashing on MQTT Explorer).
I have an ESP-WROOM-32D, it was very stable before. I got the Lock 4, Keypad 2 and the Opener.
Here are the logs:
```
System Information
Nuki Hub version: 8.34-pre-2
run: true
confVersion: 834
deviceId: 3288685188
deviceIdOp: 1196936286
nukiId: ***
nukidOp: ***
mqttbroker: 10.0.0.27
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: true
lockena: true
mqttpath: nuki
openerena: true
openercont: false
mqttoppath: nukiopener
maxkpad: 3
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
hassConfigUrl: 
dhcpena: false
ipaddr: 10.10.20.65
ipsub: 255.255.255.0
ipgtw: 10.10.20.1
dnssrv: 10.10.20.1
nwhw: 1
nwwififb: false
rssipb: 60
hostname: nukihub
nettmout: -1
restdisc: false
rstbcn: -1
lockStInterval: 1800
configInterval: 7200
batInterval: 7200
kpInterval: 1800
kpCntrlEnabled: true
kpInfoEnabled: true
aclLckOpn: 
confLckBasAcl: 
confLckAdvAcl: 
confOpnBasAcl: 
confOpnAdvAcl: 
accLvl: 0
regAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: 
crdpass: 
pubAuth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
latest: 8.33
MQTT connected: Yes
Lock firmware version: 4.1.8
Lock hardware version: 8.3
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: No
Lock has keypad: Yes
Lock ACL (Lock): Allowed
Lock ACL (Unlock): Allowed
Lock ACL (Unlatch): Allowed
Lock ACL (Lock N Go): Allowed
Lock ACL (Lock N Go Unlatch): Allowed
Lock ACL (Full Lock): Allowed
Lock ACL (Fob Action 1): Allowed
Lock ACL (Fob Action 2): Allowed
Lock ACL (Fob Action 3): Allowed
Lock config ACL (Name): Allowed
Lock config ACL (Latitude): Allowed
Lock config ACL (Longitude): Allowed
Lock config ACL (Auto Unlatch): Allowed
Lock config ACL (Pairing enabled): Allowed
Lock config ACL (Button enabled): Allowed
Lock config ACL (LED flash enabled): Allowed
Lock config ACL (LED brightness): Allowed
Lock config ACL (Timezone offset): Allowed
Lock config ACL (DST mode): Allowed
Lock config ACL (Fob Action 1): Allowed
Lock config ACL (Fob Action 2): Allowed
Lock config ACL (Fob Action 3): Allowed
Lock config ACL (Single Lock): Allowed
Lock config ACL (Advertising Mode): Allowed
Lock config ACL (Timezone ID): Allowed
Lock config ACL (Unlocked Position Offset Degrees): Allowed
Lock config ACL (Locked Position Offset Degrees): Allowed
Lock config ACL (Single Locked Position Offset Degrees): Allowed
Lock config ACL (Unlocked To Locked Transition Offset Degrees): Allowed
Lock config ACL (Lock n Go timeout): Allowed
Lock config ACL (Single button press action): Allowed
Lock config ACL (Double button press action): Allowed
Lock config ACL (Detached cylinder): Allowed
Lock config ACL (Battery type): Allowed
Lock config ACL (Automatic battery type detection): Allowed
Lock config ACL (Unlatch duration): Allowed
Lock config ACL (Auto lock timeout): Allowed
Lock config ACL (Auto unlock disabled): Allowed
Lock config ACL (Nightmode enabled): Allowed
Lock config ACL (Nightmode start time): Allowed
Lock config ACL (Nightmode end time): Allowed
Lock config ACL (Nightmode auto lock enabled): Allowed
Lock config ACL (Nightmode auto unlock disabled): Allowed
Lock config ACL (Nightmode immediate lock on start): Allowed
Lock config ACL (Auto lock enabled): Allowed
Lock config ACL (Immediate auto lock enabled): Allowed
Lock config ACL (Auto update enabled): Allowed
Opener firmware version: 
Opener hardware version: 
Opener paired: Yes
Opener PIN set: No
Opener has keypad: No
Opener ACL (Activate Ring-to-Open): Allowed
Opener ACL (Deactivate Ring-to-Open): Allowed
Opener ACL (Electric Strike Actuation): Allowed
Opener ACL (Activate Continuous Mode): Allowed
Opener ACL (Deactivate Continuous Mode): Allowed
Opener ACL (Fob Action 1): Allowed
Opener ACL (Fob Action 2): Allowed
Opener ACL (Fob Action 3): Allowed
Opener config ACL (Name): Allowed
Opener config ACL (Latitude): Allowed
Opener config ACL (Longitude): Allowed
Opener config ACL (Pairing enabled): Allowed
Opener config ACL (Button enabled): Allowed
Opener config ACL (LED flash enabled): Allowed
Opener config ACL (Timezone offset): Allowed
Opener config ACL (DST mode): Allowed
Opener config ACL (Fob Action 1): Allowed
Opener config ACL (Fob Action 2): Allowed
Opener config ACL (Fob Action 3): Allowed
Opener config ACL (Operating Mode): Allowed
Opener config ACL (Advertising Mode): Allowed
Opener config ACL (Timezone ID): Allowed
Opener config ACL (Intercom ID): Allowed
Opener config ACL (BUS mode Switch): Allowed
Opener config ACL (Short Circuit Duration): Allowed
Opener config ACL (Eletric Strike Delay): Allowed
Opener config ACL (Random Electric Strike Delay): Allowed
Opener config ACL (Electric Strike Duration): Allowed
Opener config ACL (Disable RTO after ring): Allowed
Opener config ACL (RTO timeout): Allowed
Opener config ACL (Doorbell suppression): Allowed
Opener config ACL (Doorbell suppression duration): Allowed
Opener config ACL (Sound Ring): Allowed
Opener config ACL (Sound Open): Allowed
Opener config ACL (Sound RTO): Allowed
Opener config ACL (Sound CM): Allowed
Opener config ACL (Sound confirmation): Allowed
Opener config ACL (Sound level): Allowed
Opener config ACL (Single button press action): Allowed
Opener config ACL (Double button press action): Allowed
Opener config ACL (Battery type): Allowed
Opener config ACL (Automatic battery type detection): Allowed
Network device: Built-in Wi-Fi
Uptime: 0 minutes
Heap: 72600
Stack watermarks: nw: 3052, nuki: 228, pd: 252
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.
```

iranl, 2024-04-14:
Info page shows that no Opener PIN is set, that would interfere with changing config. Although it should check for this. Are you sure PIN is set for the opener?

It also does not show Opener hardware and firmware version which seems odd.

chrisgrou, 2024-04-14:
Yes, I didn't have all these reboots with the previous version.
The info is changing by itself with almost every refresh of the page, for example these 3 screenshots are within 10":
[image]
[image]
[image]

iranl, 2024-04-15:
It looks like a resource issues when using both an opener and a lock in combination with all the new functionality for both.
I've implemented a couple of changes that should help.

Could you give 8.34-pre-3 a try?

[nuki_hub-8.34-pre-3.zip](https://github.com/technyon/nuki_hub/files/14982250/nuki_hub-8.34-pre-3.zip)

chrisgrou, 2024-04-16:
I'm away till Friday, I'll try it then and let you know

technyon, 2024-04-17:
It would sure be possible, but we want the next release to work equally well as the last. It would be great if you could do some testing for us.

chrisgrou, 2024-04-20:
The batteries of the Opener were dead when I came home and the ring detection still isn't working even after I recharged them. It's not the first time this is happening, last time it got magically fixed by itself after I had given up. I guess I'll have to talk to Nuki because it's getting frustrating.
So, although this isn't proper troubleshooting and I can't say for sure that the above issues don't affect the hub:
- The new firmware is more stable (I had a few crashes right after I flashed it but it's been online for 25 hours straight)
- The new entities appear correctly in HA
- Changing options like doorbell suppression mode/volume from the app appear on MQTT with a big delay (like something else has to happen for it to update)
- Changing options doorbell suppression mode/volume from HA aren't enforced and mirrored in the app

iranl, 2024-04-21:
Thanks for taking the time to do some more testing.

- It is expected that changes in the app appear in MQTT with a big delay, there is no signaling by the Nuki devices to the hub that anything has changed in the config. Nuki Hub checks the config on a schedule (every 2 hours based on your Nuki Hub config). You can update the config in MQTT on demand using the (hidden) query config entity in Home Assistant or by publishing a "1" to nukiopener/lock/query/config topic using something like MQTT explorer.

- The Nuki app only refreshes settings when you exit the app almost completely (back to home screen) and then open the config menu again.

- What is the data in the MQTT topic `nukiopener/configuration/commandResult` after changing any setting using Home Assistant? (Check using MQTT explorer)

chrisgrou, 2024-04-22:
No problem really! I'm again away for the week, I'll get back to you on Friday. Btw I'm fine if you'd prefer we use PMs instead of overloading this thread (like HA forum, or whatever)

iranl, 2024-04-22:
If you prefer PMs you can join the Nuki Hub on Discord (https://discord.gg/feB9FnMY) and contact me there.

chrisgrou, 2024-05-08:
(still having problems with my Opener)
Can't join the server, says the invite is invalid

iranl, 2024-05-08:
Try https://discord.gg/24HxpGBJ

PR #332 (2024-05-10): Enhance config options
- [x] Add every config and advanced config option to MQTT (JSON)
- [x] Separate config state from config actions in MQTT (same as lockstate/action and keypad codes/action)
- [x] ACL for every separate config option
- [x] Make some config/advanced config options more human readable
- [x] Allow modifying every config and advanced config option through MQTT
- [x] Modify existing config Home Assistant auto discovery topics
- [x] Add new config Home Assistant auto discovery topics
- [x] Update README
- [x] Enable/disable all button for ACL options in WebCfg
- [x] Final minor  [... cut]
closed 2024-05-10 by technyon