technyon/nuki_hub #329: MQTT Entities in HA remain when lock not connected

DaveTiff, 2024-03-19:
Hi, juts got my Nuki 4 Pro yesterday and just found your code and installed on an ESP32.

I took the battery out overnight to fully charge it, but before inserting the battery this morning I looked at HA and you would not know that the lock was unpowered. If you log onto the Hub the only clue is "Nuki Lock State = undefined"

Can this state be reflected in HA so one knows there might be a problem?

iranl, 2024-03-19:
The lock state `undefined` is the way this is reflected. You can create an automation to signal you when this MQTT state is detected.

DaveTiff, 2024-03-20:
Hi, thanks, but the "undefined" state shows up on the Hub webpage but not in HA, still says Locked or Unlocked (Last state)

iranl, 2024-03-25:
That behaviour is expected. HA does not have a "undefined" state for locks.

You can research how to use MQTT topics as triggers or conditions for automations in Home Assistant via the HA community (e.g. https://community.home-assistant.io/t/automation-based-on-mqtt-topic-solved/278674)

iranl, 2024-05-17:
Closing as completed with answer provided.

closed 2024-05-17 by iranl