technyon/nuki_hub #330: Indication When Lock Fails (Red Flashing Light on lock ring)

DaveTiff, 2024-03-19:
Hi, I have a lock that requires you to lift the handle before the NUKI can operate.

If you don't lift the handle then the NUKI stalls and flashes the top half of the light ring red for a few seconds

Is this error sent back to HA, if not can it be sent as an error state.

Regards Dave

technyon, 2024-03-20:
I think the lock state in MQTT (not HA) should change to motor blocked. Probably the auto discovery can't pick it up.

iranl, 2024-03-20:
Motor blocked state is being used by HA since 8.33, this was part of recent auto discovery changes. HA should show "jammed" if this state is signaled by Nuki. But not sure if Nuki signals motor blocked in this case. Nuki documentation seems to suggest it mainly logs this to the authorization log. 

Which you can enable in Nuki Hub / HA, see the Nuki Hub README for how publishing of the authorization log.

technyon, 2024-03-21:
@DaveTiff I think it would be useful to know what is published in the lock/state topic. Can you use an MQTT client like MQTT explorer and monitor this topic please.

DaveTiff, 2024-03-22:
Hi, thanks, have done that. But not its not that straight forward.

Senario 1: If you use the app to lock the door with the handle NOT lifted then the Topic nuki/lock/completionStatus = undefined 
and you get an error on the app asking you to lift the handle and try again. the Topic state = locked, also the LED in the Knob flash Red for a few seconds

Senario 2: If you press the button to use the "Lock & Go" function, which is what I do, with the handle NOT lifted then the then the Topic nuki/lock/completionStatus = success, the Topic state = unlockedLnga, no flashing Red LED

Senario 3: If you use the Hub via HA to lock the door with the handle NOT lifted then the Topic nuki/lock/completionStatus = success, the Topic state = locked, no flashing Red LED

Senario 4: If you use the Hub via HA to Lock And Go with the door with the handle NOT lifted then the Topic nuki/lock/completionStatus = success, the Topic state = unlocked, no flashing Red LED but Topic nuki/lock/query/lockstateCommandResult = failed

Hub Lock And Go responds differently to Lock Button Lock & Go

I will be contacting NUKI about this, also there is another slight problem with there lock and that is the Locked & Unlocked state seems to be based on the rotation position of the knob. This is a problem, because if you unlock the door by hand (which I do) you can hear the clunk as the cylinder is turned, in your mind yuou know the lock is unlocked but the Status (on the App & the Hub) still says "locked" and the LED on the know remains OFF, you have to physically rotate the know more to get the status to change to "Unlocked".

I have already contacted them about this.

Regards, Dave (MQTT Results below)
Please let me know if there is anything else you need.

Nuki Unlock With App (Normal)

nuki
maintenance
lock
action = ack
query
battery = 0
config = 0
lockstate = 0
**lockstateCommandResult = success**
state = unlocked
hastate = unlocked
binaryState = unlocked
trigger = system
lastLockAction = Unlock
**completionStatus = success**
doorSensorState = unavailable
json = {"lock_state":"unlocked","trigger":"system","last_lock_action":"Unlock","lock_completion_status":"success","door_sensor_state":"unavailable","auth_id":0,"auth_name":""}

Nuki Locked with App (Normal Handle Lifted)

nuki
maintenance
lock
action = ack
query
battery = 0
config = 0
lockstate = 0
**lockstateCommandResult = success**
state = locked
hastate = locked
binaryState = locked
trigger = system
lastLockAction = Lock
**completionStatus = success**
doorSensorState = unavailable
json = {"lock_state":"locked","trigger":"system","last_lock_action":"Lock","lock_completion_status":"success","door_sensor_state":"unavailable","auth_id":0,"auth_name":""}

Nuki Locked with App (ERROR Handle NOT Lifted Door did NOT Lock)

nuki
maintenance
lock
action = ack
query
battery = 0
config = 0
lockstate = 0
**lockstateCommandResult = success**
state = locked
hastate = locked
binaryState = locked
trigger = system
lastLockAction = Lock
**completionStatus = undefined**
doorSensorState = unavailable
json = {"lock_state":"locked","trigger":"system","last_lock_action":"Lock","lock_completion_status":"undefined","door_sensor_state":"unavailable","auth_id":0,"auth_name":""}

Nuki Locked with HA (ERROR Handle NOT Lifted Door did NOT Lock)

nuki
maintenance
lock
action = ack
query
battery = 0
config = 0
lockstate = 0
**lockstateCommandResult = success**
state = locked
hastate = locked
binaryState = locked
trigger = system
lastLockAction = Lock
**completionStatus = success**
doorSensorState = unavailable
json = {"lock_state":"locked","trigger":"system","last_lock_action":"Lock","lock_completion_status":"success","door_sensor_state":"unavailable","auth_id":0,"auth_name":""}

Nuki Locked with Lock And Go Via HA (ERROR Handle NOT Lifted Door did NOT Lock)

nuki
maintenance
lock
action = ack
query
battery = 0
config = 0
lockstate = 0
**lockstateCommandResult = failed**
state = unlocked
hastate = unlocked
binaryState = unlocked
trigger = system
lastLockAction = Unlock
**completionStatus = success**
doorSensorState = unavailable
json = {"lock_state":"unlocked","trigger":"system","last_lock_action":"Unlock","lock_completion_status":"success","door_sensor_state":"unavailable","auth_id":0,"auth_name":""}
rssi = -90

Nuki Locked with Lock And Go Via Button on Lock (ERROR Handle NOT Lifted Door did NOT Lock)

nuki
maintenance
lock
action = ack
query
battery = 0
config = 0
lockstate = 0
**lockstateCommandResult = success**
state = unlockedLnga
hastate = unlocked
binaryState = unlocked
trigger = system
lastLockAction = LockNgo
**completionStatus = success**
doorSensorState = unavailable
json = {"lock_state":"unlockedLnga","trigger":"system","last_lock_action":"LockNgo","lock_completion_status":"success","door_sensor_state":"unavailable","auth_id":0,"auth_name":""}

technyon, 2024-03-22:
For scenario 1, do you get completion status undefined consistently? This status represents the failure or success of communicating with the lock via bluetooth. Undefined means nuki was not successful retrieving the status, I don't think it's related to how you operate the lock.

DaveTiff, 2024-03-25:
Yes Nuki Locked with App (ERROR Handle NOT Lifted Door did NOT Lock)
nuki/lock/completionStatus = undefined

technyon, 2024-03-26:
Sorry I confused the completionStatus with the commandResult. The completionStatus is returned as a number and converted to a string in the code. Whatever number your lock returns, Nuki Hub doesn't know it. Can you flash this binary, it publishes the actual number as the completion result. Retry scenario 1 and post the number please.

[nuki_hub-8.34-dbg-1.zip](https://github.com/technyon/nuki_hub/files/14759055/nuki_hub-8.34-dbg-1.zip)

DaveTiff, 2024-03-28:
Scenario 1 with 8.34-dbg-1

nuki
maintenance
lock
action = --
query
battery = 0
config = 0
lockstate = 0
lockstateCommandResult = success
keypad = 0
state = locked
trigger = system
lastLockAction = Lock
completionStatus = 11
doorSensorState = unavailable
json = {"lock_state":"locked","trigger":"system","last_lock_action":"Lock","lock_completion_status":"11","door_sensor_state":"unavailable","auth_id":0,"auth_name":"������P"}
log = --
authorizationId = 0
authorizationName = --
rssi = -78
address = 54:d2:72:27:90:64
info
nukiHubIp = 192.168.1.47
nukiHubVersion = 8.34-dbg-1
firmwareVersion = 4.1.8
hardwareVersion = 7.0

DaveTiff, 2024-03-29:
Ok, thanks, will that be reported back to show in HA?

technyon, 2024-03-29:
I have no idea :D. Someone else has to answer that.

iranl, 2024-05-17:
Closing as completed with fix and answer provided

closed 2024-05-17 by iranl