technyon/nuki_hub #331: Lock wont repair after unpair

aurelmarius, 2024-03-25:
I have a nuki lock 3 pro and because of poor battery life i want to try nuki hub. Installed latest version of nuki hub on an olimex poe iso board and paired to the lock. First pair went smooth, but in the morning i saw that nuki hub was reporting state unlocked, but nuki was locked. I tried to restart the esp board but no change. Then, I tried to re-pair the lock, but not working anymore.

technyon, 2024-03-25:
Please factory reset your ESP and reflash NUKI Hub.

When paired again, please check if the smartphone app reports the correct status.

aurelmarius, 2024-03-25:
I did factory reset, reflashed nuki hub, nothing works.
[image]

aurelmarius, 2024-03-25:
Strange, after changing hub's hostname got paired.

technyon, 2024-03-26:
is it reporting the state now correctly?

aurelmarius, 2024-03-26:
Yes, untill now is reporting correctly and connection seems stable.[image]

aurelmarius, 2024-04-19:
I will close this issue now, everything seems to work as expected. Thanks.

closed 2024-04-19 by aurelmarius

technyon, 2024-04-19:
ok, perfect