technyon/nuki_hub #334: "Query interval lock state" not taken into account - Battery drain issue

sbenmansour, 2024-04-03:
Since I started using nuki_hub, I noticed a significantly higher battery usage on my Nuki Lock 3.
A battery health check on the official Android app pointed out high bluetooth traffic as the culprit.

I noticed the publish frequency on nuki/lock/json and nuki/lock/query/lockStateCommandResult was always about one message every 4 or 5 seconds, and that the value of the "Query interval lock state" parameter did not have any effect on it.
This obviously generated unnecessary bluetooth traffic that drains the battery.

Some other details about my config : 
I'm using an ethernet connection with Wiznet W5500
I enabled Smartlock management but not Opener.
I checked "Nuki Bridge is running alongside Nuki Hub"

technyon, 2024-04-04:
Are you actually using a Nuki Bridge together with Nuki Hub. If you enable "Nuki Bridge is running alongside Nuki Hub" without using a Bridge, this is expected bahavior. In that case you should disable the option, unpair and re-pair Nuki Hub.

sbenmansour, 2024-04-04:
I initially checked that option because I was using Nubli, another bridge replacement option written on nodejs I ran on a raspberry pi.
And it sure makes sense that in the case where many bridge likes are paired with a single lock, that they all have to maintain its state in real time by continuously polling it.

But now I followed your advice, unchecked the option and re-paired the lock, but there are still MQTT state publishes on the topics I mentioned earlier every ~4s.

closed 2024-04-04 by sbenmansour

technyon, 2024-04-05:
I can indeed not, because the relevant code is in the NUKI firmware which is closed source. If you enable this option, Nuki Hub will register as an app, as opposed to registering as a bridge. When a bridge queries the lock state, the lock will stop notifiying in beacons that the state has changed. When registered as an app, the lock will continue notifying that the state has changed, triggering Nuki Hub to continuously querying the state.

sbenmansour, 2024-04-05:
I see, I understand now how NukiWrapper::notify(eventType) gets called.

Thanks again.