technyon/nuki_hub #335: Reading GPIO 14 pull-up resistor issue?

pkkpp, 2024-04-10:
Hello,

I want to trigger the Nuki Lock to open when electric door opener  Door open is pressed on a Biticino 100x Doorbell system.
I have connectet the Output of the Biticino Relais to a ESP32

C-> GND
NO -> GPIO 14

When "Door Open" is pressed on the Biticino the Relais latches and Nuki lock opens.
Good so far.

But the GPIO 14 Input is wery sensitive - instable...
When I switch on or off an electrical device nearby (Light or somthing else) the Relais does not latch but  Nuki hub sends the command to unlock the nuki lock. 

what can I do?
I have heard that GPIO 14 has internal pull up/down resistors... how ar they configured?
Do I have to use external pull-up resistor?
somthinge else?

Thank You

technyon, 2024-04-10:
Well, it's a microcontroller, they are sensitive by default. The pin is configured as pull-up, no need for an external one. Your problem is probably that the leads from the relay to the ESP act as an antenna, and they pick up those external signals. Keep the wires between the ESP and the relay as short as possible. If that alone doesn't help, you can try shielding against interferences by housing the ESP and the relay inside a grounded metal cage.

iranl, 2024-05-17:
Closing as this is not an issue with Nuki Hub but ESPs in general

closed 2024-05-17 by iranl