technyon/nuki_hub #337: Feature request / Question: Homekey support

chbmuc, 2024-04-17:
First of all, I really like this project. The code is very well structured and clean.
 
What I would like to see is support for Apple Homekey. There is already a project for a homekey reader with the esp32 (https://github.com/rednblkx/HomeKey-ESP32), but it requires HomeSpan (https://github.com/HomeSpan/HomeSpan).
 
I have looked at the code of these projects. HomeSpan is an entire framework for building Homekit appliances, not just a simple library. This makes it difficult to integrate with nuki_hub. It would probably be easier to integrate the Nuki library into the HomeKey-esp32 project, but then I would lose the other benefits of nuki_hub...
 
As I'm not sure where to start, I wanted to ask if you have any ideas on how to get a Nuki hub with homekey support?

iranl, 2024-04-17:
I don't see a reason why such a project would need to be part of Nuki hub or the other way around.

It adds unnecessary (and because of resource limitations probably impossible) complexity for a fringe use case.

I would suggest using the project you mentioned and use nodered or homeassistant to let Nuki hub react on input from the homekey reader.

technyon, 2024-04-18:
The idea behind Nuki Hub is not to implement manufacturer specific protocols, but to expose features in an open protocol that allows it to connect it to your smarthome middleware. The middleware can then connect the different systems so they can interact with each other ... you'd register Nuki Hub and homekit with your smarthome software (home assistant, iobroker, OpenHab, ...), and that software would act as a hub between those systems.

If you want something specific to homekit, I think it would be a lot easier to start with the Nuki library (https://github.com/I-Connect/NukiBleEsp32), and then copy over specific code from Huki Hub to extend the functionality.

chbmuc, 2024-04-18:
Thanks for your opinion. Let me see what I can do on my own...

closed 2024-04-18 by chbmuc

x23piracy, 2024-06-20:
You xmcan directly use the home key project to send to a specified mqtt topic so nothing in between is need because the home key project directly integrates the lock into homekit.

the problem is the current nuki fw 3.9.5 which is really worse with wlan and mqtt and even the homekit integration from nuki is not really fast.