technyon/nuki_hub #346: Nuki "Lock immediately after closing" feature usable with home assistant door sensor?

TheFelix93, 2024-04-30:
Hello, I plan to use a zigbee door sensor that is integrated into HA.
Is it possible to directly configure the sensor to trigger the lock to "Lock immediately after closing" like if the original Nuki door sensor is used? I don't know how the original sensor and lock communicate, but I would expect that the nuki hub could use the input of a HA doors sensor to do the same.
 Or do I have to create an automation to achieve this?

Additional questions: Do you know if the original nuki door sensor works plug and play if I use nuki hub instead of original bridge and app?

Thank you in advance.

iranl, 2024-04-30:
You need to create an automation for this.

The Nuki Door Sensor communicates with the Nuki Lock independent from any other device including the original bridge/app or Nuki Hub.

TheFelix93, 2024-04-30:
Thank you for the quick reply.
let's assume I use the original nuki sensor:
The nuki internal feature  "Lock immediately after closing" can be only enabled via App. I assume that with my Nuki 3 (non-pro) I can use the app via BLE in parallel to the nuki hub, right?

closed 2024-04-30 by TheFelix93