technyon/nuki_hub #347: old BLE connections?

jolichter, 2024-05-01:
Question: I am having trouble disconnecting the BLE connection between my ESP32-S3 and the NUKI SL3 Pro, which I had used only for testing purposes. These connections are not displayed in the Nuki app, and therefore I cannot remove them. Could this be a problem?

Thanks!

iranl, 2024-05-02:
This should not be a problem.

If you erase the ESP it will no longer be able to connect to the lock.

Pairing a new bridge or ESP to the lock will also overwrite the old pairing in the lock as only one bridge like device can be  registered.

There is no battery penalty on the lock for having a (non-functioning/offline/reset) hub registered.

jolichter, 2024-05-02:
Thank you, good to know: Pairing a new bridge or ESP to the lock will also overwrite the old pairing in the lock as only one bridge like device can be registered.

That answers my question.

closed 2024-05-02 by jolichter