technyon/nuki_hub #348: AP Selection with Nuki Hub?

jolichter, 2024-05-02:
Background: I frequently encounter the issue of an ESP32 attempting to connect to a distant access point with a weak signal, despite the presence of one or two nearby access points. While I utilize mesh networking with AVM (Fritz!Box) devices, ESP32s, unfortunately, lack support for mesh (seamless roaming).

Therefore, I'm curious: Does the Nuki Hub offer a feature like scanning to automatically choose the nearest or strongest access point when multiple access points share the same SSID?

see e.g.: [github.com/espressif/arduino-esp32](https://github.com/espressif/arduino-esp32/commit/3253de879235e35b48be3b9cb656680837028687)

iranl, 2024-05-02:
It should already connect to the strongest AP with the configured SSID. This is also how my own setup works.

When the ESP loses connection your settings in Nuki Hub determine how the ESP reacts. Because Wifi reconnection by the ESP can be unreliable I have Nuki Hub configured to reboot on connection loss. You could play with this setting to see if it helps.

It sounds like your AVM system might be a bit agressive in kicking clients with a possibly better connection to another AP off an AP, hoping it will connect to the AP with a stronger connection. The ESP should do this, but maybe it is then kicked off by the other APs as well if the rssi/db still does not meet a certain threshold. An option is to move the ESP closer to a specific AP.

Does the AVM have functionality to create a second SSID on only certain AP's on 2.4ghz that operates outside of the mesh? You could then connect the ESP to that network

jolichter, 2024-05-02:
Thank you for your information! With my ESP8266 (WLED) I have no problems in the home network, but with ESP32 I have to adjust the configuration in the Nuki Hub so that they restart automatically if the connection is lost.
Unfortunately, AVM does not currently offer the option of creating a second SSID only on certain 2.4 GHz access points outside the mesh network.

That's really strange: My M5Stack-StampS3 shows the best Wi-Fi signal strength according to the router. The ESP32-S3 sometimes connects to the furthest and weakest signal after a restart, even though the router and another access point are much closer.

Due to the fact that the ESP32 does not connect to the strongest AP, the problem probably is caused by AVM.

As a workaround, I've now housed the M5Stack-StampS3 in a case to dampen the signal. Pretty crazy, right? On the other hand, I've optimized the antennas of my ESP8266.

jolichter, 2024-05-02:
According to AVM ([German Link](https://avm.de/service/wissensdatenbank/dok/FRITZ-Box-7590/3593_WLAN-Gerat-verbindet-sich-nicht-mit-optimalem-Mesh-Repeater/)), if WiFi devices do not support WiFi Mesh Steering, the Mesh Master cannot control them. To specifically connect these devices to a particular Mesh Repeater, you need to disable settings adoption in the Mesh Repeater and set up a separate wireless network name for it.

That could be an option, but I consider it suboptimal because it renders this repeater unusable for mesh devices.

closed 2024-05-02 by jolichter

technyon, 2024-05-03:
To give a bit background to this: We're using the Wifi stack provided by Espressif for the ESP32. It's more or less a black box, you give a few parameters like ssid and PSK, the rest it handles on its own. It is unfortunately a bit flaky, and we unfortunately have limited control over that.

What I can always recommend: Get an ESP with a LAN connection, it's much more stable than Wifi.

mundschenk-at, 2024-05-03:
Could you optionally pass a BSSID to the WiFi stack?

iranl, 2024-05-03:
See PR #349

technyon, 2024-05-04:
What about just enable this option, no need to put it into the UI. I don't see a reason why you wouldn't connect to the "best" AP ...  meaning the one with the strongest signal I guess. The Wifimanager code has some custom modifications made by me, for example the wifi configuration page is locked by username/password if credentials are configured. This needs to be applied again in the updated wifi manager code.

jolichter, 2024-05-04:
Hi, just a thought... if a user only has one AP, the connection setup would be faster since the search for other APs can be skipped. Therefore, I find this option beneficial unless the search for the best AP is not time-critical. Since many users (myself included) frequently experience connection drops with ESP32 modules and a restart is then necessary, I consider a fast boot process important. Unfortunately, I don't have a LAN connection within the BLE range of the Nuki.

iranl, 2024-05-04:
> The Wifimanager code has some custom modifications made by me, for example the wifi configuration page is locked by username/password if credentials are configured. This needs to be applied again in the updated wifi manager code.

I already ported your customizations to 2.0.17.

> What about just enable this option, no need to put it into the UI. I don't see a reason why you wouldn't connect to the "best" AP ... meaning the one with the strongest signal I guess. 

Agree with @jolichter on making this an optional feature because of the added complexity and slowing of the connection (it seems to be about 2 seconds extra) which is not necessary for all users.

PR #349 (2024-05-22): Update WiFiManager to 2.0.17 and add option to connect to strongest BSSID
Add a solution for multi AP/Mesh environments where Nuki Hub connects to a random AP instead of the AP with the strongest signal and might be repeatedly kicked of the network by Mesh systems employing access point steering. 

Fixes #348

Enable `Connect to AP with the best signal in an environment with multiple APs with the same SSID` in `Advanced MQTT and Network Configuration` to enable this new behaviour.

Binaries for testing:
~~[esp32dev-assets.zip]()~~
~~[esp32-s3-assets.zip]()~~
~~[esp32-c3-assets.zip]()~~
~~[esp3 [... cut]