technyon/nuki_hub #351: OTA for 8.34 not possible

fir3drag0n, 2024-05-10:
just the erros appears:

`Over-the-air update errored. Please check the logs for more info

Choose the updated nuki_hub.bin file to upload:`

iranl, 2024-05-10:
The binary in the ESP32 release file does not seem to be the proper binary. Please try the binary (nuki_hub_esp32.bin) from the 8.34 commit: https://github.com/technyon/nuki_hub/actions/runs/9031574144/artifacts/1490961864

@technyon: Is there a reason not to use the packages directly from GitHub actions?

I'm working on a PR to automate binary creation and uploading for releases.

fir3drag0n, 2024-05-10:
thank  you! btw your discord link does not work anymore

iranl, 2024-05-10:
Did you try https://discord.gg/24HxpGBJ ?

capandcamera, 2024-05-10:
Binary being used by OTA does not have the .bin extension, 
8.34 was uploaded as nuki_hub. so the OTA downloads 8.33 again.

nuki_hub. | version bump to 8.34 | 2 hours ago
nuki_hub.bin | version bump to 8.33 | 2 months ago

iranl, 2024-05-11:
Please try the binary that was linked in this issue or the new binary that was attached to the release

noohi, 2024-05-12:
Thanks iranl.

OK, a bit embarrassing, and easily explainable, the POE device once flashed was not LAN by default, it was waiting for the WIFI setup first.
So long has passed since it worked upgrading itself, that I forgot the basics. One would assume it prioritizes LAN if available, but that's not important now.

So the last OTA update erased the network settings, that explains all on my side.

Anyway, for others in the future, if you have a POE/LAN ESP32 device like Olimex, note that nukihub is WIFI out of the box regardless.
After flashing connect to its AP, setup WIFI, change to LAN from Nukihub settings.

Btw thanks for the maintenance of the project!

noohi, 2024-05-12:
Btw while at it, where is the network configuration to switchto v8.33?

iranl, 2024-05-12:
Should be in the same spot as always, under `Network hardware` about halfway down under `Advanced MQTT and Network Configuration`

noohi, 2024-05-12:
Thank you, all good now. (saw MQTT and eyes swiped over).
From today I officially lost the title of a power user :)

Suggestions from this:
- split MQTT and Network settings pages
- if possible to detect LAN presence, use it by default on first boot (if not set to anything specifically)

technyon, 2024-05-12:
Hi,

Nuki Hub doesn't know what kind of ESP it's running on, and thus has no knowledge what kind of networking hardware is connected. All those boards that come with LAN have an external chip one way or another, so you first have to tell Nuki Hub what kind of chip is connected unfortunately.

iranl, 2024-05-17:
Seems solved for all involved.

Starting from 8.35 release pipeline should be fully automatic so release binaries will be automatically added at all relevant locations at the same time (release, ota and webflash)

closed 2024-05-17 by iranl