technyon/nuki_hub #352: Feature Proposal: Auth data in unlock state

alexlenk, 2024-05-10:
Hello,

I'm using the json auth data to execute additional actions with a HASS automation when the unlock was successful. I find this a very useful feature, here is my config in case you want to integrate it in the code base.

**Setup:**
homeassistant/sensor/xxxxxxxx/lockstate/config:
```
{
    "dev": {
        "ids": [
            "nuki_xxxxxxxx"
        ],
        "mf": "Nuki",
        "mdl": "Opener",
        "name": "Opener"
    },
    "~": "nukiopener",
    "name": "Lock State",
    "unique_id": "xxxxxxxx_lockstate",
    "stat_t": "~/lock/json",
    "value_template": "{{ value_json.lock_state }}",
    "json_attributes_topic": "~/lock/json",
    "ent_cat": "diagnostic",
    "avty": {
        "t": "nuki/maintenance/mqttConnectionState"
    },
    "enabled_by_default": true
}
```

**Automation**

```
alias: "Access: Nuki Keypad Advanced Action"
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.opener_lock_state
    to: open
condition:
  - condition: template
    value_template: "{{ state_attr(\"input_select.people_group_owners\", \"options\") | select('match', state_attr('sensor.opener_lock_state', 'auth_name') | replace(' (Keypad)', '')) | list |\_count > 0}}"
action:
  - device_id: xxxxxxxxxxxxxxxxxxxx
    domain: button
    entity_id: yyyyyyyyyyyyyyyy
    type: press
mode: single
```

iranl, 2024-05-11:
Thanks for sharing your automation.

I don't think the sensor is necessary though, this automation should work without a separate sensor:

```
alias: "Access: Nuki Keypad Advanced Action"
description: ""
trigger:
  - platform: mqtt
    topic: "nukiopener/lock/json"
    payload: "open"
    value_template: "{{ value_json.lock_state }}"
condition:
  - condition: template
    value_template: "{{ state_attr(\"input_select.people_group_owners\", \"options\") | select('match',  trigger.payload_json.auth_name) | replace(' (Keypad)', '')) | list |\_count > 0}}"
action:
  - device_id: xxxxxxxxxxxxxxxxxxxx
    domain: button
    entity_id: yyyyyyyyyyyyyyyy
    type: press
mode: single
```

As such I don't see a reason to change the HA auto discovery code of Nuki Hub.

alexlenk, 2024-05-11:
Can you share how? When using the lock unlock state I always ran into race conditions since other values updated at different times.

iranl, 2024-05-11:
Have you looked at my comment?

I shared the complete code based on your first comment but without a separate sensor using the exact same input (nukiopener/lock/json lock_state and auth_name).

closed 2024-05-17 by iranl