technyon/nuki_hub #356: Nuki Opener cannot be configured via HA

streetjumper16, 2024-05-11:
Hello, I have updated Nuki Hub to 8.34. 

Now I try to change the configs in Home Assistant but as soon as I change something he shows me on Unavailable and then everything is as before. 

Can anyone help me here?

iranl, 2024-05-11:
What settings are you trying to change?

What does your nuki hub info page show before and after trying to change a setting?

What does the MQTT topic `nukiopener/configuration/commandResult` show after trying to change a setting?

streetjumper16, 2024-05-14:
I would like to e.g. In addition, the opener reacts and also the lock from time to time no longer until I restart. 

 Where can I see this? I'm still quite new about the MQTT.

iranl, 2024-05-14:
The `noPinSet` message means you haven't entered the PIN for the Nuki device.

To change Nuki settings you need to enter your Nuki PIN/Security code in the credentials page of the web interface of Nuki Hub.

The info page can be found on the Nuki Hub web interface by clicking on the current firmware version

iranl, 2024-05-16:
Please update to the latest master ([esp32-assets.zip](https://github.com/technyon/nuki_hub/files/15339216/esp32-assets.zip)) and report if this solves your problem. If not please provide the requested information:

What settings are you trying to change?
What does your nuki hub info page show before and after trying to change a setting?
What does the MQTT topic nukiopener/configuration/commandResult show after trying to change a setting?

iranl, 2024-05-26:
Any update on this?

iranl, 2024-05-29:
Lets follow up in #378