technyon/nuki_hub #360: Another one with Battery discharge

spaggi, 2024-05-15:
Hi,

I've been having the issue with the battery being empty after around 2 weeks. Ive see the other threads regarding this issue however in my case the "Run as appication" flag was never set.

[image]

I just have updated to the latest version.

Here are my settings:

`Nuki Hub version: 8.34
run: true
confVersion: 834
deviceId: 1798759620
deviceIdOp: 1798759620
nukiId: ***
nukidOp: 
mqttbroker: 192.168.1.185
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: false
lockena: true
lockpin: 0
mqttpath: nuki
openerena: false
openerpin: 
openercont: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
hassConfigUrl: 
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 5
nwwififb: false
rssipb: 60
hostname: nukihub
nettmout: -1
restdisc: false
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpCntrlEnabled: true
kpInfoEnabled: false
aclLckOpn: 
tcCntrlEnabled: false
tcInfoEnabled: false
confLckBasAcl: 
confLckAdvAcl: 
confOpnBasAcl: 
confOpnAdvAcl: 
accLvl: 
regAsApp: false
nrRetry: 
rtryDelay: 100
crdusr: 
crdpass: 
pubAuth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
latest: 
MQTT connected: Yes
Lock firmware version: 2.15.3
Lock hardware version: 11.1
Lock paired: Yes
Lock PIN set: No
Lock has door sensor: Yes
Lock has keypad: No
Lock ACL (Lock): Allowed
Lock ACL (Unlock): Allowed
Lock ACL (Unlatch): Allowed
Lock ACL (Lock N Go): Allowed
Lock ACL (Lock N Go Unlatch): Allowed
Lock ACL (Full Lock): Allowed
Lock ACL (Fob Action 1): Allowed
Lock ACL (Fob Action 2): Allowed
Lock ACL (Fob Action 3): Allowed
Lock config ACL (Name): Disallowed
Lock config ACL (Latitude): Disallowed
Lock config ACL (Longitude): Disallowed
Lock config ACL (Auto Unlatch): Disallowed
Lock config ACL (Pairing enabled): Allowed
Lock config ACL (Button enabled): Allowed
Lock config ACL (LED flash enabled): Allowed
Lock config ACL (LED brightness): Disallowed
Lock config ACL (Timezone offset): Disallowed
Lock config ACL (DST mode): Disallowed
Lock config ACL (Fob Action 1): Disallowed
Lock config ACL (Fob Action 2): Disallowed
Lock config ACL (Fob Action 3): Disallowed
Lock config ACL (Single Lock): Allowed
Lock config ACL (Advertising Mode): Disallowed
Lock config ACL (Timezone ID): Disallowed
Lock config ACL (Unlocked Position Offset Degrees): Disallowed
Lock config ACL (Locked Position Offset Degrees): Disallowed
Lock config ACL (Single Locked Position Offset Degrees): Disallowed
Lock config ACL (Unlocked To Locked Transition Offset Degrees): Disallowed
Lock config ACL (Lock n Go timeout): Disallowed
Lock config ACL (Single button press action): Disallowed
Lock config ACL (Double button press action): Disallowed
Lock config ACL (Detached cylinder): Disallowed
Lock config ACL (Battery type): Disallowed
Lock config ACL (Automatic battery type detection): Disallowed
Lock config ACL (Unlatch duration): Disallowed
Lock config ACL (Auto lock timeout): Disallowed
Lock config ACL (Auto unlock disabled): Allowed
Lock config ACL (Nightmode enabled): Disallowed
Lock config ACL (Nightmode start time): Disallowed
Lock config ACL (Nightmode end time): Disallowed
Lock config ACL (Nightmode auto lock enabled): Disallowed
Lock config ACL (Nightmode auto unlock disabled): Disallowed
Lock config ACL (Nightmode immediate lock on start): Disallowed
Lock config ACL (Auto lock enabled): Allowed
Lock config ACL (Immediate auto lock enabled): Disallowed
Lock config ACL (Auto update enabled): Disallowed
Network device: WT32-ETH01
Uptime: 12 minutes
Heap: 102712
Stack watermarks: nw: 7872, nuki: 1864, pd: 228
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.`

What I can see in the MQTT Log that there are messages every 3 seconds on nuki/presence/devices. Is this normal?

Any help is much appreciated!

iranl, 2024-05-15:
In the supplied info `nrRetry:` is empty.
This translates to the `Number of retries if command failed` in `Advanced Nuki Configuration` page.
This might cause a loop?

Please check that `Number of retries if command failed` is set to a positive integer (3 is default)

spaggi, 2024-05-15:
Thanks a lot for answering! @technyon I enabled MQTT logging but I cannot see anything here for the last 5 minutes
[image]
 
 @iranl thanks a lot, I changed the setting. Will see if itmakes a difference!

spaggi, 2024-05-15:
@technyon have you seen my remark before:
 
 "What I can see in the MQTT Log that there are messages every 3 seconds on nuki/presence/devices."
 
 
[image]
 
  Is this normal?

iranl, 2024-05-15:
Yes. Presence detection updates these values every few seconds. Presence detection works independently from the lock/opener (its a function of Nuki Hub, not the Nuki device) and does not cause battery drain on the Nuki device.

spaggi, 2024-05-17:
Hi, after changing the setting the discharge of 2% a day seems still to be the case.

The lock is fully locked and unlocked 2 times a day and the latch is opened maybe 5 times a day. 

Do you have a suggestion how I can diagnose this further?

technyon, 2024-05-18:
@spaggi If you don't see continuous messages coming in, that's how it should be. If there's constant querying of the lock (which drains the battery), each query will generate log messages. Keep observing the messages though, and best case keep a history of messages for a day or so.

Are you sure this is related to Nuki Hub? You could easily check by unplugging your ESP for some time and observe the battery.

spaggi, 2024-05-18:
I will go on vacation soon so this will be a good time to monitor potential drain. Ill check the messages too. Thank you! For now I will close this as there might be nothing wrong

closed 2024-05-18 by spaggi