technyon/nuki_hub #366: Last "authorizationName" in log overwrites all "authorizationName" entries in nuki/lock/log

Dielee, 2024-05-19:
First, thanks for your effort. 8.34 is working great!
Since the last 8.34 update, the "authorizationName" seems to have been overwritten by the latest "authorizationName".
In 8.33, this feature worked well.

Topic message: 

```
[
	{
		"index": 4289,
		"authorizationId": 926285825,
		"authorizationName": "Linus (Keypad)",
		"timeYear": 2024,
		"timeMonth": 5,
		"timeDay": 19,
		"timeHour": 8,
		"timeMinute": 26,
		"timeSecond": 9,
		"type": "KeypadAction",
		"action": "Unlatch",
		"completionStatus": "success"
	},
	{
		"index": 4288,
		"authorizationId": 902758402,
		"authorizationName": "Linus (Keypad)",
		"timeYear": 2024,
		"timeMonth": 5,
		"timeDay": 19,
		"timeHour": 6,
		"timeMinute": 52,
		"timeSecond": 39,
		"type": "LockAction",
		"action": "Unlock",
		"trigger": "system",
		"completionStatus": "success"
	},
	{
		"index": 4287,
		"authorizationId": 902758402,
		"authorizationName": "Linus (Keypad)",
		"timeYear": 2024,
		"timeMonth": 5,
		"timeDay": 18,
		"timeHour": 19,
		"timeMinute": 36,
		"timeSecond": 21,
		"type": "LockAction",
		"action": "Lock",
		"trigger": "system",
		"completionStatus": "success"
	},
	{
		"index": 4286,
		"authorizationId": 926285825,
		"authorizationName": "Linus (Keypad)",
		"timeYear": 2024,
		"timeMonth": 5,
		"timeDay": 18,
		"timeHour": 14,
		"timeMinute": 12,
		"timeSecond": 21,
		"type": "KeypadAction",
		"action": "Unlatch",
		"completionStatus": "success"
	},
	{
		"index": 4285,
		"authorizationId": 902758402,
		"authorizationName": "Linus (Keypad)",
		"timeYear": 2024,
		"timeMonth": 5,
		"timeDay": 18,
		"timeHour": 14,
		"timeMinute": 8,
		"timeSecond": 14,
		"type": "LockAction",
		"action": "Unlock",
		"trigger": "system",
		"completionStatus": "success"
	}
]
```

Nuki app log:

[image]

iranl, 2024-05-19:
Confirmed.

Bug was introduced as part of the fixes for in #344 in #345.

Fixed by #368

closed 2024-05-21 by technyon

Dielee, 2024-06-22:
@iranl is this fix available in 8.35 ? Thanks!

iranl, 2024-06-22:
It is

Dielee, 2024-06-22:
Works, thanks a lot!