technyon/nuki_hub #374: ring detect and ring locked loging permanent

Camouflagge, 2024-05-27:
Hello, everything is working fine for me so far. I am using the opener with keypad 2.0. The opener is only connected as a different gate (blue and purple cable). 

Unfortunately, HA constantly recognizes the ring. So in 2 seconds the ringing was detected. It only stops when you activate the RTO, which I don't need. 

At the same time, ringlocked is also constantly triggered.
Can you switch that off? 
Version current 8.34

iranl, 2024-05-27:
You can just disable the home assistant entities for ring (event and sensor) on the home assistant side.

Camouflagge, 2024-05-27:
OK, there's no other choice :D

iranl, 2024-05-27:
Your use case seems pretty fringe to me and it does seem on the surface that the ring detection in Nuki Hub is correct (e.g. the Nuki Opener is signaling that a ring signal is being received based on your wiring situation), but if this is something that occurs for multiple users we could build a disable switch in Nuki Hub for ring detection.

Camouflagge, 2024-05-27:
I already have an electric motor in the door, I just wanted to use the Keypad2 and the only option is the opener :D (potential-free contact)

closed 2024-05-27 by iranl