technyon/nuki_hub #378: Nuki hub crash by changing setting of Nuki Opener

fir3drag0n, 2024-05-29:
As described earlier, I have the problem that I cannot change settings with Nuki Opener if I change the settings

[image]

iranl, 2024-05-29:
Thanks for helping us debug this issue.

Please do the following:
- Flash the binary for your device: 
[esp32-debug.zip](https://github.com/technyon/nuki_hub/files/15486264/esp32-debug.zip)
[esp32-c3-debug.zip](https://github.com/technyon/nuki_hub/files/15486263/esp32-c3-debug.zip)
[esp32-s3-debug.zip](https://github.com/technyon/nuki_hub/files/15486265/esp32-s3-debug.zip)

- Enable `Enable MQTT logging` in `Advanced MQTT and Network Configuration`
- Open MQTT Explorer and connect to your MQTT server
- Check that you have a `nukiopener/maintanence/log` topic
- Try to change to setting for `Doorbell Suppression` and `Button Enabled`. You can use the dropdown/switch in Home Assistant for this or publish `{ "doorbellSuppression": "CM & Ring" }` or `{ "buttonEnabled": "1" }` to `nukiopener/configuration/action`.
- Report back here with the last 5-10 messages in `nukiopener/maintanence/log` before the message `MQTT connected` is seen (this means the ESP has restarted)

- It should look like this (but probably won't because of the crash, but hopefully will help pinpoint the exact location of the problem)

```
onConfigUpdateReceived: Config update recieved
onConfigUpdateReceived: Initial checks done, continuing
onConfigUpdateReceived: ACL preferences loaded
onConfigUpdateReceived: checking key name
[...]
onConfigUpdateReceived: checking key buttonEnabled
onConfigUpdateReceived: key exists
onConfigUpdateReceived: value = 1 (or 0 depending on if you are enabling or disabling the button)
onConfigUpdateReceived: Changing enableButton
onConfigUpdateReceived: result: 1
onConfigUpdateReceived: checking key ledEnabled
[...]
onConfigUpdateReceived: Success
Reading config. Result: success
Reading advanced config. Result: success
```

sjors-lemniscap, 2024-05-30:
Just tested with the esp32_debug binary. I can reproduce the crash reported by @fir3drag0n by changing either the `Doorbell Suppression` or `Button Enabled` setting in HA (but also publishing it directly to the MQTT `configuration/action` topic).

Worth noting (for future reference) that the logs are published through the `nuki/maintanence/log` topic. There are unfortunately no new MQTT messages before the `MQTT connected` message (which indicates a restart) when changing the settings meaning that the ESP32 is already crashing early in the process.

iranl, 2024-05-30:
Should be fixed by #385. Have been able to pinpoint the faulty code.

Please test:
[esp32.zip](https://github.com/technyon/nuki_hub/files/15505861/esp32.zip)
[esp32-s3.zip](https://github.com/technyon/nuki_hub/files/15505858/esp32-s3.zip)
[esp32-c3.zip](https://github.com/technyon/nuki_hub/files/15505860/esp32-c3.zip)

technyon, 2024-06-01:
Merged into master

PR #385 (2024-06-01): Fix opener config
Closes #378
closed 2024-06-01 by technyon