technyon/nuki_hub #382: Nuki Keypad display authorization code information

sjors-lemniscap, 2024-05-30:
As explained on Discord the actual authorization code is not exposed in MQTT topics such as `lock/log` and `keypad/json` as a security measure.

It would be great to have an additional checkbox on the "**Nuki General Access Control**" page which allows the actual authorization code to be exposed.

iranl, 2024-05-30:
Note that the code is not available as part of  the authorisation data published in `lock/log`. Nuki devices don't publish the code there.

sjors-lemniscap, 2024-05-30:
> Note that the code is not available as part of the authorisation data published in `lock/log`. Nuki devices don't publish the code there.

Makes sense, `keypad/json` would be sufficient - don't think there are any other places.

PR #384 (2024-06-01): Publish keypad code and rolling log, update BLEScanner to 1.1.0
- Closes #382 
- Closes #383 
- Updates BLEScanner to 1.1.0
closed 2024-06-01 by technyon