technyon/nuki_hub #395: [Feature Request] Adding firmware version to Home Assistant

duhow, 2024-06-06:
Having current code as:
https://github.com/technyon/nuki_hub/blob/71314ffccf2beab38c819e0a151113df0fdaa277/src/Network.cpp#L761-L770

The idea would be to send the `nukiDevice` variable and add something like:
```c
json["dev"]["hw"] = _nukiConfig->hardwareVersion().c_str();
json["dev"]["sw"] = _nukiConfig->firmwareVersion().c_str();
```

📝 **Details here:** https://github.com/home-assistant/core/blob/dev/homeassistant/components/mqtt/abbreviations.py#L263-L280

duhow, 2024-06-06:
Additionally for the Nuki Hub, it is also possible to add an `update` entity:

https://www.home-assistant.io/integrations/update.mqtt/

iranl, 2024-06-06:
We already have a Home Assistant update entity.

https://github.com/technyon/nuki_hub/blob/71314ffccf2beab38c819e0a151113df0fdaa277/src/Network.cpp#L984

Because HTTPS OTA is not implemented (yet, it is  on my To-Do list) it only notifies you of new versions and links to the release notes. When HTTPS OTA is implemented you will also be able to update with one click from HA.

As to adding HW and SW version to device config I have no objection. 
Seeing as you basically already have the code worked out (you would need to add hw/sw in the arguments of `publishHASSConfig` and call it with these values from `Nuki(Opener)Wrapper`) is this something you need us to implement or can you create a PR yourself?

duhow, 2024-06-06:
I can try copy the strings and pass them to the function, however I was expecting something more like "sending the whole device variable data", but the few attempts I did were not successful...

So I raised this issue more to have an agreement on, how would should we proceed best?

mundschenk-at, 2024-06-06:
I haven't been following all the recent PRs in detail, but one thing I wanted to change for some time is untangling Nuki **Hub** and the Nuki Smart **Lock** in HA. Ideally, there should be separate devices in HA for the Nuki Hub, the Smart Lock and the Opener (and maybe the Keypad and the Door Sensor, though I don't know if the BLE API includes enough information about those to warrant handling them as separate devices).

The Lock and Opener would be connected via the Hub. (I have it all written down on paper, but making a nice RFC about it was delayed all the time.), and each device could display its correct HW and FW version via the HA device registry. There's currently a draft PR for enabling a [new device-based discovery schema](https://github.com/home-assistant/core/pull/118757) for HA (it was merged already, but got pulled out of 2024.6.0 at the last minute), which would greatly reduce the number of discovery messages that need to be sent.