technyon/nuki_hub #397: ESP32 S3 does not show up as wifi device after successfully flashing via browser

22chrs, 2024-06-06:
Using the web flash tool works is successful, but the esp does not show up as wifi device. (I tried to repower the device after the flash.) confirmed this with 2 devices. 

Fun fact: using the esp web flash tool from home assitant for a new esp home device is successful with the device showing as wifi device after this.

I use this device:
https://de.aliexpress.com/item/1005006504766603.html

Any idea?

iranl, 2024-06-06:
Please provide serial logs using a program like hterm

iranl, 2024-06-06:
Note that if the device was not erased and was used previously with any other firmware (e.g. ESPHome) it will just connect with previously saved wifi credentials and not open up the WIFI configuration portal.

22chrs, 2024-06-10:
Thanks for your help. I definitly erased the device. This is the log after booting:

ELF file SHA256: 76d60dae6f83b890

E (329) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377c48
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
E (301) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.

assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)

Backtrace: 0x4037803a:0x3fceb180 0x403806c1:0x3fceb1a0 0x40386c19:0x3fceb1c0 0x4207b4fa:0x3fceb2f0 0x403779ab:0x3fceb320 0x403cd83b:0x3fceb350 0x403cdafa:0x3fceb380 0x403c9925:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED

iranl, 2024-06-10:
Seems like something went wrong with flashing.
Please flash again using the web flash or if that doesn't work using the release binaries at https://github.com/technyon/nuki_hub/releases/tag/8.34

technyon, 2024-06-16:
@22chrs Were you able to resolve the issue?

closed 2024-06-20 by technyon

22chrs, 2024-07-28:
Not for now. I will try later again.

PR #402 (2024-06-20): Change ESP32-S3 to 4MB build
## Description:
Builds S3 for devices with 4MB instead of 8MB flash. Smaller flash size should work without problems on devices with more flash, but this does not work the other way around, even when the partition table fits within 4MB.

Tested without issue using both the webflash binary and separate binaries on a S3 with 16MB flash.

**Related issue:** fixes #401 fixes #397 

## Checklist:
  - [x] The pull request is done against the latest master branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change com [... cut]