technyon/nuki_hub #401: S3 Mini Flashing Error using Web Flasher

mutasim, 2024-06-19:
I'm unable to boot from my S3 Mini when flashing using the web flasher. The logs show the following:

```
E (301) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.

assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)

Backtrace: 0x4037803a:0x3fceb180 0x403806c1:0x3fceb1a0 0x40386c19:0x3fceb1c0 0x4207b4fa:0x3fceb2f0 0x403779ab:0x3fceb320 0x403cd83b:0x3fceb350 0x403cdafa:0x3fceb380 0x403c9925:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED

ELF file SHA256: 76d60dae6f83b890

E (329) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377c48
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
E (301) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.

assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)

Backtrace: 0x4037803a:0x3fceb180 0x403806c1:0x3fceb1a0 0x40386c19:0x3fceb1c0 0x4207b4fa:0x3fceb2f0 0x403779ab:0x3fceb320 0x403cd83b:0x3fceb350 0x403cdafa:0x3fceb380 0x403c9925:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED

ELF file SHA256: 76d60dae6f83b890
```

I have tried flashing twice, still no luck. Shall I tried the other flashing method?

technyon, 2024-06-19:
You posted the problem:

`E (301) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.`

The ESP you bought has 4MB of Flash, but 8MB are required.

mutasim, 2024-06-19:
Ah ok, thanks - any suggestions on a reliable ESP model?

iranl, 2024-06-19:
Note that the release binaries might not work as well.
Bootloader probable needs to be built with flash size set to 4MB.
So this might need new binaries (for both webflash and manual flash) after all

iranl, 2024-06-19:
Please flash using the binaries in https://github.com/technyon/nuki_hub/actions/runs/9583173692/artifacts/1617236784

mutasim, 2024-06-19:
Amazing - that worked. Thanks!! 👍

PR #402 (2024-06-20): Change ESP32-S3 to 4MB build
## Description:
Builds S3 for devices with 4MB instead of 8MB flash. Smaller flash size should work without problems on devices with more flash, but this does not work the other way around, even when the partition table fits within 4MB.

Tested without issue using both the webflash binary and separate binaries on a S3 with 16MB flash.

**Related issue:** fixes #401 fixes #397 

## Checklist:
  - [x] The pull request is done against the latest master branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change com [... cut]
closed 2024-06-20 by technyon