technyon/nuki_hub #403: Restart if Bluetooth beacons not received, disabled by default

MihaiKrieger, 2024-06-20:
### PROBLEM DESCRIPTION
I have a brand new installation of NukiHub on ESP32 D1 mini. It has been working great for one week, until today, when I discovered that the BT signals have flatlined.
[image]

In the official documentation it days that _Restart if Bluetooth beacons not received_ has a default setting of 60 (seconds).
I have checked mine. It was disabled. I had not intentionally disabled this setting.
[image]

After a restart of the board, it started working normally again. Opening the issue just to bring this information to the dev's attention.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.comhttps://github.com/technyon/nuki_hub/discussions/technyon/nuki_hub/issues)
- [x ] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ x] Device used (e.g., ESP32-S3): ESP32 D1 Mini
- [ ] Nuki Hub binary firmware version number used: 8.34
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info:
```
Nuki Hub version: 8.34
run: true
confVersion: 834
deviceId: 2209271902
deviceIdOp: 2209271902
nukiId: ***
nukidOp: 
mqttbroker: 192.168.1.2
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: true
lockena: true
lockpin: 1
mqttpath: nuki
openerena: false
openerpin: 
openercont: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
hassConfigUrl: 
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
nwwififb: false
rssipb: 60
hostname: NukiHub
nettmout: -1
restdisc: true
rstbcn: 120
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpCntrlEnabled: true
kpInfoEnabled: false
aclLckOpn: 
tcCntrlEnabled: false
tcInfoEnabled: false
confLckBasAcl: 
confLckAdvAcl: 
confOpnBasAcl: 
confOpnAdvAcl: 
accLvl: 
regAsApp: false
nrRetry: 0
rtryDelay: 100
crdusr: ***
crdpass: ***
pubAuth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
latest: 8.35
MQTT connected: Yes
Lock firmware version: 4.2.8
Lock hardware version: 7.0
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: No
Lock has keypad: No
Lock ACL (Lock): Allowed
Lock ACL (Unlock): Allowed
Lock ACL (Unlatch): Allowed
Lock ACL (Lock N Go): Allowed
Lock ACL (Lock N Go Unlatch): Allowed
Lock ACL (Full Lock): Allowed
Lock ACL (Fob Action 1): Allowed
Lock ACL (Fob Action 2): Allowed
Lock ACL (Fob Action 3): Allowed
Lock config ACL (Name): Disallowed
Lock config ACL (Latitude): Disallowed
Lock config ACL (Longitude): Disallowed
Lock config ACL (Auto Unlatch): Disallowed
Lock config ACL (Pairing enabled): Allowed
Lock config ACL (Button enabled): Allowed
Lock config ACL (LED flash enabled): Allowed
Lock config ACL (LED brightness): Disallowed
Lock config ACL (Timezone offset): Disallowed
Lock config ACL (DST mode): Disallowed
Lock config ACL (Fob Action 1): Disallowed
Lock config ACL (Fob Action 2): Disallowed
Lock config ACL (Fob Action 3): Disallowed
Lock config ACL (Single Lock): Allowed
Lock config ACL (Advertising Mode): Disallowed
Lock config ACL (Timezone ID): Disallowed
Lock config ACL (Unlocked Position Offset Degrees): Disallowed
Lock config ACL (Locked Position Offset Degrees): Allowed
Lock config ACL (Single Locked Position Offset Degrees): Disallowed
Lock config ACL (Unlocked To Locked Transition Offset Degrees): Disallowed
Lock config ACL (Lock n Go timeout): Disallowed
Lock config ACL (Single button press action): Disallowed
Lock config ACL (Double button press action): Disallowed
Lock config ACL (Detached cylinder): Disallowed
Lock config ACL (Battery type): Disallowed
Lock config ACL (Automatic battery type detection): Disallowed
Lock config ACL (Unlatch duration): Disallowed
Lock config ACL (Auto lock timeout): Disallowed
Lock config ACL (Auto unlock disabled): Allowed
Lock config ACL (Nightmode enabled): Disallowed
Lock config ACL (Nightmode start time): Disallowed
Lock config ACL (Nightmode end time): Disallowed
Lock config ACL (Nightmode auto lock enabled): Disallowed
Lock config ACL (Nightmode auto unlock disabled): Disallowed
Lock config ACL (Nightmode immediate lock on start): Disallowed
Lock config ACL (Auto lock enabled): Allowed
Lock config ACL (Immediate auto lock enabled): Disallowed
Lock config ACL (Auto update enabled): Disallowed
Network device: Built-in Wi-Fi
Uptime: 10 minutes
Heap: 78612
Stack watermarks: nw: 5036, nuki: 1860, pd: 252
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
N/A, just an observation

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Fresh installations to have the Restart if _Bluetooth beacons not received_ setting set to 60 by default, as per official documentation.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-06-20:
Thanks for the report. Will update the docs or the defaults for 8.36

technyon, 2024-06-20:
Let's better update the default, it makes sense to have this enable.

closed 2024-07-05 by technyon