technyon/nuki_hub #406: Nuki Lock state undefined since version 8.35

MihaiKrieger, 2024-06-20:
### PROBLEM DESCRIPTION
With the update to the new 8.35 FW, the Nuki Lock state is **undefined**.
Repairing the lock did not solve the problem.
Power cycling the ESP32 Board did not solve the problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [ x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ ] Device used (e.g., ESP32-S3): ESP32 D1 mini
- [ ] Nuki Hub binary firmware version number used: 8.35
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info:
```
System Information
Nuki Hub version: 8.35
Nuki Hub build: 9596492875.1136.1 (release)
run: true
confVersion: 835
deviceId: 4178833015
deviceIdOp: 2209271902
nukiId: ***
nukidOp: 
mqttbroker: 192.168.1.2
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: true
websrvena: false
lockena: true
lockpin: 1
mqttpath: nuki
openerena: false
openerpin: 
openercont: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
maxtc: 
opmaxtc: 
enabtlprst: false
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
hassConfigUrl: 
buffsize: 
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
nwwififb: false
rssipb: 60
hostname: NukiHub
nwbestrssi: false
nettmout: -1
restdisc: true
rstbcn: 120
lockStInterval: 1800
tcPerEntry: false
kpPerEntry: false
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpCntrlEnabled: true
kpInfoEnabled: false
kpPubCode: false
tcCntrlEnabled: false
tcInfoEnabled: false
cnfInfoEnabled: true
regAsApp: false
regOpnAsApp: false
nrRetry: 0
rtryDelay: 100
crdusr: ***
crdpass: ***
disnonjson: false
pubAuth: false
pubdbg: false
prdtimeout: 60
offHybrid: false
hybridTimer: 600
hybridAct: false
hybridRtry: false
hasmac: false
macb0: 
macb1: 
macb2: 
latest: 8.35
tsksznetw: 
tsksznuki: 
authmaxentry: 
kpmaxentry: 
tcmaxentry: 
MQTT connected: Yes
Lock firmware version: 
Lock hardware version: 
Lock paired: Yes
Lock valid PIN set: Yes
Lock has door sensor: No
Lock has keypad: No
Lock ACL (Lock): Allowed
Lock ACL (Unlock): Allowed
Lock ACL (Unlatch): Allowed
Lock ACL (Lock N Go): Allowed
Lock ACL (Lock N Go Unlatch): Allowed
Lock ACL (Full Lock): Allowed
Lock ACL (Fob Action 1): Allowed
Lock ACL (Fob Action 2): Allowed
Lock ACL (Fob Action 3): Allowed
Lock config ACL (Name): Disallowed
Lock config ACL (Latitude): Disallowed
Lock config ACL (Longitude): Disallowed
Lock config ACL (Auto Unlatch): Disallowed
Lock config ACL (Pairing enabled): Allowed
Lock config ACL (Button enabled): Allowed
Lock config ACL (LED flash enabled): Allowed
Lock config ACL (LED brightness): Disallowed
Lock config ACL (Timezone offset): Disallowed
Lock config ACL (DST mode): Disallowed
Lock config ACL (Fob Action 1): Disallowed
Lock config ACL (Fob Action 2): Disallowed
Lock config ACL (Fob Action 3): Disallowed
Lock config ACL (Single Lock): Allowed
Lock config ACL (Advertising Mode): Disallowed
Lock config ACL (Timezone ID): Disallowed
Lock config ACL (Unlocked Position Offset Degrees): Disallowed
Lock config ACL (Locked Position Offset Degrees): Disallowed
Lock config ACL (Single Locked Position Offset Degrees): Disallowed
Lock config ACL (Unlocked To Locked Transition Offset Degrees): Disallowed
Lock config ACL (Lock n Go timeout): Disallowed
Lock config ACL (Single button press action): Disallowed
Lock config ACL (Double button press action): Disallowed
Lock config ACL (Detached cylinder): Disallowed
Lock config ACL (Battery type): Disallowed
Lock config ACL (Automatic battery type detection): Disallowed
Lock config ACL (Unlatch duration): Disallowed
Lock config ACL (Auto lock timeout): Disallowed
Lock config ACL (Auto unlock disabled): Allowed
Lock config ACL (Nightmode enabled): Disallowed
Lock config ACL (Nightmode start time): Disallowed
Lock config ACL (Nightmode end time): Disallowed
Lock config ACL (Nightmode auto lock enabled): Disallowed
Lock config ACL (Nightmode auto unlock disabled): Disallowed
Lock config ACL (Nightmode immediate lock on start): Disallowed
Lock config ACL (Auto lock enabled): Allowed
Lock config ACL (Immediate auto lock enabled): Disallowed
Lock config ACL (Auto update enabled): Disallowed
Network device: Built-in Wi-Fi
BSSID of AP: C2:09:D0:1E:F7:A1
Uptime: 3 minutes
Heap: 76664
Stack watermarks: nw: 7092, nuki: 5420, pd: 7092
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

### TO REPRODUCE
_Steps to reproduce the behavior:_ Update to the new firmware via the OTA interface from the board.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._ Nuki lock state should be _locked_, _unlocked_, _locking_, or _unlocking_, based on the situation. However, it remains in _undefined_ state.

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
I have rolled back to the 8.34 version, and it started working again. 
Post downgrade screenshot:

[image]

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2024-06-20:
That's very strange, and sure didn't happen during testing. Could you get MQTT or serial logs with fw 8.35. For serial logs you can use hterm and connect 115200 baud.

iranl, 2024-06-20:
Is the state only undefined in the webconfigurator or also in MQTT?

Serial  logs would be helpfull

MihaiKrieger, 2024-06-20:
I'm sorry I did not have the time to install 8.35 again and gather the logs for you. Making the door unavailable during a busy evening would put me in a bad light at home. 
I don't have any MQTT logs because none were generating. I would have to collect the serial ones.
@iranl - Other observations:

1. In MQTT, the lock was showing as available, and locked. I think it was because that was the state of the door before updating the firmware.
2. MQTT was appearing as "connected" in the device's page in HA. _MQTT Connected - on_
3. The _undefined_ state was **only** in the web configurator, nowhere else. In the native app, the correct state of the lock was displayed, and in HA, the door had the _locked_ state.
4. When I called the _unlock_ service in HA, the door would indeed unlock! However, the updated state would not display. The MQTT state was still _locked_ and the web configurator state would still be _undefined_.
5. No other subsequent actions would take effect. I called the _lock_ service from HA, but the lock wouldn't react at all.
6. During this time, the MQTT log was not being populated with any new entries
7. No other actions would take effect from HA. E.g: toggling _LED enabled_. on 8.34 version, this works correctly. My lock has a PIN defined in the web configurator.

I hope this helps further. I'm doing my best to get around to reinstall 8.35, reproduce the issue  and collect the logs.

iranl, 2024-06-20:
Your observations point to an issue with BLE communications.

Considering your other issue (#403) where you didn't recieve BLE beacons (which is not very common in my experience) i'm not yet convinced this is 8.35 related and not hardware/situation dependant.

The fact that you are able to unlock the door also point in the direction that some BLE communication is possible, even if it is only sending and not recieving.
How far away is your device from the lock? Can you try to reduce this distance?

MihaiKrieger, 2024-06-20:
Strange, to only have the issue on the new FW.  The distance between lock and esp board is around 1.5 -2 meters, with a BT strength of -64 dBm. 
It is stable again with 8.34 FW, and with respect to my other issue, it is no longer an issue, now that I have corrected the settings and the board restarts itself correctly. 

Below an evolution of the BT strength for a couple of hours. 
The flatline is when I had the problem earlier today, again, solved nicely with a reboot. :) 

[image]

iranl, 2024-06-20:
Can't reproduce this on 8.35 with:
- ESP32 D1 Mini (Hybrid mode)
- Second ESP32 D1 Mini (App mode)
- ESP32 S3 (Hybrid mode)
- Second ESP32 S3 (App mode)
- ESP32 C3 (App mode)

Using Lock 4.0 and Lock 4.0 Pro

Very interested in serial logs (might need logs from the debug binary) when you get the chance

pipip, 2024-06-20:
I have the same issue with Nuki Smart Lock 3.0 Pro and M5Stack Atom.
--> Nuki Lock state: undefined

iranl, 2024-06-21:
@pipip Please provide a serial log while running 8.35

gerardovf, 2024-06-21:
While I try to get a serial log, I would like to point out that, after deleting the 'homeassistant' topic in the broker the nuki hub never sends it complete again (just homeassistant/status -> online)
The "lock/query/lockstateCommandResult" topic sends "undefined" payload.

iranl, 2024-06-21:
Homeassistant discovery is only updated when a valid config from the Nuki device is received.
Info page from @MihaiKrieger suggests that a valid config is never received (There is no firmware/hardware version of the Nuki device).
So that would match with your observation.

It is possible that a  serial log when on a release binary will not give much information.
A serial log while running a debug firmware (from https://github.com/technyon/nuki_hub/actions/runs/9598815492) will provide more information.

iranl, 2024-06-21:
I may have found the issue in the info page supplied by @MihaiKrieger.

nrRetry is set to 0. 

There was a change in 8.35 with how this value is used.
Setting this to 0 will mean that most commands are never sent.

This setting is basically not named correctly as of this time as it is not interpreted as number of retries but as maximum number of total tries of a command.

I'll make a PR to fix this.

Please check the value or `nrRetry` on your info page and if this is 0 set it to at least 1 on the "Nuki configuration" page with the setting "Number of retries if command failed" (default is 3)

iranl, 2024-06-21:
Yes that value should not be zero on 8.35.

Do note that this is considered a bug, introduced by changes by me in 8.35.

It should be perfectly fine to have this at 0 and will be again in 8.36

If this works no serial logs would have to be collected.

Should this work for @MihaiKrieger and not for others please open your own issue and post correct info as per the issue template

MihaiKrieger, 2024-06-21:
I can confirm this works. Posted a new info page above, but somehow the formatting ended up weird. 
it is perfectly working as expected with the value "3".

[image]

iranl, 2024-06-21:
It is up to you. If you don't want commands to be retried you can set it to 0.

Best practice is the default ( = 3)

gerardovf, 2024-06-21:
> Yes that value should not be zero on 8.35. Set it to atleast 1, but default and advised value is 3
> 
> Do note that this is considered a bug, introduced by changes by me in 8.35.
> 
> It should be perfectly fine to have this at 0 and will be again in 8.36
> 
> If this works no serial logs would have to be collected.
> 
> Should this work for @MihaiKrieger and not for others please open your own issue and post correct info as per the issue template

That was the solution! Setting nrRetry to 3 solved the problem. THANKS a lot!!!

closed 2024-06-22 by technyon

FrankSchoene, 2024-07-17:
Setting retries does the trick for me. As a note in the release notes it would have saved me 3 hours :-)

PR #411 (2024-06-22): Fix nr of retries
## Description:

In 8.35 when number of retries is set to 0 no commands will be executed (except lock actions)

**Related issue (if applicable):** fixes #406 

## Checklist:
  - [x] The pull request is done against the latest master branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works
  - [x] I accept the [CLA](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md#contributor-license-agreement-cla).