technyon/nuki_hub #408: AuthorizationId doesn't change on rolling auth log

mutasim, 2024-06-20:
Version: 8.35

I've been experimenting with the Hass rolling_authorization_log sensor, the state + AuthorizationId attribute doesn't change for me when trying different codes/fingerprints.

The rest of the attributes do update correctly, but the state and AuthorizationId stay the same. Same with the AuthorizationId on lock/log on MQTT.

(Separately - I wonder if something like a timestamp would be a better state for the auth log sensor in Hass. I think this would make tracking events easier in the case where the same code/fingerprint is used again later that day for example. In the current implementation I can't see a clean/easy way to handle that scenario)

propi62, 2024-06-20:
an easy way to trigger automations if code/fingerprint is entered.
also trigger automation whene last key entered is the same as before.
if homeassistant reboots the last key entered should not trigger the automation.

iranl, 2024-06-21:
State/AuthorizationId not changing with different codes/fingerprints is correct.
The keypad has one authorization (id) which is the same for all actions executed through the keypad, regardless of used code and method. Trigger and codeId are specific to certain codes and actions (e.g. fingerprint, arrow or code).

I do agree that it might not be the best value for the HA sensor.
I'll change it to index which will count up with every new log entry.

There are some more needed improvements on the rolling log as rebooting the ESP currently resets the rolling log and replays the max auth log count (default 5)

propi62, 2024-06-25:
my automation triggers whene i restart nukihub or homeassistant.

closed 2024-07-05 by technyon