technyon/nuki_hub #410: How upgrade?

zaratustraelimpio, 2024-06-21:
Hi, 

how can I update the version of my nuki hub to new versions? I have 8.34 and I am having some problems and I have seen that if I access through the browser there is the "Upgrade firmware" option but it asks for a .bin file, which file should I do it with? with the complete .zip file? or is it one that is inside this one?,

thanks, 

regards.

MihaiKrieger, 2024-06-21:
No. You click "Download latest binary from GitHub". a file with the extension .bin will be downloaded. You have to upload that one.

zaratustraelimpio, 2024-06-21:
> No. You click "Download latest binary from GitHub". a file with the extension .bin will be downloaded. You have to upload that one.

Where i can download? Can you show me the link?

jolichter, 2024-06-21:
> Where i can download? Can you show me the link?

Hi, can't you see this?
[image]

iranl, 2024-06-21:
The "Download latest binary from GitHub" button is available on the nukihub/ota (Firmware upgrade) page, but only when the option "Check for updates every 24h" is enabled in the "MQTT and network" settings. This is where @MihaiKrieger is referring to.

When using the release zip (where @jolichter is referring to) you need to upload the nuki_hub_esp32[-S3/C3].bin file from inside the zip file.

You can also find this binary on github in the /ota directory.

closed 2024-06-21 by iranl

sebax80, 2024-06-24:
where is the upload file

[image]

iranl, 2024-06-24:
Please just read the comments above.

`You can also find this binary on github in the /ota directory`

sebax80, 2024-06-24:
Please tell me what should I do to get help to upgrade?

MihaiKrieger, 2024-06-24:
1. Open the IP address of your Nukihub (local IP on your network)
You shoould see this:

[image]

2. click on the option „Firmware update”. You should get this:

[image]

3. Click on: „Download latest binary from GitHub” . Like this:
[image]

4. Download the file called **nuki_hub_esp32.bin** to your desktop.

5. Choose the file you have just downloaded from here: 
[image]

6. With the file inserted in that field, click on „Upload file”. The web interface does its job than.
[image]