technyon/nuki_hub #413: Nuki Hub gets unavailable for a couple of hours a day - also tries to reconnect every minute afterwards

DeckardCain2014, 2024-06-26:
### PROBLEM DESCRIPTION
Nuki Hub is several times a day unavailable via MQTT for several hours.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32 NodeMCU Module
- [x] Nuki Hub binary firmware version number used: 8.35
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/info:
```
System Information
Nuki Hub version: 8.35
Nuki Hub build: 9596492875.1136.1 (release)
run: true
confVersion: 835
deviceId: 4004615175
deviceIdOp: 4004615175
nukiId: ***
nukidOp: 
mqttbroker: 192.168.178.170
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: true
websrvena: false
lockena: true
lockpin: 1
mqttpath: nuki
openerena: false
openerpin: 
openercont: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
maxtc: 
opmaxtc: 
enabtlprst: false
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
hassConfigUrl: 
buffsize: 
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
nwwififb: false
rssipb: 60
hostname: nukihub
nwbestrssi: true
nettmout: 500
restdisc: true
rstbcn: 60
lockStInterval: 1800
tcPerEntry: false
kpPerEntry: false
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpCntrlEnabled: true
[... 184 lines cut ...]
2024-06-26 18:42:12: New client connected from 192.168.178.81:60194 as nukihub (p2, c0, k15, u'mqttUser').
2024-06-26 18:42:34: Client nukihub has exceeded timeout, disconnecting.
2024-06-26 18:42:47: New connection from 192.168.178.81:60195 on port 1883.
2024-06-26 18:42:47: New client connected from 192.168.178.81:60195 as nukihub (p2, c0, k15, u'mqttUser').
2024-06-26 18:43:10: Client nukihub has exceeded timeout, disconnecting.
2024-06-26 18:43:23: New connection from 192.168.178.81:60196 on port 1883.
2024-06-26 18:43:23: New client connected from 192.168.178.81:60196 as nukihub (p2, c0, k15, u'mqttUser').
2024-06-26 18:43:46: Client nukihub has exceeded timeout, disconnecting.
2024-06-26 18:43:58: New connection from 192.168.178.81:60197 on port 1883.
2024-06-26 18:43:58: New client connected from 192.168.178.81:60197 as nukihub (p2, c0, k15, u'mqttUser').
2024-06-26 18:44:22: Client nukihub has exceeded timeout, disconnecting.
2024-06-26 18:47:32: New connection from 192.168.178.81:51014 on port 1883.
2024-06-26 18:47:32: New client connected from 192.168.178.81:51014 as nukihub (p2, c0, k15, u'mqttUser').
2024-06-26 19:02:14: Saving in-memory database to /data//mosquitto.db.
2024-06-26 19:32:15: Saving in-memory database to /data//mosquitto.db.
2024-06-26 20:02:16: Saving in-memory database to /data//mosquitto.db.
2024-06-26 20:13:26: New connection from 192.168.178.81:54669 on port 1883.
2024-06-26 20:13:26: Client nukihub already connected, closing old connection.
2024-06-26 20:13:26: New client connected from 192.168.178.81:54669 as nukihub (p2, c0, k15, u'mqttUser').
2024-06-26 20:15:23: New connection from 192.168.178.81:59567 on port 1883.
2024-06-26 20:15:23: Client nukihub already connected, closing old connection.
2024-06-26 20:15:23: New client connected from 192.168.178.81:59567 as nukihub (p2, c0, k15, u'mqttUser').
```

**(Please, remember to close the issue when the problem has been addressed)**

StefanX13, 2024-06-28:
Same here. If mosquitto Broker output is needed please just tell.

iranl, 2024-06-28:
I suggest trying the latest PR with Arduino Core 3 and IDF 5.1.
Besides many changes to the networking stack it also includes nuki hub specific improvements regarding MQTT.

You can find the binaries here: https://github.com/technyon/nuki_hub/actions/runs/9711394389

Do note that these binaries need to be flashed using USB/serial once because of changes to the partition table.

DeckardCain2014, 2024-06-29:
I tried flashing it with the Espressif Flash Download Tools, however it shows only 'null' in the command line and red fields in the UI

iranl, 2024-06-30:
I find that to be a horrible tool and exclusively use esptools.

See https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html or https://www.flamingo-tech.nl/2021/03/21/installing-and-using-esptools/ for how to install esptools.

You can just use the command in the how-to-flash.txt file. You only need to change the COM port.

DeckardCain2014, 2024-07-01:
I did that now. However, the file seems to big for my ESP32:
```
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
```

> esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file: nuki_hub.partitions.bin

DeckardCain2014, 2024-07-01:
Worked now. Different machine, different tool. I will update here how stable it is

DeckardCain2014, 2024-07-01:
Basically it works now. However, the main lock topic is unavailable
[image]

fir3drag0n, 2024-07-02:
Is there a solution to unknown state?

fir3drag0n, 2024-07-06:
> I suggest trying the latest PR with Arduino Core 3 and IDF 5.1. Besides many changes to the networking stack it also includes nuki hub specific improvements regarding MQTT.
>
> You can find the [...]

with this binary, I cannot get a connection by ethernet anymore, specifically with a M5Stack Atom PoE Kit - W5500.
Nuki Hub does not respond anymore. If I disconnect it from the module, I can reach it by wifi.

technyon, 2024-07-06:
I can conform that core 3 breaks both W5500 and LAN8720 ethernet support.

iranl, 2024-07-09:
Lets keep this open until we have confirmation this is fixed in latest builds

DeckardCain2014, 2024-07-10:
I just flashed the latest build (9876220444.1180.1), trying this now.

DeckardCain2014, 2024-07-14:
Works like a charm since then!

iranl, 2024-07-14:
Thanks for reporting back.

Closing as completed

closed 2024-07-14 by iranl

adamlc, 2024-07-21:
Unfortunately I'm still having issues with the lock being unavailable. 

While the builds from main are better. They still haven't fixed it for me!

technyon, 2024-07-22:
@adamlc Could you get serial logs while this is happening?

adamlc, 2024-07-22:
I'll see what I can do to capture them :)

adamlc, 2024-07-26:
Just a quick update. I've now had different behaviour without any changes.

MQTT is now reported as connected but updates aren't sent and everything seems out of sync with HA. Unfortunately I've not been able to get any logs at the time it happens.

Because of the instability I've had to sadly revert to 8.33 as that was rock solid before. All the new improvements to 8.35 did look fantastic and I hope with some more time I can debug.

adamlc, 2024-07-29:
Back on `8.33` which has since been rock solid!

technyon, 2024-07-31:
And the problems you had with 8.35, or the 9.0 beta, or both?

adamlc, 2024-08-30:
Sorry for the late reply on this. Seems to now be resolved on the stable 9.0 builds 👍

DeckardCain2014, 2024-08-30:
It's really stable now with 9.0! Thanks for all the work 🍾

PR #419 (2024-07-09): Fixes/Enhancement for MQTT and WiFi reconnect + network  timout + MQTT retain
## Description:

**Related issue (if applicable):** Possibly fixes #413 

## Checklist:
  - [x] The pull request is done against the latest master branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works
  - [x] I accept the [CLA](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md#contributor-license-agreement-cla).