technyon/nuki_hub #417: nuki hub status does not change

zaratustraelimpio, 2024-07-08:
Hi, 

I have a nuki hub with an esp32 board (8.35) and when I lock or unlock the Nuki lock, the status does not update, as well as the battery percentage, does anyone know what could be going on?,

thanks,

regards.

[image]

iranl, 2024-07-08:
Use the issue template if you need support

iranl, 2024-07-09:
Closing for not complying to the [Contributing policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md)

Please reopen when you complied to the policy.

zaratustraelimpio, 2024-07-10:
### PROBLEM DESCRIPTION
I have a nuki hub with an esp32 board and when I lock or unlock the Nuki lock, the status does not update, as well as the battery percentage.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [X] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): ESP-WROOM-32 ESP-32
- [X] Nuki Hub binary firmware version number used: 8.35
- [X] Pre-compiled
- [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info:
```
System Information
Nuki Hub version: 8.35
Nuki Hub build: 9596492875.1136.1 (release)
run: true
confVersion: 835
deviceId: 3115457327
deviceIdOp: 3115457327
nukiId: ***
nukidOp: 
mqttbroker: 192.168.1.35
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: true
websrvena: false
lockena: true
lockpin: 0
mqttpath: nuki
openerena: false
openerpin: 
openercont: false
mqttoppath: nukiopener
maxkpad: 
opmaxkpad: 
maxtc: 
opmaxtc: 
enabtlprst: false
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
hassConfigUrl: 
buffsize: 
dhcpena: false
ipaddr: 192.168.1.92
ipsub: 255.255.255.0
ipgtw: 192.168.1.1
dnssrv: 192.168.1.1
nwhw: 1
nwwififb: false
rssipb: 60
hostname: nukihub
nwbestrssi: false
nettmout: -1
restdisc: false
rstbcn: -1
lockStInterval: 1800
tcPerEntry: false
kpPerEntry: false
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpCntrlEnabled: true
kpInfoEnabled: false
kpPubCode: false
tcCntrlEnabled: false
tcInfoEnabled: false
cnfInfoEnabled: false
regAsApp: false
regOpnAsApp: false
nrRetry: 0
rtryDelay: 100
crdusr: ***
crdpass: ***
disnonjson: false
pubAuth: false
pubdbg: false
prdtimeout: 60
offHybrid: false
hybridTimer: 600
hybridAct: false
hybridRtry: false
hasmac: false
macb0: 
macb1: 
macb2: 
latest: 8.35
tsksznetw: 
tsksznuki: 
authmaxentry: 
kpmaxentry: 
tcmaxentry: 
MQTT connected: Yes
Lock firmware version: 
Lock hardware version: 
Lock paired: Yes
Lock valid PIN set: No
Lock has door sensor: No
Lock has keypad: No
Lock ACL (Lock): Allowed
Lock ACL (Unlock): Allowed
Lock ACL (Unlatch): Allowed
Lock ACL (Lock N Go): Allowed
Lock ACL (Lock N Go Unlatch): Allowed
Lock ACL (Full Lock): Allowed
Lock ACL (Fob Action 1): Allowed
Lock ACL (Fob Action 2): Allowed
Lock ACL (Fob Action 3): Allowed
Lock config ACL (Name): Disallowed
Lock config ACL (Latitude): Disallowed
Lock config ACL (Longitude): Disallowed
Lock config ACL (Auto Unlatch): Disallowed
Lock config ACL (Pairing enabled): Allowed
Lock config ACL (Button enabled): Allowed
Lock config ACL (LED flash enabled): Allowed
Lock config ACL (LED brightness): Disallowed
Lock config ACL (Timezone offset): Disallowed
Lock config ACL (DST mode): Disallowed
Lock config ACL (Fob Action 1): Disallowed
Lock config ACL (Fob Action 2): Disallowed
Lock config ACL (Fob Action 3): Disallowed
Lock config ACL (Single Lock): Allowed
Lock config ACL (Advertising Mode): Disallowed
Lock config ACL (Timezone ID): Disallowed
Lock config ACL (Unlocked Position Offset Degrees): Disallowed
Lock config ACL (Locked Position Offset Degrees): Disallowed
Lock config ACL (Single Locked Position Offset Degrees): Disallowed
Lock config ACL (Unlocked To Locked Transition Offset Degrees): Disallowed
Lock config ACL (Lock n Go timeout): Disallowed
Lock config ACL (Single button press action): Disallowed
Lock config ACL (Double button press action): Disallowed
Lock config ACL (Detached cylinder): Disallowed
Lock config ACL (Battery type): Disallowed
Lock config ACL (Automatic battery type detection): Disallowed
Lock config ACL (Unlatch duration): Disallowed
Lock config ACL (Auto lock timeout): Disallowed
Lock config ACL (Auto unlock disabled): Allowed
Lock config ACL (Nightmode enabled): Disallowed
Lock config ACL (Nightmode start time): Disallowed
Lock config ACL (Nightmode end time): Disallowed
Lock config ACL (Nightmode auto lock enabled): Disallowed
Lock config ACL (Nightmode auto unlock disabled): Disallowed
Lock config ACL (Nightmode immediate lock on start): Disallowed
Lock config ACL (Auto lock enabled): Allowed
Lock config ACL (Immediate auto lock enabled): Disallowed
Lock config ACL (Auto update enabled): Disallowed
Network device: Built-in Wi-Fi
BSSID of AP: 28:D1:27:4E:95:E8
Uptime: 13102 minutes
Heap: 49688
Stack watermarks: nw: 8176, nuki: 5352, pd: 8176
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.
```

### TO REPRODUCE
press the unlock button in homeassistant or press the lock or unlock button in the Nuki app

### EXPECTED BEHAVIOUR
if the nuki lock is blocked, the unlock button should appear. If the lock is unlocked, the button should show lock

### SCREENSHOTS
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2024-07-10:
Hi, from your configuration I can see that the locks PIN isn't configured. You must have configured a PIN in the smartphone app before, please enter this PIN under credentials.

iranl, 2024-07-10:
nrOfRetry = 0. This is a bug in 8.35 that is already solved in master and solution is provided in the pinned issue #406

technyon, 2024-07-10:
@zaratustraelimpio Please change the "Number of retries if command failed" value to something other than 0, for example 3 makes sense.

zaratustraelimpio, 2024-07-10:
> @zaratustraelimpio Please change the "Number of retries if command failed" value to something other than 0, for example 3 makes sense.

Hi, yes, that is the solution, now it works correctly!!! closed the thread.

[image]

closed 2024-07-10 by zaratustraelimpio