technyon/nuki_hub #422: Presence

githubwesseln, 2024-07-13:
### PROBLEM DESCRIPTION
- since V8.35 the presence of some ble devices is not detected. In V8.34 these devices are detected.
- the nuki-device is detected in V8.35

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [ x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ x] Device used (e.g., ESP32-S3): ESP32
- [ x] Nuki Hub binary firmware version number used: 8.35
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info:
```

iranl, 2024-07-13:
Presence detection had changes in 8.35 that breaks some PD functionality.

Presence detection is currently slated to be removed in 8.36 as it causes issues for several users and was never really part of the core features of nuki hub.

I suggest considering moving your presence detection to a second esp32 running OpenMQTTGateway, ESPHome, Tasmota or ESPresence

kotylo, 2024-07-14:
shame, it was the only way to reliably reboot if it "hangs" on Wifi. Although if the newer versions don't crash or reconnect automatically, than that's ok, important is stability over Wifi...

technyon, 2024-07-14:
@kotylo How do you use it to reboot the ESP? If you a topic that regularly changes, you can use the uptime counter.

kotylo, 2024-07-14:
I just remember only playing with one of those options worked for stability and although every XXX minutes reboot happens, it doesn't hang anymore. I thought BT Beacons = Presence detection, and if they are gone, Nuki Hub would reboot automatically.
[image]

technyon, 2024-07-15:
The "Restart if no beacons" option will not be removed, and work as usual.

iranl, 2024-07-16:
Closing as not planned

closed 2024-07-16 by iranl