technyon/nuki_hub #431: Add support for ESP32-C3 ethernet WT ESP32C3-S5 DM5051NP (ETH01-Evo)

drstil, 2024-07-22:
### PROBLEM DESCRIPTION
Is there a reason, why ESP32-C3 SoC / DM5051NP are not supported via ethernet connection?

> Note: LAN8720 modules are only supported on the ESP32, not on the ESP32-S3, ESP32-C3 or ESP-C6

In my special case I have a ETH01-Evo DM9051NP WT ESP32C3-S5 board supporting PoE. Is there a way how I could add the support by myself? Is there any hardware restriction?

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32-C3
- [x] Nuki Hub binary firmware version number used: 8.35
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/info:
```
System Information
Nuki Hub version: 8.35
Nuki Hub build: 9596430697.1135.1 (release)
run: true
confVersion: 835
deviceId: 1950785077
deviceIdOp: 1950785077
nukiId: 
nukidOp: 
mqttbroker: homeassistant.local
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: true
checkupdates: false
websrvena: false
lockena: true
lockpin: 
mqttpath: nuki-2
openerena: false
openerpin: 
openercont: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
maxtc: 
opmaxtc: 
enabtlprst: false
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
hassConfigUrl: 
buffsize: 
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 5
nwwififb: false
rssipb: 60
hostname: nukihub-2
nwbestrssi: false
nettmout: -1
restdisc: true
rstbcn: -1
lockStInterval: 1800
tcPerEntry: false
kpPerEntry: false
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpCntrlEnabled: true
kpInfoEnabled: false
kpPubCode: false
tcCntrlEnabled: false
tcInfoEnabled: false
cnfInfoEnabled: true
regAsApp: false
regOpnAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: 
crdpass: 
disnonjson: false
pubAuth: false
pubdbg: false
prdtimeout: 60
offHybrid: false
hybridTimer: 600
hybridAct: false
hybridRtry: false
hasmac: false
macb0: 
macb1: 
macb2: 
latest: 
tsksznetw: 
tsksznuki: 
authmaxentry: 
kpmaxentry: 
tcmaxentry: 
MQTT connected: Yes
Lock firmware version: 
Lock hardware version: 
Lock paired: No
Lock valid PIN set: -
Lock has door sensor: No
Lock has keypad: No
Lock ACL (Lock): Allowed
Lock ACL (Unlock): Allowed
Lock ACL (Unlatch): Allowed
Lock ACL (Lock N Go): Allowed
Lock ACL (Lock N Go Unlatch): Allowed
Lock ACL (Full Lock): Allowed
Lock ACL (Fob Action 1): Allowed
Lock ACL (Fob Action 2): Allowed
Lock ACL (Fob Action 3): Allowed
Lock config ACL (Name): Disallowed
Lock config ACL (Latitude): Disallowed
Lock config ACL (Longitude): Disallowed
Lock config ACL (Auto Unlatch): Disallowed
Lock config ACL (Pairing enabled): Allowed
Lock config ACL (Button enabled): Allowed
Lock config ACL (LED flash enabled): Allowed
Lock config ACL (LED brightness): Disallowed
Lock config ACL (Timezone offset): Disallowed
Lock config ACL (DST mode): Disallowed
Lock config ACL (Fob Action 1): Disallowed
Lock config ACL (Fob Action 2): Disallowed
Lock config ACL (Fob Action 3): Disallowed
Lock config ACL (Single Lock): Allowed
Lock config ACL (Advertising Mode): Disallowed
Lock config ACL (Timezone ID): Disallowed
Lock config ACL (Unlocked Position Offset Degrees): Disallowed
Lock config ACL (Locked Position Offset Degrees): Disallowed
Lock config ACL (Single Locked Position Offset Degrees): Disallowed
Lock config ACL (Unlocked To Locked Transition Offset Degrees): Disallowed
Lock config ACL (Lock n Go timeout): Disallowed
Lock config ACL (Single button press action): Disallowed
Lock config ACL (Double button press action): Disallowed
Lock config ACL (Detached cylinder): Disallowed
Lock config ACL (Battery type): Disallowed
Lock config ACL (Automatic battery type detection): Disallowed
Lock config ACL (Unlatch duration): Disallowed
Lock config ACL (Auto lock timeout): Disallowed
Lock config ACL (Auto unlock disabled): Allowed
Lock config ACL (Nightmode enabled): Disallowed
Lock config ACL (Nightmode start time): Disallowed
Lock config ACL (Nightmode end time): Disallowed
Lock config ACL (Nightmode auto lock enabled): Disallowed
Lock config ACL (Nightmode auto unlock disabled): Disallowed
Lock config ACL (Nightmode immediate lock on start): Disallowed
Lock config ACL (Auto lock enabled): Allowed
Lock config ACL (Immediate auto lock enabled): Disallowed
Lock config ACL (Auto update enabled): Disallowed
Network device: Built-in Wi-Fi
BSSID of AP: 54:AF:97:1A:F2:9B
Uptime: 0 minutes
Heap: 102684
Stack watermarks: nw: 10264, nuki: 7524, pd: 10264
Restart reason FW: NetworkDeviceCriticalFailure
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

### TO REPRODUCE
Use listed device.

### EXPECTED BEHAVIOUR
Support of listed device.

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2024-07-22:
Do you have a link to the board?

iranl, 2024-07-22:
DM9051 is not supported by Nuki Hub and is not planned.
You can of course try to add support yourself and create a PR.

technyon, 2024-07-23:
Looks like an interesting board, just why don't they add a USB to serial converter. Wiring via a USB dongle usually involves a lot of trial and error.

We'd need to add driver code and configuration for this specific module, it should be possible to make it work, but unfortunately I don't have one and they're hard to get hold off here.

For reference, someone seems to have made it work:

https://github.com/AI6YP/eth01-evo/issues/2

drstil, 2024-07-24:
@technyon 

This board is one of the cheapest for having ethernet with also POE-support, so very nice! Flashing with a CP2102 adapter was straight forward. 

If there is anything I could support, please let me know! If you would be nearer to me (Germany), I could have send you one of my boards. But with Bangkok the shipping fee would be higher than the whole board. Ordering [here](https://de.aliexpress.com/item/1005006132611720.html) (without POE-Module?) is hard for you? 
But maybe I could test it for you?

iranl, 2024-07-25:
@technyon Will you be working on this? 

If so we should label this as a feature request and enhancement. Otherwise it should be closed.

Although I get the appeal of a cheap PoE solutions the DM9051NP only seems to be used in S2 (not supported because of no BLE) and C3 ESP's which imho are not really worth the time being the slowest of the supported ESP32 versions.

drstil, 2024-07-26:
@iranl In principle the software support the ESP32-C3:

> Tested stable builds are provided for the ESP32, ESP32-S3 and ESP32-C3.

Do you think the ESP32-C3 has to less performance for nuki_hub? Have you ever done performance testings so that you can say what are the minimum requirements and if it makes a difference in production with ESP32 vs ESP32-C3 (e. g. in latency of mqtt, what are the performance rich processes?)?

iranl, 2024-07-26:
I added the support for all devices other than the original ESP32 including the C3 (and wrote the text you're quoting to me).

The C3 and C6 are about 10 times as slow as the ESP32 and ESP32-S3 and they don't support running 2 tasks concurrently as they are single instead of dual core. Although they work in general they don't offer the best experience compared with the ESP32 and S3.

Ethernet chip compatibility is completely separate from this. The 8720 and W5500 are much more common for ESP's than the DM9051 and are available as separate modules that can be added to different boards. 

If PoE for a low price is desired I would look at a ESP32-S3 with an added W5500 and an Ethernet PoE to Ethernet and USB-C power splitter. That would be about 12 euros on Ali or double on Amazon.de 

As to adding the DM9051: @technyon and I are the ones doing most of the development here. I don't see a good case to work on this, so you will have to work on this yourself and create a PR or hope technyon is willing to work on this.

technyon, 2024-07-26:
I'll look into it once I get the module. I'd see how hard or easy it is to add the code, maybe it's not too hard to do. Also, we'd already have the DM9051 driver in case other boards with the chip pop up.

technyon, 2024-08-10:
Unfortunately the ESP I found wasn't the ETH01-Evo, but the older WT32-ETH01 model. I'll look again if I can find it somewhere.

drstil, 2024-08-12:
Thank you for the feedback! Hope you find the right one somewhere! 😃

iranl, 2024-08-16:
@drstil: You should be able to give these binaries a try: https://github.com/technyon/nuki_hub/actions/runs/10417095071/artifacts/1819235732

They should support Ethernet on your C3.

closed 2024-08-16 by technyon

drstil, 2024-08-16:
@iranl Thank you very much for implementing, works like a charm! 😃

iranl, 2024-08-16:
Np, credits in large part to @technyon

technyon, 2024-08-17:
My ETH01-Evo hasn't even arrived yet, and it's already working :D

PR #455 (2024-08-16): Remove old and modified libs, switch to ESPAsyncWebserver, add support for ESP32-H2 and multiple Ethernet modules
## Description:

-Modify WebCfgServer and WifiManager to use AsyncWebserver
-Ethernet changes based on the work by @technyon in branch https://github.com/technyon/nuki_hub/tree/DM9051
-Remove Arduino core Webserver and Ethernet libs

@technyon: Created this as WIP/draft PR to prevent us doing double work on related issues. Your changes for DM9051 and my changes for AsyncWebserver together should create the option to remove the old libs.

Ethernet completely untested ATM, in the pro [... cut]