technyon/nuki_hub #433: 8.35: battery voltage not provided to Home Assistant anymore

bcutter, 2024-07-23:
### PROBLEM DESCRIPTION
After updating NH from v8.27 to v8.35, the battery voltage does not arrive in Home Assistant anymore - instead the entity's state is `unknown`.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): ESP32 (LOLIN32 Lite Board V1.0 ESP-32 Rev1)
- [X] Nuki Hub binary firmware version number used: 8.35
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Provide the output of http://nukihubIP/info:
```
System Information

Nuki Hub version: 8.35
Nuki Hub build: 9596430697.1135.1 (release)
run: true
confVersion: 835
deviceId: [MASKED]
deviceIdOp: [MASKED]
nukiId: ***
nukidOp: 
mqttbroker: [MASKED]
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: true
websrvena: false
lockena: true
lockpin: 1
mqttpath: nuki_door_b
openerena: false
openerpin: 
openercont: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
maxtc: 
opmaxtc: 
enabtlprst: false
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
hassConfigUrl: 
buffsize: 
dhcpena: true
ipaddr: [MASKED]
ipsub: 255.255.255.0
ipgtw: [MASKED]
dnssrv: [MASKED]
nwhw: 1
nwwififb: false
rssipb: -1
hostname: ESP-Nuki-Hub-Door-B
nwbestrssi: true
nettmout: -1
restdisc: true
rstbcn: 60
lockStInterval: 3600
tcPerEntry: false
kpPerEntry: false
configInterval: 3600
batInterval: 7200
kpInterval: 1800
kpCntrlEnabled: true
kpInfoEnabled: false
kpPubCode: false
tcCntrlEnabled: false
tcInfoEnabled: false
cnfInfoEnabled: true
regAsApp: false
regOpnAsApp: false
nrRetry: 10
rtryDelay: 100
crdusr: ***
crdpass: ***
disnonjson: false
pubAuth: true
pubdbg: false
prdtimeout: 60
offHybrid: false
hybridTimer: 600
hybridAct: false
hybridRtry: false
hasmac: true
macb0: -41
macb1: 80
macb2: 112
latest: 8.35
tsksznetw: 
tsksznuki: 
authmaxentry: 
kpmaxentry: 
tcmaxentry: 
MQTT connected: Yes
Lock firmware version: 
Lock hardware version: 
Lock paired: Yes
Lock valid PIN set: Yes
Lock has door sensor: No
Lock has keypad: No
Lock ACL (Lock): Allowed
Lock ACL (Unlock): Allowed
Lock ACL (Unlatch): Allowed
Lock ACL (Lock N Go): Allowed
Lock ACL (Lock N Go Unlatch): Allowed
Lock ACL (Full Lock): Allowed
Lock ACL (Fob Action 1): Allowed
Lock ACL (Fob Action 2): Allowed
Lock ACL (Fob Action 3): Allowed
Lock config ACL (Name): Disallowed
Lock config ACL (Latitude): Disallowed
Lock config ACL (Longitude): Disallowed
Lock config ACL (Auto Unlatch): Disallowed
Lock config ACL (Pairing enabled): Allowed
Lock config ACL (Button enabled): Allowed
Lock config ACL (LED flash enabled): Allowed
Lock config ACL (LED brightness): Allowed
Lock config ACL (Timezone offset): Disallowed
Lock config ACL (DST mode): Disallowed
Lock config ACL (Fob Action 1): Disallowed
Lock config ACL (Fob Action 2): Disallowed
Lock config ACL (Fob Action 3): Disallowed
Lock config ACL (Single Lock): Allowed
Lock config ACL (Advertising Mode): Disallowed
Lock config ACL (Timezone ID): Disallowed
Lock config ACL (Unlocked Position Offset Degrees): Disallowed
Lock config ACL (Locked Position Offset Degrees): Disallowed
Lock config ACL (Single Locked Position Offset Degrees): Disallowed
Lock config ACL (Unlocked To Locked Transition Offset Degrees): Disallowed
Lock config ACL (Lock n Go timeout): Allowed
Lock config ACL (Single button press action): Allowed
Lock config ACL (Double button press action): Allowed
Lock config ACL (Detached cylinder): Disallowed
Lock config ACL (Battery type): Disallowed
Lock config ACL (Automatic battery type detection): Disallowed
Lock config ACL (Unlatch duration): Allowed
Lock config ACL (Auto lock timeout): Disallowed
Lock config ACL (Auto unlock disabled): Disallowed
Lock config ACL (Nightmode enabled): Disallowed
Lock config ACL (Nightmode start time): Disallowed
Lock config ACL (Nightmode end time): Disallowed
Lock config ACL (Nightmode auto lock enabled): Disallowed
Lock config ACL (Nightmode auto unlock disabled): Disallowed
Lock config ACL (Nightmode immediate lock on start): Disallowed
Lock config ACL (Auto lock enabled): Disallowed
Lock config ACL (Immediate auto lock enabled): Disallowed
Lock config ACL (Auto update enabled): Disallowed
Network device: Built-in Wi-Fi
BSSID of AP: [MASKED]
Uptime: 0 minutes
Heap: 76228
Stack watermarks: nw: 7096, nuki: 5536, pd: 7096
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

### TO REPRODUCE
Update NH to 8.35.

### EXPECTED BEHAVIOUR
`sensor.lock_door_b_battery_voltage` is available and gives the actual voltage.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
There's another major issue since NH has been updated to 8.35. Seperate to this, but I'll link it here once created.

MQTT Nuki Hub topic: information is available (captured from the Nuki Lock):
[image]

MQTT `/homeassistant/sensor/[nuki-id]/battery_voltage` topic:
```
{"dev":{"ids":["nuki_XXXXXXXX"],"mf":"Nuki","mdl":"SmartLock","name":"Door B"},"~":"nuki_door_b","name":"Battery voltage","unique_id":"XXXXXXXX_battery_voltage","dev_cla":"voltage","stat_t":"~/battery/advancedJson","stat_cla":"measurement","ent_cat":"diagnostic","avty":{"t":"nuki_door_b/maintenance/mqttConnectionState"},"unit_of_meas":"V","val_tpl":"{{value_json.level}}"}
```

Does that look correct? So why is it missing in HA?

I also wiped both topics once (one by another step by step) and restarted NH, which made the topics reappear but the entity in HA is still `unknown`.

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-07-23:
Duplicate of #414 
Already fixed in#407

closed 2024-07-23 by iranl

bcutter, 2024-07-23:
@iranl That's basically good news, thanks a lot for your quick response.

But pardon me as I'm not familiar with tracking PRs: is https://github.com/technyon/nuki_hub/pull/407 already merged into a release I can use? I don't see a 8.36 or another stable release :-(

iranl, 2024-07-23:
9.00 beta

bcutter, 2024-07-23:
> 9.00 beta

How to get it and didn't you mention bluetooth issues in that 9.00 beta earlier this day?

I'm willing to test but I fear switching one issue with another :-)

mundschenk-at, 2024-07-23:
Unlikely, but no-one can guarantee 🤷

bcutter, 2024-07-23:
> Unlikely, but no-one can guarantee 🤷

I could certainly downgrade, couldn't I?

1. I don't understand why issues are solved when a PR fixing that issue has not been made available in a stable release yet. If it would have been listed as an open issue I would have spotted it when checking all open issues during planning and before executing my update orgy.
2. On the other hand at https://github.com/technyon/nuki_hub?tab=readme-ov-file#upgrading-using-over-the-air-update-ota I saw notes about 8.36 which needs a flash
> Note for users upgrading from Nuki Hub 8.35 or lower:
Updating to version 8.36 requires a change to the partition table of the ESP32.
Please follow the instructions for the [First time installation](https://github.com/technyon/nuki_hub?tab=readme-ov-file#first-time-installation) once when updating to Nuki Hub 8.36 from an earlier version.
Your settings will not be affected when updating using the above instructions (do not select erase device when updating using Webflash).
--> Is 9.0 beta also affected by this?
3. I still have ZERO information on how to get the beta version. I never had to until now and the docs do not contain a single word unfortunately. In short: help appreciated.

mundschenk-at, 2024-07-23:
> > Unlikely, but no-one can guarantee 🤷
> 
> I could certainly downgrade, couldn't I?

Yes, by serial flashing (not OTA).

> 1. I don't understand why issues are solved when a PR fixing that issue has not been made available in a stable release yet. If it would have been listed as an open issue I would have spotted it when checking all open issues during planning and before executing my update orgy.

Because the issue is fixed in the code, even if there has not been a non-beta release yet. 

> 2. On the other hand at https://github.com/technyon/nuki_hub?tab=readme-ov-file#upgrading-using-over-the-air-update-ota I saw notes about 8.36 which needs a flash
> 
> > Note for users upgrading from Nuki Hub 8.35 or lower:
> > Updating to version 8.36 requires a change to the partition table of the ESP32.
> > Please follow the instructions for the [First time installation](https://github.com/technyon/nuki_hub?tab=readme-ov-file#first-time-installation) once when updating to Nuki Hub 8.36 from an earlier version.
> > Your settings will not be affected when updating using the above instructions (do not select erase device when updating using Webflash).
> 
> --> **Is 9.0 beta also affected by this?** So a downgrade to 8.35 is impossible (or much more complicated)?

8.36 and 9.0 are the same thing, just renumbered. So yes, you would need to do a serial flash instead of an OTA update.

> 3. I still have ZERO information on how to get the beta version. I never had to until now and the docs do not contain a single word unfortunately. In short: help appreciated.

I am not sure if there is a better way yet (@iranl maybe working on it), but you can get the built assets from the [Actions](https://github.com/technyon/nuki_hub/actions) tab (currently https://github.com/technyon/nuki_hub/actions/runs/10051235124 has the latest binaries).

bcutter, 2024-07-23:
I decided to work around this (rather minor) issue and live with the missing battery voltage until a stable release (8.36) fixing this arrives.

My real serious issue since 8.35 is with https://github.com/technyon/nuki_hub/issues/434 🥵

bcutter, 2024-10-19:
Fixed by updating to NH v9.01. ...which was quite a pain:

(see https://github.com/technyon/nuki_hub/issues/434#issuecomment-2424203899)