technyon/nuki_hub #434: 8.35: config changes triggered in Home Assistant triggers complete MQTT reload for that device (renders all entities unavailable)

bcutter, 2024-07-23:
### PROBLEM DESCRIPTION
After updating NH from v8.27 to v8.35, when one config (e. g. `switch.lock_door_b_led_enabled`) is changed, first the switch immediately turns back and after few seconds all entities are unavailable - it takes 30 seconds for them to come back. Nuki Hub seems to NOT restart during that time according to the `sensor.lock_door_b_uptime_nukihub` sensor.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): ESP32 (LOLIN32 Lite Board V1.0 ESP-32 Rev1)
- [X] Nuki Hub binary firmware version number used: 8.35
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Provide the output of http://nukihubIP/info:
```
System Information

Nuki Hub version: 8.35
Nuki Hub build: 9596430697.1135.1 (release)
run: true
confVersion: 835
deviceId: [MASKED]
deviceIdOp: [MASKED]
nukiId: ***
nukidOp: 
mqttbroker: [MASKED]
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: true
websrvena: false
lockena: true
lockpin: 1
mqttpath: nuki_door_b
openerena: false
openerpin: 
openercont: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
maxtc: 
opmaxtc: 
enabtlprst: false
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
hassConfigUrl: 
buffsize: 
dhcpena: true
ipaddr: [MASKED]
ipsub: 255.255.255.0
ipgtw: [MASKED]
dnssrv: [MASKED]
nwhw: 1
nwwififb: false
rssipb: -1
hostname: ESP-Nuki-Hub-Door-B
nwbestrssi: true
nettmout: -1
restdisc: true
rstbcn: 60
lockStInterval: 3600
tcPerEntry: false
kpPerEntry: false
configInterval: 3600
batInterval: 7200
kpInterval: 1800
kpCntrlEnabled: true
kpInfoEnabled: false
kpPubCode: false
tcCntrlEnabled: false
tcInfoEnabled: false
cnfInfoEnabled: true
regAsApp: false
regOpnAsApp: false
nrRetry: 10
rtryDelay: 100
crdusr: ***
crdpass: ***
disnonjson: false
pubAuth: true
pubdbg: false
prdtimeout: 60
offHybrid: false
hybridTimer: 600
hybridAct: false
hybridRtry: false
hasmac: true
macb0: -41
macb1: 80
macb2: 112
latest: 8.35
tsksznetw: 
tsksznuki: 
authmaxentry: 
kpmaxentry: 
tcmaxentry: 
MQTT connected: Yes
Lock firmware version: 
Lock hardware version: 
Lock paired: Yes
Lock valid PIN set: Yes
Lock has door sensor: No
Lock has keypad: No
Lock ACL (Lock): Allowed
Lock ACL (Unlock): Allowed
Lock ACL (Unlatch): Allowed
Lock ACL (Lock N Go): Allowed
Lock ACL (Lock N Go Unlatch): Allowed
Lock ACL (Full Lock): Allowed
Lock ACL (Fob Action 1): Allowed
Lock ACL (Fob Action 2): Allowed
Lock ACL (Fob Action 3): Allowed
Lock config ACL (Name): Disallowed
Lock config ACL (Latitude): Disallowed
Lock config ACL (Longitude): Disallowed
Lock config ACL (Auto Unlatch): Disallowed
Lock config ACL (Pairing enabled): Allowed
Lock config ACL (Button enabled): Allowed
Lock config ACL (LED flash enabled): Allowed
Lock config ACL (LED brightness): Allowed
Lock config ACL (Timezone offset): Disallowed
Lock config ACL (DST mode): Disallowed
Lock config ACL (Fob Action 1): Disallowed
Lock config ACL (Fob Action 2): Disallowed
Lock config ACL (Fob Action 3): Disallowed
Lock config ACL (Single Lock): Allowed
Lock config ACL (Advertising Mode): Disallowed
Lock config ACL (Timezone ID): Disallowed
Lock config ACL (Unlocked Position Offset Degrees): Disallowed
Lock config ACL (Locked Position Offset Degrees): Disallowed
Lock config ACL (Single Locked Position Offset Degrees): Disallowed
Lock config ACL (Unlocked To Locked Transition Offset Degrees): Disallowed
Lock config ACL (Lock n Go timeout): Allowed
Lock config ACL (Single button press action): Allowed
Lock config ACL (Double button press action): Allowed
Lock config ACL (Detached cylinder): Disallowed
Lock config ACL (Battery type): Disallowed
Lock config ACL (Automatic battery type detection): Disallowed
Lock config ACL (Unlatch duration): Allowed
Lock config ACL (Auto lock timeout): Disallowed
Lock config ACL (Auto unlock disabled): Disallowed
Lock config ACL (Nightmode enabled): Disallowed
Lock config ACL (Nightmode start time): Disallowed
Lock config ACL (Nightmode end time): Disallowed
Lock config ACL (Nightmode auto lock enabled): Disallowed
Lock config ACL (Nightmode auto unlock disabled): Disallowed
Lock config ACL (Nightmode immediate lock on start): Disallowed
Lock config ACL (Auto lock enabled): Disallowed
Lock config ACL (Immediate auto lock enabled): Disallowed
Lock config ACL (Auto update enabled): Disallowed
Network device: Built-in Wi-Fi
BSSID of AP: [MASKED]
Uptime: 0 minutes
Heap: 76228
Stack watermarks: nw: 7096, nuki: 5536, pd: 7096
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

### TO REPRODUCE
Update NH to 8.35. Make a config change through any Home Assistant configuration entity.

### EXPECTED BEHAVIOUR
Change is performed immediately and the changed entity updates within few seconds, all other entities are not affected and no entity renders `unavailable` for 30 seconds.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
This makes controlling a Nuki lock from Home Assistant using Nuki Hub pretty useless.
- I can directly compare the behaviour to another Nuki Hub still running NH v8.27 - which works flawlessly when doing config changes.
- I enabled MQTT integration debug logging and performed a config change for that Nuki Hub device and hat a look at the log afterwards but can't really find anything suspicous.

**(Please, remember to close the issue when the problem has been addressed)**

mundschenk-at, 2024-07-23:
Can't reproduce with build `9909162422.1186.1` (an early `8.36`/`9.0` build). But how often do you change your config settings that this seriously affects your use of the lock in HA?

bcutter, 2024-07-23:
Well it's more that it doesn't really give confidence when the stability in Home Assistant is so massively negatively affected by an update of a 3rd party component (from outside HA trough MQTT). I've never seen this before in the last years - and this issue makes using HA to configure Nuki locks pretty useless.

What else can I provide or test?

bcutter, 2024-07-23:
I believe this is not really acceptable:

https://github.com/user-attachments/assets/06ef174d-ebb8-467c-a8be-efbb4cb7df54

Do you want me to update step by step from 8.27 up to 8.35 with a stop-by at 8.28, 8.29 etc.? I have a second Nuki Hub on 8.27. But this way we'd also only know which release introduced this behaviour - is that helpful at all?

Or any other ideas?

mundschenk-at, 2024-07-23:
At this point in the release cycle, I don't think it is worthwhile to put more energy into investigating this issue. If you can reproduce it with 9.0 when that is released, it will become another matter (but I don't think you will see issue with 9.0).

bcutter, 2024-07-23:
More facts after testing:
1. reloading the whole MQTT integration takes a long time until the NH 8.35 updated device shows values in normal (not grayed out) color again.
2. after that, most entities have NO valid state at all, most are `unknown` and therefore configuration interface is strange too:

[image]

[image]

3. I need to manually press the buttons `Query battery` and `Query config` etc. and wait quite a while (1 to 2 minutes) to get status back for many entities, however, many others stay in `unknown` state now:
- sensor.lock_door_b_nuki_hub_build
- sensor.lock_door_b_nuki_hub_ip
- sensor.lock_door_b_nuki_hub_latest
- update.lock_door_b_nuki_hub_firmware_update
- sensor.lock_door_b_rolling_authorization_log

Something's **fundamentally broken** here. Worst update experience so far :-)

And the fact no one else ran into this makes me feel very uncomfortable to just "sit and wait for 9.0". I hope this is reasonable.

mundschenk-at, 2024-07-23:
You can try cleaning up the `nuki` and related config topics in the broker while NH is switched off. There might be some retained messages that wreak havoc with HA discovery. Other than that, I really don't know what to tell you. You might need to renew your support contract with @technyon and @iranl 🤷

bcutter, 2024-07-23:
I tend to ignore the passive funny support contract comments :-)

> You can try cleaning up the `nuki` and related config topics in the broker while NH is switched off. There might be some retained messages that wreak havoc with HA discovery.

How would I do that?

1. Power off Nuki Hub
2. Start MQTT Explorer and delete all topics from
- `/nuki_door_b/`
- `/homeassistant/sensor/nuki-id/*`
- same as line before for `lock`, `binary`, `switch`, `number`, `update`, `button`, `update` and `select` instead of `sensor`
3. Power up Nuki Hub again - and pray everything is recreated

**Is that correct?**

As it's my productive system I really don't want to loose any data (settings / references, custom icons or I don't know what else gets removed when basically "purging" the MQTT).

mundschenk-at, 2024-07-23:
> I tend to ignore the passive funny support contract comments :-)

Maybe don't label things as "unacceptable" or "worst update experience" then?

> How would I do that?
>
> 1. Power off Nuki Hub
> 2. Start MQTT Explorer and delete all topics from
>
> * `/nuki_door_b/`
> * `/homeassistant/sensor/nuki-id/*`
> * same as line before for `lock`, `binary`, [...]

Yes.

> As it's my productive system I really don't want to loose any data (settings / references, custom icons or I don't know what else gets removed when basically "purging" the MQTT).

If you want to make sure that HA does not lose any entity customizations, you can shut down HA or disable the MQTT integration while you clean up the retained messages. It's unlikely to be necessary, but if you want to go really sure, that's the way to ensure that HA can't do anything funky. 

Anyway, MQTT messages (even retained ones) are only a kind of message bus, so they are always ephemeral.

bcutter, 2024-07-23:
Unacceptable refers to the situation and few comments like "just wait for the glorious 9.0 release which is unknown when to arrive or if it will fix things at all" :-D :-D :-D

Back to business. Additional notes:

- The missing entities (still `unknown`) only switch to their actual values once I restart the Nuki Hub. Immediately.
- Restarting the Mosquitto broker changes nothing. Reloading the MQTT integration changes nothing. Restarting Home Assistant changes nothing.

**That plus the fact a Nuki Hub restart at least triggers a value update from MQTT to Home Assistant leaves me quite wondering - and more or less clearly pointing at Nuki Hub - which also is the only thing changed (likely to have introduced this).**

I need to decice which way I walk tomorrow:
a) purging MQTT content as described above
b) downgrade Nuki Hub from 8.35 step by step to see if/when this resolves (at latest at 8.26 I guess)
c) upgrade my second Nuki Hub from 8.26 to 8.35 in minor steps and see if and when the same behaviour starts
d) test 9.0 beta release (I could not find the compiled esp32 `.bin` yet)
e) wait for 9.0 stable / 8.36 release

Plenty of options basically. Unfortunately I already spent half of the day only for Nuki Hub stuff. Let's call it a day for today.

---

Appendix: restarting the MQTT broker surprisingly triggered the update entity:
[image]

The MQTT topic looks like this:
[image]

Where the hell is 8.26 coming from?!? It never got updated since the update today (8.26 to 8.35).
**That's not a Nuki Hub bug (but speaks more for option a) above), isn't it?**

mundschenk-at, 2024-07-24:
Well, nobody told you to do those additional steps, so please don't complain that they took time. The mysterious value indicates that you have old state messages retained by the broker and should clear out those topics (I have noticed a similar thing on my installation, it seems the `retained` flag was changed sometime in the past for some topics).

technyon, 2024-07-24:
@bcutter Please acknowledge that everyone here is doing their best to help you. What Nuki Hub does is complex on many levels, and problems can occur. HA auto discovery in itself is complicated enough already to potentially cause problems. Every release is tested as much as we can without being a commercial project. I know it can be frustrating if things don't work, but that's in the nature of setting up your own smart home.

I can't be all that helpful since I'm not a HA user, but maybe deleting the MQTT nodes related to Nuki Hub sounds reasonable, since a fresh Nuki Hub setup doesn't show these problems. Maybe export customizations you made before so it's easy to add them later.

bcutter, 2024-07-24:
I'll test to purge the MQTT content (Option a) from above) and turn back.

The fact that after restarting HA nearly all entities are unavailable/unknown reminded me of the very root issue which blocked me (with old HA setup before 2023.8) from updating NH to 8.27 and beyond:

https://github.com/technyon/nuki_hub/issues/227

But that time only the lock entity was affected - and the instability issue I see currently wasn't a thing either. So just thinking out loud - as mentioned I will report back after the MQTT topics cleanup test.

bcutter, 2024-07-24:
After going through https://github.com/technyon/nuki_hub/issues/434#issuecomment-2246236813 (with disabling the MQTT before and enabling it again after the process), nothing has changed:

- all entities' values still `unkown` after MQTT initialization in HA, manual update necessary to trigger entity update in HA by a) restarting NH + b) `Query config` + c) `Query battery`
- making a config change in HA renders all entities of the NH device unavailable for ~ 30 seconds

So a) from this list did do nothing unfortunately.

> a) purging MQTT content as described above
b) downgrade Nuki Hub from 8.35 step by step to see if/when this resolves (at latest at 8.26 I guess)
c) upgrade my second Nuki Hub from 8.26 to 8.35 in minor steps and see if and when the same behaviour starts
d) test 9.0 beta release (I could not find the compiled esp32 .bin yet)
e) wait for 9.0 stable / 8.36 release

Which option do you recommend now next?

bcutter, 2024-07-24:
OK I continued with

>  b) downgrade Nuki Hub from 8.35 step by step to see if/when this resolves (at latest at 8.26 I guess)

I downgraded NH from 8.35 to 8.34. Results:

a) https://github.com/technyon/nuki_hub/issues/433 is fixed (or not yet introduced), battery voltage is back
[image]

b) behaviour regarding this issue is at least improved: no full device entities reset for ~ 30 seconds, instead it just flickers for a moment and after few seconds the command seems to be executed and confirmed back (by MQTT to HA). That's the way it looks and feels at least:

https://github.com/user-attachments/assets/ddfae711-96b6-429c-98c7-badd8ca4e86a

Not yet good (not as fast and direct as with 8.26), but much more stable.

So no idea what introduced (or increased the outcome) of this issue in https://github.com/technyon/nuki_hub/releases/tag/8.35, but 8.34 seems to be the better choice for now.

Here's a config/system information comparison of v8.35 (before the downgrade) on the left and v8.34 (after the downgrade) on the right:

[image]

1. Do you spot any difference which might in combination with 8.35 changes look "suspicous"?
2. Can someone provide a working `.bin` for ESP32 of the 8.36 / 9.0 beta? I want to test if this issue only is/was 8.35 release specific or if we continue to carry that around in future releases.

technyon, 2024-07-25:
Also, here's a link to the latest beta binaries:

https://github.com/technyon/nuki_hub/actions/runs/10086733497

Note that you'll need to flash via serial because the partition scheme has changed. Setting will not be overwritten if you don't do a factory reset.

bcutter, 2024-07-25:
> Expected behaviour when changing a setting in Home Assistant:
> 
>     * `nuki/configuration/action` changes (e.g. `{ "ledEnabled": "1" }`) and changes back to `--`
> 
>     * `nuki/configuration/commandResult` changes  (e.g. `{ "ledEnabled": "success", "general": "success" }`)
> 
>     * `nuki/configuration/basicJson` and `nukihub/configuration/advancedJson` changes reflecting the changed setting
> 
>     * The setting in Home Assistant changes to the correct state
> 
> 
> Also keep an eye on `nuki/maintenance/mqttConnectionState`, this should remain `online` at all times
> 
> Please report back with any differences between 8.34 and 8.35 while observing the above.

### 8.34 with Nuki Hub number 1:
1. HA: make a change to `switch.lock_door_X_led_enabled`
2. ✅ Immediately: `nuki/configuration/action` changes to `{ "ledEnabled": "1" }` and changes back to `--`
3. ✅ Shortly after 2: `nuki/configuration/commandResult` changes to `{"ledEnabled":"success","general":"success"}`
4. ✅ Shortly after 2: `nuki/configuration/basicJson` changes (`ledEnabled`and `currentTime`), `nukihub/configuration/advancedJson` remains unchanged (as the tested LED is not stored here)
5. ✅ HA: LED switch changes to desired state - whole process takes roughly **3 seconds**
6. ✅ `nuki/maintenance/mqttConnectionState` remains `online` during the whole process

### 8.35 with Nuki Hub number 2:
1. HA: make a change to `switch.lock_door_X_led_enabled`
2. ✅ Immediately: `nuki/configuration/action` changes to `{ "ledEnabled": "1" }` (nothing more!)
3. ❗Nothing happens for roughly 23 seconds.
4. ❗`nuki/maintenance/mqttConnectionState` switches from `online` to `offline` and all entities of the device turn `unavailable` in Home Assistant.
5. ❗18 seconds where nothing happens (neither in MQTT nor in HA) - seems like the Nuki Hub is doing "something".
6. ❗Now kind of a "big flash update" happens in MQTT where many topics get updated within one second, among others:
* `nuki/maintenance/mqttConnectionState` switches from `offline` to `online`
* `nuki/configuration/action` changes back to `--`
* `nuki/configuration/basicJson` gets updated
* `nuki/configuration/commandResult` gets updated
* at the same time in HA: all entities are available again and LED switch changes to desired state ✅ - whole process takes roughly **44 seconds**.

If I wouldn't be concerned bout few visible details (IDs, lock names etc.) I would share a video recording of HA on the left and MQTT Explorer on the right side. Along with the time code you can see exactly what I described above in plain text.

So obviously something happens in Nuki Hub because of the
* big interruption when processing the command, right from the beginning
* `nuki/maintenance/mqttConnectionState` going `offline`
* huge amount of time where nothing happens in MQTT (I guess the Nuki Hub is doing something)

**How to (remotely) gather logs of the Nuki Hub/ESP itself, would that be helpful?** I could compare the 8.34 NH with the 8.35 NH that way.

---

> Also, here's a link to the latest beta binaries:
> 
> https://github.com/technyon/nuki_hub/actions/runs/10086733497
> 
> Note that you'll need to flash via serial because the partition scheme has changed.

Oh it's a nightly, I thought it's a branded beta - doesn't matter. **Would I be able to downgrade** (probably doing a serial flash again) to 8.35 (or 8.34 which is kind of stable instead of the horrible 8.35) **again without loosing data?**

iranl, 2024-07-25:
Thanks for the update. Based on your report I found the offending code:

https://github.com/technyon/nuki_hub/blob/4d4edaf126f3b8c6c58288bb576d164ee88b1132/src/NukiWrapper.cpp#L1158

Setting this value to true causes a delay of up to 30 seconds in processing the next command. It should have been set when another BLE command was running but in this case was always set when processing config changes.

This code was added as part of #389 and removed as part of #407.
8.35 was released in between these and contains the problematic code.

So this is already fixed as part of 9.00.

You would be able to downgrade to 8.35 using a serial flash if you update to a 9.00 pre-release without losing data.
9.00 will also allow you to export and import all your configuration and pairing data.

The best binary to be on right now if you wan't to test 9.00 would be https://github.com/technyon/nuki_hub/actions/runs/10095506815 imho as this includes latest fixes for BLE connectivity. I also run this build myself on my non-development ESPs.

bcutter, 2024-08-03:
OK I'm sitting here for half an hour trying to figure out how to flash the https://github.com/technyon/nuki_hub/actions/runs/10095506815/artifacts/1739771030.

1. https://technyon.github.io/nuki_hub/ does not let me choose a specific version, instead, it just wants
[image]

2. Espressif Flash Download Tools is very complicated (right the start screen: factory? developer? what?!?)

3. esptool: following the `how-to-flash.txt` the commands (`--before default_reset --after hard_reset`) indicate, I will loose my configuration

So finally, here's the question: **what is the most convenient way to flash that beta version without loosing the whole configuration?** That information will be needed for the release notes anyway, won't it...

iranl, 2024-08-03:
1. Webflash will only be updated when 9.0 is officialy released. We might do beta and/or dev version over webflash in the future.

3. No default and/or hard reset don't erase settings

bcutter, 2024-08-03:
Sad to hear (web flash).

So once again:

> **what is the most convenient way to flash that beta version without loosing the whole configuration?**

And when will 9.0 make it to a stable release? I need to decide if I really want to spent more hours on this or sit and wait with 8.34.

iranl, 2024-08-03:
I'd say remain on 8.34 and use webflash when 9.0 is released.

9.0 will be released when we feel it is done, could be in a week, could be in a month depending on how many bugs are found.

bcutter, 2024-08-03:
> I'd say remain on 8.34 and use webflash when 9.0 is released.

That's exactly what I decided to do 15 minutes ago.

Looking at https://github.com/technyon/nuki_hub/milestone/3 it seems there's not much to do left, but I guess that's not an arithmetic thing (like "only two issues left, let's push it next week" :-)).

I'll update to stable 9.0 once available by webflash, test things quickly and turn back here to hopefully report "issue is fixed, thanks". Until then it's fine for me to leave this issue open, thank you.

iranl, 2024-08-04:
Closing with the release of 9.0.

closed 2024-08-04 by iranl

bcutter, 2024-08-04:
I've seen the 9.0 stable update (already discovered by one of two NHs).

Unfortunately because of the breaking change

> The presence detection feature has been removed.

I first need to try migrating two presence detection scenarios away from NH before I can finally test 9.0 release.

Because of this removal it might be one now has to double the devices (one ESP for NH, one for presence detection), which also doubles maintenance, power sockets, power costs, ... and increases complexity. Understandable decision (based on the issues presence detection created according to the release notes because for me it was working very very good) but from a users perspective not really sustainable.

Off-topic over, I'll turn back with 9.0 test experience feedback once presence use-case drop is solved.

bcutter, 2024-10-19:
Fixed by updating to NH v9.01. ...which was quite a pain:

Flashing DID erase all the data (of course I did NOT tick the "erase" box)! Therefore needed:
1) Wi-Fi setup from scratch
2) Re-pairing the Nuki lock, setting the PIN
3) Full remaining config from scratch

What a pain in the evening guys. Updating two Nuki Hubs took me two hours.

That's not what I expected from the release notes, the need for a full re-configuration is not outlined there:
https://github.com/technyon/nuki_hub/releases/tag/9.0

At least now we seem to have a substantial config export/import (and therefore a hopefully reliable backup) function. Should minimize future "do everything from scratch" times…