technyon/nuki_hub #447: Disable the "Unlatch" discovery

Balooforever, 2024-08-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hi everyone,

Thank a lot for your works !

I'm beginner with Nuki Hub 9.0 and i have only one little problem. I can't disable the "Unlatch" discovery to my home assistant. My door can't do this and i want to disable the button on home assistant.

How i can do it ?

I have "unceck" "Unlatch" in nuki hub but nothing happen..

Thank a lot :)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [X] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): M5 Atom Light
- [X] Nuki Hub binary firmware version number used: 9.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info:
```

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Expose lock to home assistant

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-08-06:
Edit your comment with the ouput or http://nukihubip/info or your issue will be closed again in accordance with the Contributing policy.
Do not open new issues but edit your earlier comment.

Also provide information of what exact home asissistant entity you expect to be removed with it's entity id and/or screenshots of the place(s) you expect this to be removed

Balooforever, 2024-08-06:
I don't understand why add sur "info" with this issue but, OK, i have add it, contributing policy is little hardcore oO

I want to remove this in purple : 

[image]

Nuki hub expose "Unlatch", but my door doesn't have unlatch, i doesn't want it to home assistant.

iranl, 2024-08-06:
The contributing policy explains why we need the template filled.

What you are asking is not possible in 9.0

I've added it to PR #445, once this is merged you can update to a development version using the firmware update page on the web configurator or wait for release/beta of 9.01

Balooforever, 2024-08-06:
Thank, no urgency, wainting release, thx :)

PR #445 (2024-08-11): OTA and Debug improvements
## Description:

- Automatically commit debug builds to binary branch
- Allow updating to debug builds by manually browsing to http://NUKIHUBIP/otadebug (no link in the menu)
- Always publish restart reasons to MQTT (also when debugon is not enabled) on boot and create a Home Assistant sensor for these topics
- Track latest version based on currently running version on Web configurator page and MQTT update
-- If latest release >= current version (irrespective of release/beta/master): latest release
-- else if currently running version is beta: latest beta
-- else if  [... cut]
closed 2024-08-11 by technyon