technyon/nuki_hub #453: Keypad 2.0  Keypad status does not show activity

razserv2010, 2024-08-12:
### PROBLEM DESCRIPTION

Keypad status stay empty and does not show activity

> System Information
Nuki Hub version: 9.00
Nuki Hub build: 10235422676.1.1
Nuki Hub build type: Release
run: true
confVersion: 900
nukiId: ***
nukidOp: 
mqttbroker: 192.168.0.***
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: false
websrvena: false
lockena: true
lockpin: 1
mqttpath: nuki
openerena: false
openerpin: 
openercont: false
mqttoppath: 
maxkpad: 3
opmaxkpad: 
maxtc: 
opmaxtc: 
enabtlprst: false
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
hassConfigUrl: 
buffsize: 
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 5
nwwififb: false
rssipb: 60
hostname: nukihub
nwbestrssi: false
nettmout: 30
restdisc: true
rstbcn: -1
lockStInterval: 1800
tcPerEntry: true
kpPerEntry: true
configInterval: 3600
batInterval: 1800
kpInterval: 1
kpCntrlEnabled: true
kpInfoEnabled: true
kpPubCode: true
tcCntrlEnabled: true
tcInfoEnabled: true
cnfInfoEnabled: true
regAsApp: false
regOpnAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: 
crdpass: 
disnonjson: false
pubAuth: false
pubdbg: false
prdtimeout: 60
offHybrid: false
hybridTimer: 600
hybridAct: false
hybridRtry: false
hasmac: false
macb0: 
macb1: 
macb2: 
latest: 8.35
tsksznetw: 
tsksznuki: 
authmaxentry: 
kpmaxentry: 
tcmaxentry: 
updMqtt: false
showSecr: false
bleTxPwr: 9
recNtwMqttDis: false
MQTT connected: Yes
Lock firmware version: 4.2.8
Lock hardware version: 8.0
Lock paired: Yes
Lock valid PIN set: Yes
Lock has door sensor: No
Lock has keypad: Yes
Lock ACL (Lock): Allowed
Lock ACL (Unlock): Allowed
Lock ACL (Unlatch): Allowed
Lock ACL (Lock N Go): Allowed
Lock ACL (Lock N Go Unlatch): Allowed
Lock ACL (Full Lock): Allowed
Lock ACL (Fob Action 1): Allowed
Lock ACL (Fob Action 2): Allowed
Lock ACL (Fob Action 3): Allowed
Lock config ACL (Name): Disallowed
Lock config ACL (Latitude): Disallowed
Lock config ACL (Longitude): Disallowed
Lock config ACL (Auto Unlatch): Allowed
Lock config ACL (Pairing enabled): Allowed
Lock config ACL (Button enabled): Allowed
Lock config ACL (LED flash enabled): Allowed
Lock config ACL (LED brightness): Allowed
Lock config ACL (Timezone offset): Disallowed
Lock config ACL (DST mode): Disallowed
Lock config ACL (Fob Action 1): Disallowed
Lock config ACL (Fob Action 2): Disallowed
Lock config ACL (Fob Action 3): Disallowed
Lock config ACL (Single Lock): Allowed
Lock config ACL (Advertising Mode): Disallowed
Lock config ACL (Timezone ID): Disallowed
Lock config ACL (Unlocked Position Offset Degrees): Disallowed
Lock config ACL (Locked Position Offset Degrees): Disallowed
Lock config ACL (Single Locked Position Offset Degrees): Disallowed
Lock config ACL (Unlocked To Locked Transition Offset Degrees): Disallowed
Lock config ACL (Lock n Go timeout): Disallowed
Lock config ACL (Single button press action): Allowed
Lock config ACL (Double button press action): Allowed
Lock config ACL (Detached cylinder): Disallowed
Lock config ACL (Battery type): Allowed
Lock config ACL (Automatic battery type detection): Allowed
Lock config ACL (Unlatch duration): Allowed
Lock config ACL (Auto lock timeout): Allowed
Lock config ACL (Auto unlock disabled): Allowed
Lock config ACL (Nightmode enabled): Disallowed
Lock config ACL (Nightmode start time): Disallowed
Lock config ACL (Nightmode end time): Disallowed
Lock config ACL (Nightmode auto lock enabled): Disallowed
Lock config ACL (Nightmode auto unlock disabled): Disallowed
Lock config ACL (Nightmode immediate lock on start): Disallowed
Lock config ACL (Auto lock enabled): Allowed
Lock config ACL (Immediate auto lock enabled): Disallowed
Lock config ACL (Auto update enabled): Allowed
Network device: WT32-ETH01
Uptime: 10 minutes
Heap: 92144
Restart reason FW: ConfigurationUpdated
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [ ] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ ] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ ] Device used (e.g., ESP32-S3): _____
- [ ] Nuki Hub binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info:
```

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-08-13:
This topic will only be filled if you have  enabled `Publish authorization log`.
Based on your info page you have this option disabled.

Also note that this home assistant sensor in general does not really over too much usefull information.
Most usefull information about keypad usage (among others) will be in the rolling authorization log sensor (check the home assistant attributes of this sensor after enabling `Publish authorization log`).

**Note**: Having "Query interval keypad (seconds)" at 1 is not a good idea, This will lead to very high battery drain as Nuki Hub will constantly be quering your keypad for users. These values will not change unless you updated your keypad settings/users through the official Nuki app. When updating keypad users through Nuki Hub it will auto query after updating.

razserv2010, 2024-08-13:
> Keypad status does not show activity.

I enabled the `Publish authorization log`
But still does not show activity

iranl, 2024-08-14:
Are you looking at the right sensor?

As I already wrote the Home Assistant `Keypad status` sensor does not generally show usefull information.

Look at the `Rolling authorization log` sensor in Home Assistant. It's value should be a number (of the last auth log entry) and it's properties should contain further information. Every auth log entry should be visible in the sensor history.

razserv2010, 2024-08-14:
> Are you looking at the right sensor?
>
> As I already wrote the Home Assistant `Keypad status` sensor does not generally show usefull information.
>
> Look at the `Rolling authorization log` sensor in [...]

There is only one such sensor, Keypad status.
But through the `Rolling authorization log`  I saw and it satisfies me, thanks

closed 2024-08-14 by iranl