technyon/nuki_hub #454: 9.0 boot loop

AlbrechtL, 2024-08-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After flashing the version 9.0 the ESP stays in a boot loop. I have a ESP32 D1 Mini Nodemcu. The old version 8.35 run well.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [ ] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ ] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ ] Device used (e.g., ESP32-S3): _____
- [ ] Nuki Hub binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info:
```

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Power on ;-)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Shall boot

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Serial output
```
W (1099) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
�ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7116
ho 0 tail 12 room 4
load:0x40078000,len:15612
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3868
entry 0x4008064c
I (64) boot: ESP-IDF 5.1.4.240629 2nd stage bootloader
I (64) boot: compile time Aug  4 2024 10:21:46
I (64) boot: Multicore bootloader
I (69) boot: chip revision: v3.0
I (73) boot.esp32: SPI Speed      : 40MHz
I (78) boot.esp32: SPI Mode       : DIO
I (82) boot.esp32: SPI Flash Size : 4MB
I (87) boot: Enabling RNG early entropy source...
I (92) boot: Partition Table:
I (96) boot: ## Label            Usage          Type ST Offset   Length
I (103) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (111) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (118) boot:  2 app0             OTA app          00 10 00010000 00270000
I (126) boot:  3 app1             OTA app          00 11 00280000 00150000
I (133) boot:  4 spiffs           Unknown data     01 82 003d0000 00020000
I (141) boot:  5 coredump         Unknown data     01 03 003f0000 00010000
I (148) boot: End of partition table
I (153) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=86c94h (552084) map
I (360) esp_image: segment 1: paddr=00096cbc vaddr=3ffbdb60 size=05a7ch ( 23164) load
I (370) esp_image: segment 2: paddr=0009c740 vaddr=40080000 size=038d8h ( 14552) load
I (376) esp_image: segment 3: paddr=000a0020 vaddr=400d0020 size=133850h (1259600) map
I (832) esp_image: segment 4: paddr=001d3878 vaddr=400838d8 size=19998h (104856) load
I (890) boot: Loaded app from partition at offset 0x10000
I (890) boot: Disabling RNG early entropy source...
I (902) cpu_start: Multicore app
I (902) cpu_start: Pro cpu up.
I (902) cpu_start: Starting app cpu, entry point is 0x40082ad0
I (0) cpu_start: App cpu up.
I (922) cpu_start: Pro cpu start user code
I (922) cpu_start: cpu freq: 160000000 Hz
I (923) cpu_start: Application information:
I (927) cpu_start: Project name:     nukihub
I (932) cpu_start: App version:      331ff3e-dirty
I (938) cpu_start: Compile time:     Aug  4 2024 10:20:35
I (944) cpu_start: ELF file SHA256:  0f6d83b6939f6362...
I (950) cpu_start: ESP-IDF:          5.1.4.240629
I (955) cpu_start: Min chip rev:     v0.0
I (960) cpu_start: Max chip rev:     v3.99 
I (965) cpu_start: Chip rev:         v3.0
I (970) heap_init: Initializing. RAM available for dynamic allocation:
I (977) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (983) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (989) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (995) heap_init: At 3FFCBF80 len 00014080 (80 KiB): DRAM
I (1001) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1007) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1014) heap_init: At 4009D270 len 00002D90 (11 KiB): IRAM
I (1022) spi_flash: detected chip: generic
I (1025) spi_flash: flash io: dio
I (1033) coexist: coex firmware version: d96c1e51f
I (1035) app_start: Starting scheduler on CPU0
I (1040) app_start: Starting scheduler on CPU1
mI (1040) main_task: Started on CPU0
I (1050) main_task: Calling app_main()
I (1078) main_Partition size: 2555904
Partition subtype: 16
Nuki Hub version 9.00
Nuki Hub build 10235422676.1.1
I (1085) BTDM_INIT: BT controller compile version [1a95809]
I (1088) BTDM_INIT: Bluetooth MAC: 08:b6:1f:38:1e:9e
I (1095) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
W (1099) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
�ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

flashing
```
# sudo esptool --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq keep --flash_size detect 0xe000 boot_app0.bin 0x1000 bootloader.bin 0x10000 nuki_hub_esp32.bin 0x280000 nuki_hub_updater_esp32.bin 0x8000 nuki_hub.partitions.bin
esptool.py v4.7.0
Serial port /dev/ttyUSB0
Connecting.....
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:b6:1f:38:1e:9c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x001edfff...
Flash will be erased from 0x00280000 to 0x003b0fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 713.6 kbit/s)...
Hash of data verified.
Compressed 26704 bytes to 16691...
Wrote 26704 bytes (16691 compressed) at 0x00001000 in 0.5 seconds (effective 429.7 kbit/s)...
Hash of data verified.
Compressed 1954368 bytes to 1179992...
Wrote 1954368 bytes (1179992 compressed) at 0x00010000 in 19.3 seconds (effective 809.8 kbit/s)...
Hash of data verified.
Compressed 1246944 bytes to 753386...
Wrote 1246944 bytes (753386 compressed) at 0x00280000 in 12.2 seconds (effective 819.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 149...
Wrote 3072 bytes (149 compressed) at 0x00008000 in 0.1 seconds (effective 327.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-08-13:
When searching for your restart reason (`rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)`) multiple sources suggest doing a full erase of the ESP and reflashing and that this can occur when switching from Arduino to ESP-IDF (which we did with 9.0, although I haven't encountered your issue on any of my 6 devices).

You can use https://technyon.github.io/nuki_hub/ for erasing and flashing.

AlbrechtL, 2024-08-13:
I also tried https://technyon.github.io/nuki_hub/ - same behavior.

When I'm using 8.35 I see the following:
```
Nuki Hub version 8.35
Nuki Hub build 9596430697.1135.1 (release)

Brownout detector was triggered

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

The interesting thing is that I never saw `Brownout detector was triggered` in my other projects. From what I know is that the brownout detector is triggered when there is a power issue.  But I never had a power issue before with these USB cables.

iranl, 2024-08-13:
When the esp is recalibrating RF it uses more power and this can cause a brownout on a power supply that usually works.

Try another power supply.

closed 2024-08-13 by AlbrechtL