technyon/nuki_hub #462: Nuki Hub "Nuki Lock state" updates in web front end, but does not propagate to Home Assistant

theorician, 2024-08-22:
### PROBLEM DESCRIPTION
On the Nuki Hub's device site, the "Nuki Lock state" updates correctly (to 'locked', 'unlocked', 'motorBlocked' etc) but this does not always make it down to the lock's entity state in Home Assistant.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32 D1 mini
- [x] Nuki Hub binary firmware version number used: 9.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info:
```
System Information

Nuki Hub version: 9.00
Nuki Hub build: 10235422676.1.1
Nuki Hub build type: Release
run: true
confVersion: 900
deviceId: ***
deviceIdOp: ***
nukiId: ***
nukidOp: 
mqttbroker: ***
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: true
websrvena: false
lockena: true
lockpin: 1
mqttpath: nuki
openerena: false
openerpin: 
openercont: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
maxtc: 
opmaxtc: 
enabtlprst: false
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: 
hassConfigUrl: 
buffsize: 4096
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
nwwififb: false
rssipb: 60
hostname: nukihub
nwbestrssi: false
nettmout: 60
restdisc: true
rstbcn: -1
lockStInterval: 1800
tcPerEntry: false
kpPerEntry: false
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpCntrlEnabled: true
kpInfoEnabled: false
kpPubCode: false
tcCntrlEnabled: false
tcInfoEnabled: false
cnfInfoEnabled: true
regAsApp: false
regOpnAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: ***
crdpass: ***
disnonjson: false
pubAuth: false
pubdbg: false
prdtimeout: 
offHybrid: false
hybridTimer: 600
hybridAct: false
hybridRtry: false
hasmac: false
macb0: 
macb1: 
macb2: 
latest: 9.00
tsksznetw: 12288
tsksznuki: 8192
authmaxentry: 5
kpmaxentry: 10
tcmaxentry: 10
updMqtt: true
showSecr: false
bleTxPwr: 
recNtwMqttDis: false
MQTT connected: Yes
Lock firmware version: 3.10.7
Lock hardware version: 6.11
Lock paired: Yes
Lock valid PIN set: Yes
Lock has door sensor: No
Lock has keypad: No
Lock ACL (Lock): Allowed
Lock ACL (Unlock): Allowed
Lock ACL (Unlatch): Allowed
Lock ACL (Lock N Go): Allowed
Lock ACL (Lock N Go Unlatch): Allowed
Lock ACL (Full Lock): Allowed
Lock ACL (Fob Action 1): Allowed
Lock ACL (Fob Action 2): Allowed
Lock ACL (Fob Action 3): Allowed
Lock config ACL (Name): Allowed
Lock config ACL (Latitude): Allowed
Lock config ACL (Longitude): Allowed
Lock config ACL (Auto Unlatch): Allowed
Lock config ACL (Pairing enabled): Allowed
Lock config ACL (Button enabled): Allowed
Lock config ACL (LED flash enabled): Allowed
Lock config ACL (LED brightness): Allowed
Lock config ACL (Timezone offset): Allowed
Lock config ACL (DST mode): Allowed
Lock config ACL (Fob Action 1): Allowed
Lock config ACL (Fob Action 2): Allowed
Lock config ACL (Fob Action 3): Allowed
Lock config ACL (Single Lock): Allowed
Lock config ACL (Advertising Mode): Allowed
Lock config ACL (Timezone ID): Allowed
Lock config ACL (Unlocked Position Offset Degrees): Allowed
Lock config ACL (Locked Position Offset Degrees): Allowed
Lock config ACL (Single Locked Position Offset Degrees): Allowed
Lock config ACL (Unlocked To Locked Transition Offset Degrees): Allowed
Lock config ACL (Lock n Go timeout): Allowed
Lock config ACL (Single button press action): Allowed
Lock config ACL (Double button press action): Allowed
Lock config ACL (Detached cylinder): Allowed
Lock config ACL (Battery type): Allowed
Lock config ACL (Automatic battery type detection): Allowed
Lock config ACL (Unlatch duration): Allowed
Lock config ACL (Auto lock timeout): Allowed
Lock config ACL (Auto unlock disabled): Allowed
Lock config ACL (Nightmode enabled): Allowed
Lock config ACL (Nightmode start time): Allowed
Lock config ACL (Nightmode end time): Allowed
Lock config ACL (Nightmode auto lock enabled): Allowed
Lock config ACL (Nightmode auto unlock disabled): Allowed
Lock config ACL (Nightmode immediate lock on start): Allowed
Lock config ACL (Auto lock enabled): Allowed
Lock config ACL (Immediate auto lock enabled): Allowed
Lock config ACL (Auto update enabled): Allowed
Network device: Built-in Wi-Fi
BSSID of AP: ***
Uptime: 7 minutes
Heap: 77156
Stack watermarks: nw: 8260, nuki: 5096
Restart reason FW: ConfigurationUpdated
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

### TO REPRODUCE

Not entirely sure - I think this happens when the motor blocks. So:

1. Start the lock in the unlocked state and send a lock command via HA
2. Have the motor be blocked
3. The lock's state stays 'unlocked' in HA, but 'motorBlocked' on the Nuki Hub website
4. Send another lock command via HA
5. The lock usually sorts itself out after one or two tries of this - so that the lock is successfully in the 'locked' state. The website shows this as well. 
6. The HA state remains in 'unlocked'.

### EXPECTED BEHAVIOUR
I expect it to change to 'locked' once the lock really is locked.

### SCREENSHOTS

[image]
[image]

### ADDITIONAL CONTEXT
I tried restarting the hub in the hopes that it reads the state and sends it out. This does not work either. In general the website state and the provided mqtt state for HA are not consistent. 

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-08-22:
I'd start with (re-)enabling Home assistant in Nuki Hub, as according to your info page it is disabled at the moment (value of `hassdiscovery` is empty, should be `homeassistant`).

If the issue persists afterwards look at the value of MQTT topics with something like MQTT explorer and check it these correspond with the state of the lock, webserver and home assistant

theorician, 2024-08-22:
@iranl Damn, you're fast. Thanks so much.

This was clearly user error. When setting it up, this text:

> Home Assistant discovery topic (empty to disable; usually homeassistant)

read as 

> Home Assistant discovery topic (empty **for default,** usually homeassistant)

to me, so I just left it blank. Sorry. Works far better now - and also shows the 'locking/unlocking' intermediate states.

closed 2024-08-22 by theorician

iranl, 2024-08-22:
We have had many folks making the same mistake, probably should change this behaviour as I have basically never seen anyone change their ha discovery topic from the default.

Should maybe change this to a checkbox.

In 9.01 the info page has also changed to be much more readable which would have shown Home Assistant is disabled in your case

theorician, 2024-08-22:
@iranl Probably having 'enable home assistant autodisco' as a checkbox, and then 'disco topic (blank for default)' is the least surprising option here. Still, I feel bad. Couldn't even read the text properly. Sorry :)