technyon/nuki_hub #469: "Unable to Add Keypad Code via MQTT - Command Returns 'noCodeSet' Despite Successful Delete and Update Operations"

cyrililo, 2024-09-01:
First of all, I would like to express my gratitude for this project. I greatly appreciate the work that has gone into it. Once I am able to resolve this issue, I would like to contribute to and support the project further.

### PROBLEM DESCRIPTION

I am experiencing an issue when trying to add a keypad code via MQTT on the Nuki Hub. While the delete and update operations work perfectly, the add operation consistently fails with the response "noCodeSet". This occurs even when following the correct sequence of commands and ensuring that all the necessary parameters are set.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used   **ESP32** 
- [x] Nuki Hub binary firmware version number used: **9.00**
 - [x] Pre-compiled
- [x] Provide the output of http://nukihubIP/info:

### System Information

- Nuki Hub version: 9.00
- Nuki Hub build: 10235422676.1.1
- Nuki Hub build type: Release
- run: true
- confVersion: 900
- deviceId: [*****]
- deviceIdOp: [******]
- nukiId: [*********]
- nukidOp: 
- mqttbroker: [********]
- mqttport: 1883
- mqttuser: [****]
- mqttpass: [******]
- mqttlog: false
- checkupdates: true
- websrvena: false
- lockena: true
- lockpin: 1
- mqttpath: nuki
- openerena: false
- openerpin: 
- openercont: false
- mqttoppath: 
- maxkpad: 
- opmaxkpad: 
- maxtc: 
- opmaxtc: 
- enabtlprst: false
- mqttca: 
- mqttcrt: 
- mqttkey: 
- hassdiscovery: 
- hassConfigUrl: 
- buffsize: 4096
- dhcpena: true
- ipaddr: 
- ipsub: 
- ipgtw: 
- dnssrv: 
- nwhw: 1
- nwwififb: false
- rssipb: 60
- hostname: nukihub
- nwbestrssi: false
- nettmout: 60
- restdisc: true
- rstbcn: -1
- lockStInterval: 1800
- tcPerEntry: false
- kpPerEntry: false
- configInterval: 3600
- batInterval: 1800
- kpInterval: 1800
- kpCntrlEnabled: true
- kpInfoEnabled: false
- kpPubCode: false
- tcCntrlEnabled: false
- tcInfoEnabled: false
- cnfInfoEnabled: true
- regAsApp: false
- regOpnAsApp: false
- nrRetry: 3
- rtryDelay: 100
- crdusr: [masqué]
- crdpass: [masqué]
- disnonjson: false
- pubAuth: false
- pubdbg: false
- prdtimeout: 
- offHybrid: false
- hybridTimer: 600
- hybridAct: false
- hybridRtry: false
- hasmac: false
- macb0: 
- macb1: 
- macb2: 
- latest: 9.00
- tsksznetw: 12288
- tsksznuki: 8192
- authmaxentry: 5
- kpmaxentry: 10
- tcmaxentry: 10
- updMqtt: true
- showSecr: false
- bleTxPwr: 
- recNtwMqttDis: false
- MQTT connected: Yes
- Lock firmware version: 3.10.7
- Lock hardware version: 6.11
- Lock paired: Yes
- Lock valid PIN set: Yes
- Lock has door sensor: No
- Lock has keypad: No
- Lock ACL (Lock): Allowed
- Lock ACL (Unlock): Allowed
- Lock ACL (Unlatch): Allowed
- Lock ACL (Lock N Go): Allowed
- Lock ACL (Lock N Go Unlatch): Allowed
- Lock ACL (Full Lock): Allowed
- Lock ACL (Fob Action 1): Allowed
- Lock ACL (Fob Action 2): Allowed
- Lock ACL (Fob Action 3): Allowed
- Lock config ACL (Name): Allowed
- Lock config ACL (Latitude): Allowed
- Lock config ACL (Longitude): Allowed
- Lock config ACL (Auto Unlatch): Allowed
- Lock config ACL (Pairing enabled): Allowed
- Lock config ACL (Button enabled): Allowed
- Lock config ACL (LED flash enabled): Allowed
- Lock config ACL (LED brightness): Allowed
- Lock config ACL (Timezone offset): Allowed
- Lock config ACL (DST mode): Allowed
- Lock config ACL (Fob Action 1): Allowed
- Lock config ACL (Fob Action 2): Allowed
- Lock config ACL (Fob Action 3): Allowed
- Lock config ACL (Single Lock): Allowed
- Lock config ACL (Advertising Mode): Allowed
- Lock config ACL (Timezone ID): Allowed
- Lock config ACL (Unlocked Position Offset Degrees): Allowed
- Lock config ACL (Locked Position Offset Degrees): Allowed
- Lock config ACL (Single Locked Position Offset Degrees): Allowed
- Lock config ACL (Unlocked To Locked Transition Offset Degrees): Allowed
- Lock config ACL (Lock n Go timeout): Allowed
- Lock config ACL (Single button press action): Allowed
- Lock config ACL (Double button press action): Allowed
- Lock config ACL (Detached cylinder): Allowed
- Lock config ACL (Battery type): Allowed
- Lock config ACL (Automatic battery type detection): Allowed
- Lock config ACL (Unlatch duration): Allowed
- Lock config ACL (Auto lock timeout): Allowed
- Lock config ACL (Auto unlock disabled): Allowed
- Lock config ACL (Nightmode enabled): Allowed
- Lock config ACL (Nightmode start time): Allowed
- Lock config ACL (Nightmode end time): Allowed
- Lock config ACL (Nightmode auto lock enabled): Allowed
- Lock config ACL (Nightmode auto unlock disabled): Allowed
- Lock config ACL (Nightmode immediate lock on start): Allowed
- Lock config ACL (Auto lock enabled): Allowed
- Lock config ACL (Immediate auto lock enabled): Allowed
- Lock config ACL (Auto update enabled): Allowed
- Network device: Built-in Wi-Fi
- BSSID of AP: [masqué]
- Uptime: 7 minutes
- Heap: 77156
- Stack watermarks: nw: 8260, nuki: 5096
- Restart reason FW: ConfigurationUpdated
- Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.

### TO REPRODUCE

1. Use MQTT Explorer to connect to the Nuki Hub.
2. Send the following JSON to the `nuki/keypad/actionJson` topic to add a new keypad code:
   ```json
   {
     "name": "TestUser",
     "code": "654321",
     "enabled": 1,
     "action": "add"
   }

![Capture d'écran 2024-09-01 170447](https://github.com/user-attachments/assets/0d41d8be-4257-46c3-b8d9-f94c44a3addf)

Observe the commandResultJson topic which returns the message "noCodeSet", indicating the add operation failed.

### EXPECTED BEHAVIOUR

When sending a JSON command to add a new keypad code via MQTT, I expected the Nuki Hub to successfully add the code and confirm the operation by returning a success message in the `commandResultJson` topic.

Specifically, I expected the following:
- The new code ("654321") should be added to the keypad configuration.
- The `commandResultJson` topic should return a confirmation message, such as `"success"`, indicating that the operation was completed without issues.
- The new code should be visible in the corresponding `codes` topics under `nuki/keypad`.

However, despite sending the correct JSON payload and following the proper sequence, the Nuki Hub returns `"noCodeSet"`, and the code is not added.

### ADDITIONAL CONTEXT
- I have ensured that the JSON payload is correctly formatted and adheres to the expected structure (e.g., 6-digit code, no zeros, does not start with "12").
- The delete and update commands work as expected, but the add command consistently fails with `"noCodeSet"`.
- I attempted to retrieve more detailed logs via MQTT Explorer, but the Nuki Hub only provides limited feedback under the `commandResultJson` topic.
- The Nuki Hub firmware and software are up to date, and the device is connected to the MQTT broker successfully.
- Given that the add command fails but delete and update work, there may be a bug specifically affecting the add operation.

Please let me know if additional information or specific logs are required to help diagnose this issue.

iranl, 2024-09-01:
This has already been fixed in current master. You can either wait for 9.01 to be released or update to the latest development build (9.01-master16) from the ota page

closed 2024-09-01 by iranl

cyrililo, 2024-09-01:
Voici une réponse que vous pouvez utiliser pour remercier et confirmer que la mise à jour a résolu le problème :

Thank you for the quick response!

I have updated to the latest development build (9.01-master16) as suggested, and I can confirm that the issue is now resolved. The add command for keypad codes is working perfectly.

I appreciate the support and all the work that goes into maintaining this project. Thanks again!

cyrililo, 2024-09-01:
Thank you for the quick response!

I have updated to the latest development build (9.01-master16) as suggested, and I can confirm that the issue is now resolved. The add command for keypad codes is working perfectly.

I appreciate the support and all the work that goes into maintaining this project. Thanks again!